pascallanger/DIY-Multiprotocol-TX-Module #301: Multistm opentx aetr inv 1.3.0.45 V911S protocol

blankmac, 2019-11-30:
The last commit to the V911S protocol breaks binding for me on the Jumper T16 running OpenTx 2.3.3 N54.  If I roll back to 1.3.0.44, then it will bind and function without any problems.  Is there something else I can try with the current change?  Or would it be possible to have both methods as sub protocols?

pascallanger, 2019-12-02:
I was also thinking on having both selectable, it's not possible due to how the code is currently written. But I'll have a look when time permits.

blankmac, 2019-12-02:
Sounds great, thank you.   I’ll revert the commit and compile on my own for now.  :-D

blankmac, 2019-12-30:
Hey Pascal, any movement on this?  I did donate (not that it means you owe me anything, lol, people use the crap out of my work without forking over a little...)  If I knew more about this hardware I would take a crack at it myself but I have no sense of what is appropriate...thx.

pascallanger, 2020-01-02:
I've done some modifications time ago but there was a bug in OpenTX preventing me to push it.
I think my code has now made it to the OpenTX nightlies so I'll do some tests and push the changes in Multi.
Basically a RF freq tuning value at 0 will use the nrf24l01, a value different of 0 will use the CC2500 to emulate the XN297L @ 250Kbps.
Pascal

pascallanger, 2020-01-12:
Please test Multi v1.3.0.53 just released. It should fix your issue.
Protocols using the XN297L@250K emulation layer (V911S, KF606, GD00x,...): XN297L@250kbps is emulated by default with the NRF24L01. If option (freq tune) is different from 0, the CC2500 module (if installed) will be used instead with option being the usual frequency tuning.

blankmac, 2020-01-12:
Test and working.  Thank you!

closed 2020-01-12 by blankmac