pascallanger/DIY-Multiprotocol-TX-Module #304: Extend Telemetry for GAM and EAM Values

Flying-Bolt, 2019-12-04:
Is there already a planning in which version the other telemetry sensor values will be implemented by GAM (General Engine Module) or EAM (Electric Air Module)?

pascallanger, 2019-12-04:
I don't have any HoTT hardware anymore (sent back to the owner). If you have sensors then we could look to see if the RX is reporting something to the radio using telemetry when you plug them. If not then I have no way to know what to send to the RX to get the info. If it does then we can see if it's possible to decode them.
To do this you would need to connect your module to a computer using serial, flash it in debug mode and copy the output of the serial monitor.
It requires time and effort so if you are willing to look into it we can try.
Pascal

HazzaHFPV, 2019-12-06:
Hello Pascal,

I am also interested in getting as much functionality into the HOTT protocol as possible as I do have many HOTT receivers, but I am about to switch to the Jumper T16 pro and I also have two irangex IRX4 Plus modules which I used to upgrade my old 35Mhz radio to 2.4Ghz.
I use the "GR-12L SUMD +T" micro receiver in all 5 of my quadcopters. I don't have a GAM or EAM but the flight controllers flashed with betaflight report back as a EAM when HoTT telemetry is selected. They also deliver GPS data to the receiver. My Graupner radio is currently send in for service (MZ-24), so I have a few receivers lying around (GR-12L SUMD +T, GR-16, GR-32) doing nothing for some time (and I do not fly much during the winter).
I don't have the knowledge on how to analyse or even record the relevant data, but would be willing to try or give my support in any way possible.
Best regards
Harry

HazzaHFPV, 2019-12-06:
I don't have a clue about such stuff but would data from within the files of the following Made4RC github help?

https://github.com/Made4RC/HoTTServer
https://github.com/Made4RC/HoTTClient

pascallanger, 2019-12-07:
I really need a RX and a couple of sensors to figure out how it works. If you are willing to ship these parts to me in France I'll have a look.
Pascal

HazzaHFPV, 2019-12-07:
I don't own any sensors at this moment. I currently only use the HOTT telemetry data supplied by my betaflight flight-controllers (F4 and F7 boards). They identify as EAM (and GPS if a GPS is connected to the FC).
However, as my interest in this is quite high, I would be willing to buy some sensors and send them to you, maybe I can find some used ones for a reasonable price.
Would an EAM be sufficient  or would you need a GAM as well?

Flying-Bolt, 2019-12-09:
Did you get my email with the source code?

pascallanger, 2019-12-09:
> Did you get my email with the source code?

I don't recall seeing an email. Have you used my yahoo address? What source code are you atlking about?

pascallanger, 2019-12-09:
> I don't own any sensors at this moment. I currently only use the HOTT telemetry data supplied by my betaflight flight-controllers (F4 and F7 boards). They identify as EAM (and GPS if a GPS is [...]

I need to look at the HoTT word more closely. I don't know what EAM and GAM really means in term of hardware. Iwas under impression that you can plug a simple voltage sensor or a variometer. But listening to you it looks more like the Hitec stations where you plug all the sensors. Is it what you mean by that?
I really need a RX in fact. May be with the link you sent and wiring a "HoTTServer" will give me some clue on how telemetry works over the air.

HazzaHFPV, 2019-12-09:
I am not yet an expert on this matter, as I don't have any sensors/modules, but I do use the HoTT telemetry in my multi-rotors which run on the betaflight firmware.
Here is what I know:

There are quite a few HoTT-sensors that you can use which you simply plug into the T-socket of the receiver such as a vario, GPS, RPM-sensor, voltage-sensor, termperature, etc. As far as I know you can connect several sensors using a Y-adapter cable for example. <- So your assumption was absolutely correct!

Then there are modules, such as the Electric Air Module (EAM). This module already includes the following sensors and is made with electric crafts in mind:

-vario (+ altimeter)
-Vbat
-Current
-capacity
-single-cell monitoring up to 14S
-option for 2 temperature sensors and 2 current sensors

The General Air Module (GAM) is very similar, includes many of the same sensors with the additional option for fuel-metering and RPM sensors for combustion-engines.

Each module can be accessed via the telemetry-screen on an HoTT-transmitter and programmed to output different warnings (maximum altitude for example, etc)

When using an F3, F4 or F7 flight-controller for multi-rotors, flashed with the betaflight firmware, you can configure it to output telemetry data in the HOTT-format. You then connect the telemetry pad of the Rx with the telemetry UART of the flight controller, which will identify itself as an EAM and delivers the following telemetry-data (provided the flight controller has the relevant sensors on board, or connected):

-vbat
-current
-vario (altitude)
-GPS

TL;DR 
So there are following sources for telemetry data:
- Rx itself (RSSI, Signal information, temperature) 
- Single sensors (vario, GPS, etc.) which you just plug into the Rx
- Modules (which include a number of sensors and options)
- Flight controllers (betaflight), which identify as EAM/GPS

I wrote this from limited knowledge and from memory, hope my claims are largely correct.

I can provide you a Receiver (a "GR-12L SUMD +T" and a "GR-16") and soon with a EAM and GAM which I ordered, if it helps. I would love to do more to help, but I simply don't have the skills.

Flying-Bolt, 2019-12-09:
> > Did you get my email with the source code?
> 
> I don't recall seeing an email. Have you used my yahoo address? What source code are you atlking about?

Let me surprise you. What is your Yahoo email address?

Flying-Bolt, 2019-12-09:
> > > Did you get my email with the source code?
> > 
> > 
> > I don't recall seeing an email. Have you used my yahoo address? What source code are you atlking about?
> 
> Let me surprise you. What is your Yahoo email address?

:D I found your eMail address in your source code: pascal_langer@yahoo.fr right?

pascallanger, 2019-12-09:
yes, the one you can see if you do a donnation!

Flying-Bolt, 2019-12-09:
detonation...boom...and the last thing you see is your email address. Cool :+1:  (joke)

Flying-Bolt, 2019-12-11:
I got error message using your Email address:

This is the mail system at host mailout08.t-online.de.

I'm sorry to have to inform you that your message could not
be delivered to one or more recipients. It's attached below.

For further assistance, please send mail to postmaster.

If you do so, please include this problem report. You can
delete your own text from the attached returned message.

                   The mail system

<pascal_langer@yahoo.fr>: host mx-eu.mail.am0.yahoodns.net[188.125.72.74] said:
    421 4.7.0 [TSS04] Messages from 194.25.134.20 temporarily deferred due to
    user complaints - 4.16.55.1; see
    https://help.yahoo.com/kb/postmaster/SLN3434.html (in reply to MAIL FROM
    command)

any idea? detonation?

pascallanger, 2019-12-11:
It basically says that your IP had been reported to send spam and had been temporarily disabled...

HazzaHFPV, 2019-12-12:
Hello Pascal,

I know it is a hassle and that Graupner HoTT is not used by too many people within the hobby, but if you would be interested in pursuing this further I have now bought and can provide you with some hardware if needed to do so and I am willing to send them to you. I live in Germany, so sending to France is not a problem. If interested you can find my public E-Mail address in my profile and contact me directly. A donation from my side is of course assured :)

A betaflight developer pointed out to me that the following might help as well (but I guess you might already be aware of it):

https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/hott.c
https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/hott.h

I also found these projects that seem to include HoTT data:

https://github.com/obayer/MultiWii-HoTT/blob/master/MultiWii_2_1/HoTTv4.ino
https://github.com/3yc/hott-for-ardupilot/tree/master/PX4/hott-px4-code/px4-hott-module

Harry

hgo-be, 2019-12-20:
Hello Pascal,
Implementation of the graupner hott sensor modules could be very simple as the implementation can be read from the open source project http://www.nongnu.org/dataexplorer. The hott protocol is implemented in the HottAdapter. 

Hendrik

pascallanger, 2019-12-20:
Sadly not since that doesn't tell me how the data is sent over the air... All the links you are sending are about what the RX reads on its input. But this is not the point here, the goal is to know what to send to the RX so it sends this data over. It does not do it by itself and the data is also coded to be transported over the air. So I don't see how it can be "very simple".
May be I'm missing something. If that's as simple as you say please submit a PR with the appropriate code change in Multi and OpenTX.
Pascal

hgo-be, 2019-12-20:
Sorry , I was not aware of the two way communication.
Hendrik 
PS  I appreciate your superb work, I have donated 10€ last week.

wimalopaan, 2020-02-16:
Do you need a Hott-TX and Rx? Or just some sensors?

pascallanger, 2020-02-16:
I have everything I need, just need some time to work on it.

wimalopaan, 2020-02-16:
Oh, then I misinterpreted the previous posts. I read that you had to pass the Hott stuff back to the owner. 

Since Graupner/SJ in Germany is effectively dead and in the foreseeing future the will be only some distributing support from Robbe and D-Power, I think that a lot of people now are interested in using some other TX with the SJ-RX stuff. 

In my personal opinion this is a really great feature! Thanks a lot!

HazzaHFPV, 2020-02-16:
You interpreted correctly, but in the meantime someone else supplied a new set of equipment to Pascal.

HazzaHFPV, 2020-03-16:
@pascallanger just wondering if your received my last mail or if it might have landed in your spam-folder :) I know there is a lot going on around the project at the moment and that you are really busy, but maybe you had a chance to have another glance at the Hott protocol?

pascallanger, 2020-03-16:
@HazzaHFPV Sorry but I'm extremely busy right now. The coronavirus is putting a lot of pressure on all my professional projects and my time is fully dedicated to it. I'm not expecting much progress within the next 2 weeks or so. I apologize but...

HazzaHFPV, 2020-03-16:
No worries and absolutely no need to apologize. Just hoped for a little update and, as always, thanks for the hard work :)

HaleyACS, 2020-03-27:
@pascallanger In case you need some more hardware, let me know. I have plenty Graupner stuff, and I live in France (Picardie, Clermont de L'oise).

Herman-54, 2020-04-16:
I have a Kakute F7 1.5 flightcontroller and a GR-16 Hott receiver from Graupner. The problem for as long as I build this 5" freestyle quadcopter is, that there is no connection between those 2. Can anybody help me? Thanks, Herman

HazzaHFPV, 2020-04-16:
Hi Herman,
I have both, a gr-16 and an kakute F7 here at home. I guess I can help you. How did you connect the GR-16 with the board? Feel free to e-mail me, you can find my address in my profile so that we don't spam this topic :) greetings Harry

pascallanger, 2020-04-21:
I'm starting to look at the HoTT protocol. Sorry it took so long but just getting back to it today.
I'm already getting some info from the ESC (Batt, rpm, temp,...) so I'm getting somewhere ;-)

HazzaHFPV, 2020-04-21:
Awesome! Pascal, you rule ;)

And for anyone else who might have a similar problem as Herman and happend to land here:
Don't forget to define the Signal output ("CH OUT TYPE: ...") in the telemetry menu of the receiver.
Further information can be found in the manuals of the receivers.

pascallanger, 2020-04-22:
@HazzaHFPV I've spent 4-5 hours yesterday asking why I couldn't get the ESC text config to appear. Finally I took the mz-24 where I can see that it can't access it either. Is it expected?

HazzaHFPV, 2020-04-22:
I am sorry to say that I can only work from memory. I only bought the ESC to have as many sensors as possible for you and have not used it thoroughly myself, only tested it for the setup that I sent you. I have another MZ-24 which I borrowed from a friend currently. I will write you an E-Mail with some pictures, maybe we can figure this out together this way.

HaleyACS, 2020-04-22:
On my MC-32 I have to go into the "RX Telemetry" menu and go right until it won't  go any further. There I can set the channel out type: CH OUT TYPE:SUMI

pascallanger, 2020-04-22:
> I am sorry to say that I can only work from memory. I only bought the ESC to have as many sensors as possible for you and have not used it thoroughly myself, only tested it for the setup that I sent you. I have another MZ-24 which I borrowed from a friend currently. I will write you an E-Mail with some pictures, maybe we can figure this out together this way.
> 
> EDIT:
> I just sent you the promised mail with pictures. In the telemet. menu you have to scroll to the last page, select the sensor, which makes a new page appear, which you can then access by pressing "ENT" once more. This should bring you to the settings-page of that sensor.

This is what I've checked on the mz-24 at 2am last night and even when you click on "ESC" it does not display the text menu at the end of the RX menu. If I do the same with the GPS and EAM they do display correctly, so I'm thiking I'm doing the right thing and the original Graupner is not dowing anything better than me... They are saying in the doc that the RX and ESC versions should match to get the text to display, not sure if that could be the case here as I don't know how (heven't looked) on how to upgrade the ESC. I'll just continue the implementation like if it does work and will see later.

pascallanger, 2020-04-22:
> On my MC-32 I have to go into the "RX Telemetry" menu of the ESC and go right until it won't go any further. There I can set the channel out type: CH OUT TYPE:SUMI

Yeah this is another thing that through me off and made me lost a lot of time. You need to go throught the RX menu if you want to see the sensor you selected. I've noticed that if you do DEC (or INC I can't remember)  on the first RX page, it goes to the sensor directly. There doesn't seem to have a way to tell the TX just give me the text page of the sensor I'm asking for...

pascallanger, 2020-04-22:
I'm planning to auto detect the sensors like they do. Meanning that only sensors plugged to the RX a the time of a HoTT model load (or TX powered on with a HoTT model) will be polled. It's waiting for the RX to be up right to assess the sensors in any case.

Otherwise I can poll all sensors all the time which is easier but that slows down the updates of the sensors... Not sure if this is really a problem in fact...

HazzaHFPV, 2020-04-22:
> They are saying in the doc that the RX and ESC versions should match to get the text to display, not sure if that could be the case here as I don't know how (haven't looked) on how to upgrade the ESC. I'll just continue the implementation like if it does work and will see later.

I remember spending an evening updating all firmwares and testing that all the sensors worked and are accessible. I even vaguely remember programming the ESC, but it was with some trouble I think. It could well be that I made a mistake with the ESC or it might be that this particular ESC is not playing along nicely. The only further suggestions that I could find in a German forum was to make sure that TX and RX are not too close to each other (at least 1m apart) and one guy reported having trouble programming his ESC in a HF-noisy environment (WiFi) and that it worked better in the field, but that sounds more like a coincidence and placebo-effect than anything else. I noticed that I have the Graupner USB interface cable for updating sensors and Receivers still lying around here. If you require it, I could mail it to you. Don't understand why I didn't include it in the original package.
Did I include the USB stick with firmware and manuals? If not I could mail you the different firmware *.bin files to try, but just continuing and having a further look later is probably the least frustrating path. I know that trial and error with different firmwares eats up a lot of time.

> I'm planning to auto detect the sensors like they do. Meaning that only sensors plugged to the RX at the time of a HoTT model load (or TX powered on with a HoTT model) will be polled. It's waiting for the RX to be up right to assess the sensors in any case.

Sounds good, but I know too little about these things as to give constructive input, which I am really sorry about :/ All I know is that once you get everything is working, Graupner is great. BUT it can be finicky and annoying to get to that point sometimes. I trust your professional judgement will find the best way, and I thank you for it :)

HaleyACS, 2020-04-22:
There was a major change in the firmware history of the Rx that required a recent Firmware on the Tx, so I think you may have hit that one.

Regarding the sensor detection, HoTT does it the way you described (Autodetect at Power-on). Once the sensors are detected, you will not add another one. IMHO, efficiency is the better approach here. If, once the sensors are detected, we can free CPU cycles for the rest of the system, that is always better.

HazzaHFPV, 2020-04-22:
I can almost (98% sure) guarantee that the Tx, Rx and sensors are on the latest Firmware. The only one that I might not quite be so certain about is the ESC.
Firmware Versions are
Tx MZ-24 Pro: v2.029
Rx GR-16: v7a07

HaleyACS, 2020-04-23:
Latest for MZ-24 Pro is: v2.029
Latest for GR-17: 7a07

So you are set with the latest. Should be fine.

pascallanger, 2020-04-23:
> Regarding the sensor detection, HoTT does it the way you described (Autodetect at Power-on). Once the sensors are detected, you will not add another one. IMHO, efficiency is the better approach [...]

It does not free CPU but instead increase the refresh rate of the connected sensor. 
Example: if you have 1 sensor only, you can poll only this sensor instead of trying all the time the 6 possible sensors. 1 server poll is 50ms, so if you poll 1 sensor you have values every 50ms, if you poll the 6, you have values every 300ms. Add to that the loss of over air and you could receive your single sensor every minute or so...

What I've done is the following:
 - at model load and after RX is connected, it scans through all the 6 sensors for a few seconds. Then it stops searching and only poll that have been found. Therefore you can't add a sensor after that search period.
 - if the RX is disconnected for 1 sec, the sensor scan will be relaunched. It allows you to plug a new sensor, power cycle the RX and get the new sensor without having to fiddle with the radio.

There is at power up a sporadic sensor that appears every 10-20 power up... I think this is a bug in their RX software. I don't really want to filter it out so it will stay like this...

Moving now to the Text config mode, leaving the decoding of sensors which is the hardest part for the end.

HaleyACS, 2020-04-23:
Ok. That makes sense (The d sensor polling). thanks for clarifying!
And the "There is at power up a sporadic sensor that appears every 10-20 power up" may be a bug that HoTT has not detected yet.

pascallanger, 2020-04-23:
Text config mode is working for RX and all sensors. You can cycle through the sensors with the Menu/MDL key.
The cycle goes through all the sensors even if not discovered. I hope it's ok since that makes it easier for me...
On the big color screen the sensor being viewed is displayed in the title bar. I need to figure out a way to display it on smaller screens (it cycles but you don't know where you are in the cycle).

The telemetry is now reworked on the OpenTX code side. It can accept all the incoming telemetry from all sensors. Now it needs to be able to decode all these data...

HaleyACS, 2020-04-23:
Cool :) for the text mode!

"The cycle goes through all the sensors even if not discovered. I hope it's ok since that makes it easier for me..."
The question will be: Does the Sensor actually provide telemetry?
The discussions we had in the lounge about that was: it is better to have displayed only what is discovered. This way - if you have the wrong sensor connected, you'll see it ;}

Anyway - having a working sample is the first step :) So do as you think is best!

pascallanger, 2020-04-24:
The text mode now cycles only through the RX and discovered sensors.

HazzaHFPV, 2020-04-30:
This all sounds really good. I know I sound like a broken record, but thanks for all the effort.

sbuhre, 2020-05-04:
Hi Pascal,
I'm also interested in seeing the telemetry feature for HoTT. If needed I can do some testing: 
I have GR12,GR16 receivers and Voltage-Module, Vario-Modul, PowerBox Competition SRS, Unilog 2 (very popular in Germany) ready for testing.
Stephan

HazzaHFPV, 2020-05-13:
Hi @pascallanger 
I have now started setting up my Jumper T16 with the internal Multi-Module and I have noticed something odd while testing the HoTT protocol with my "Gr-12L SMD +T" receivers (product No. S1045 and S1046) - the small one which I also sent you.

My setup:
- Jumper T16 with OpenTX 2.3.7
- Internal Multi-Module firmware Version 1.3.0.95
- Frequency Tuning performed (+26)
- Graupner S1045 and S1046 receivers (these are the same, difference is that one comes pre-wired). On latest firmware.

The Symptoms:
The receiver is experiencing what seems like mini signal-losses? When bound the red LED on the receiver should stay off. This is what I confirmed by borrowing a Graupner MZ-24 from a friend. However, if I bind the receiver with the T16 using the Multi-Module, the LED comes on very briefly about every 7 seconds. Mostly it just flashes briefly, but once in a while it stays on a little longer for about 0.5 - 1.0 seconds or so.

Behaviour in flight:
The Quadcopter flies normally for the majority of the time. Those really brief signal-losses/failsafes are unnoticed (meaning the quick flashes). Once in a while the quad freezes in flight for about 0.5 - 0.8 seconds before I regain control. The delay for a full failsafe (disarming) in betaflight is a little longer. I think, so it did not lead to a crash yet, but when you are inverted over a tree it gets your heartrate up when the quad doesnt react.

What I have tried:
- unbound and rebound the receiver with the T16.
- bound the receiver with a Graupner MZ-24 to verify the behaviour (LED does not flash)
- verified firmware of the receiver
- repeated the frequency tuning procedure (result stayed the same)
- tried 3 different receivers of the same kind. Behaviour stayed the same.
- disabled Telemetry in the T16 menu.

More Data:
Here is a short video of the flashing:
https://www.youtube.com/watch?v=pTtmoKq7Lb0

Here is an extract of a betaflight Log, RC Commands received by the Flight Controller:
https://www.youtube.com/watch?v=DA22eQ_7F-Y&feature=youtu.be

Here is a Betaflight Blackbox Log if it helps, one event can be seen at timestamp 00:49 and lasts about 0.85 seconds:
https://www.dropbox.com/s/kxfjybtys4kv9fc/LOG00030.BFL?dl=0

I tried my best to eliminate as many variables as possible and I think I could narrow it down to the T16. I do not observe that behaviour when using the same radio with FrSky xm+ receivers (D16 EU LBT protocol).
You are more qualified than I am to verify or rule out if there is a bug in the HoTT protocol.
Anyway, just wanted to share my observation and thank you for your help. I also dropped you an E-Mail about the equipment, but I'm not sure if you have seen it yet.
All the best and kind regards
Harry

HazzaHFPV, 2020-05-14:
Okay. So I found out that this behaviour can only be observed when I configure Failsafe in the T16 to "Hold", "Custom" or "no pulses".
The Failsafe settings then seem to be transmitted to the receiver every 7 seconds I guess? The LED does not flash if I set the Failsafe to "Receiver" or "Not set".

When I define the Failsafe positions in the MZ-24 and safe them, the LED on the receiver only flashes once briefly and then stays off.

Tomorrow I will test the flight behaviour once more and do another Blackbox Log with Failsafe set to "Receiver" in the T16, with which the failsafe settings don't seem to be sent every 7 seconds. Maybe it upsets the Receiver once in a while having to receive and save the failsafe positions on such a regular basis?

tfode, 2020-05-14:
> 
> 
> Okay. So I found out that this behaviour can only be observed when I configure Failsafe in the T16 to "Hold", "Custom" or "no pulses".

Had the same issue not long ago (...), this is how to actually set it up: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#HoTT---57

HazzaHFPV, 2020-05-14:
@tfode 
Damn. Thanks for the hint! I normally am quite thorough with reading instructions, but this I completely missed. I do feel stupid now...

HazzaHFPV, 2020-05-18:
Hey @pascallanger 
I unfortunately am failing to reach you via E-Mail. I know you are really busy, but I would like to talk about the returning of the Graupner equipment. I have a few projects standing still atm and due to corona currently no job, which means I can't simply buy what I need. The situation regretfully makes me have to shift my priorities, so I am willing to give up on extending the HoTT Telemetry in the Multi-Protocol project.
Anyway, drop me an E-Mail when you have time.
On a final note: I hope this does not sound harsh in any way, I do appreciate all the effort you are putting into this.
Harry

pascallanger, 2020-05-23:
Current status of Multi v1.3.1.1 release:
- Full telemetry support including auto sensors discovery
- Full RX and connected sensors text config using the HoTT LUA script
Tested with the following HoTT sensors: Vario, GPS, ESC, EAM and GAM.
It should be available in OpenTX 2.3.8 (PR opentx/opentx#7627)

Todo: Implement LBT on the HoTT protocol

pascallanger, 2020-05-24:
Telemetry video:
[image]
Text mode video:
[image]

Flying-Bolt, 2020-05-24:
well done

wimalopaan, 2020-05-24:
> Telemetry video:
> Telemtry video <https://www.youtube.com/watch?v=geEbRglUB1A>
> Text mode video:
> Text mode video <https://www.youtube.com/watch?v=81wd8NlF3Qw>

How do I switch the Sensor?

I only get the values of the receiver in Binary-mode.

In text-mode the receiver-pages and the next sensor page ist displayed.

pascallanger, 2020-05-24:
@wimalopaan Have you compiled OpenTX with the PR mentionned above? Have you updated the Graupner HoTT LUA script from the PR above? Have you upgraded your module to v1.3.1.1?
If yes to all above then press Menu has written on the TX screen when you launch the Graupner HoTT LUA script. Menu being whatever you press to get to the model config.

HaleyACS, 2020-05-25:
Hi Pascal,
I have heard from some people that they re unable to bind the "Falcon 12" Rx.
In case you are interested, I could send you one over (PS - I do not own a T16 so can't test. But I am always open to Opensource development).

pascallanger, 2020-05-25:
@HaleyACS This is strange...
Why this RX will be different from a binding standpoint? Working on my bench, I have a GR-12, GR-12L and GR-16, all working fine. 
Do you know if they were running a recent version of the Multi firmware?
Do you know if they have been using the RF freq tune? This is really important for HoTT since al lmy modules need around a +50 to work and don't bind with the default value of 0. (the doc says to try -40,0,+40 to get a bind then fine tune it)
Do you know if the LED on the RX was doing anything special?
I'm not against the shipping, just trying to rule out any issues before doing so.
Pascal

[5 more replies]

closed 2020-06-02 by pascallanger