pascallanger/DIY-Multiprotocol-TX-Module #307: Jumper T16 with inner model can not bind with AC900 receiver

SamFisher940425, 2020-01-06:
AC900 receiver use CC2500 chip. It can support S-FHSS (8ch), D16 non-EU (14ch), D16 EU-LBT (14ch) modes. I can bind it using my Frsky x9d se (2018), but Jumper T16 with inner 4in1 model can not bind it in D16 non-EU and D16 EU-LBT modes. I have tried to adjust Freq fine tune, but it still couldn't work. But at the same time, my Jumper T16 can bind with Frsky xm+ receiver. So, I'm confused now......by the way, In S-FHSS mode, T16 and AC900 can bind and work well.

pascallanger, 2020-01-07:
Strange... I've never heard an issue on binding with these protocols...
Is the RX doing anything when you launch a bind (LED change)? You tried to bind with a rf freq of -40 and +40?
I'm not sure what to try and I don't have this RX to play with.
Pascal

SamFisher940425, 2020-01-07:
thanks for your answer. I have tried to bind with all of rf freq (from -128 to +127), but it doesn't work. The LED on RX still blink, because it is waiting to bind.
Jumper T16 and Frsky xm+ can work well.
Frsky X9D and AC900 can work well.
Jumper T16 and AC900 can not bind with each other......It's so strange.....
I suspect that there is something wrong with the receiver protocol, what do you think?

pascallanger, 2020-01-07:
-128 to +127 are too big values, you've tried with other values, correct?
The RX LED does not even change...
I don't know what can differ between the X9D and Multi for D16. I'll see what I can come up with.
Pascal

SamFisher940425, 2020-01-07:
thank you so much......I have tried every number between -128 to +127, it still can not work. haha

pascallanger, 2020-01-07:
When you bind with the X9D are you using any of the options like 8ch, no telemetry or change channels to 9-16 instead of 1-8
Pascal

SamFisher940425, 2020-01-07:
uh....X9D.....channel range is ch1-16(118ms), and both ch1-8 telem on and ch1-8 telem off can bind with AC900, ch9-16 telem off/on can not bind with AC900.

pascallanger, 2020-01-07:
I've verified and I can't see a difference between a Frsky radio and what multi sends...
Can you try to bind with different RX numbers?
Frsky has a bug on some RX which are not working with some rx num so may be it applies here as well.
After that I'm clueless.

pascallanger, 2020-01-28:
No more comments for more than 20 days. Assuming the problem fixed itself.

closed 2020-01-28 by pascallanger