pascallanger/DIY-Multiprotocol-TX-Module #308: X-Lite Pro Fails to Intialize MPM Lite

Errant33, 2020-01-07:
On opentx version 2.3.4 with multi-stm-opentx-taer-inv-v1.3.0.47 loaded on the MPM lite, the transmitter is using MPM Lite to do FRSky D8 protocol when this occurs

-If you switch between internal module and mpm lite will cause the mpm lite to have its status lights turn amber instead of green. 
- This also occurs when the model is on FRSky D8 from the MPM lite and it switches between different protocols on the MPM lite or to another FRsky d8 protocol. 
- This persists thru reboots
- The model is set to model 00
- To resolve and get the MPM lite initializing back to a green status light I am putting the transmitter on a model that is a r9m configured model, then switching it back to a model for one of the MPM lite settings

pascallanger, 2020-01-07:
It's a known issue where the bootloader which is being shipped is either really old or has been modified: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/237
At the bottom of the page above you'll find a document on how to do the upgrade which I'm linking here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/3594850/Vantac.MPM.Lite.Red.LED.Issue.docx.
Pascal

closed 2020-01-07 by pascallanger

gretel, 2020-02-01:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/3594850/Vantac.MPM.Lite.Red.LED.Issue.docx