pascallanger/DIY-Multiprotocol-TX-Module #312: xlite pro DSM prot.invalid

matejbludsky, 2020-01-11:
Hi, thanks so much for the software ! I upgraded my board to the latest version and when i select DSM (2 or X doesn't matter) when i hit bind it will just say prot. invalid in the menu and i cannot bind anything. 

Thx for support 

[image]

pascallanger, 2020-01-11:
Prot invalid appears only when you bind but it doesn't prevent to bind...
I haven't looked at why it displays that.
Your DSM issue is different. Please retry and make sure you are not too close to the RX when binding since there is a negotiation between the RX and TX at bind time.
Pascal

matejbludsky, 2020-01-11:
Thx for the response, you were correct i made a mistake during the binding.. it works now!

closed 2020-01-12 by pascallanger