pascallanger/DIY-Multiprotocol-TX-Module #315: Why not support radiolink?

frizzle101101, 2020-01-14:
Radio link seems to use the cc2500 why is their fhss protocol not available

pascallanger, 2020-01-14:
Like for adding any protocols, I need at least a TX, a RX and some sensors if telemetry is supported.
So basically no one has ever proposed to send me any Radio link equipment.
Pascal

frizzle101101, 2020-01-14:
Is it difficult to reverse engineering the protocol's?

pascallanger, 2020-01-14:
Yes, it's days/weeks of work.

frizzle101101, 2020-01-14:
Can i ask what ur strategies are for that?

pascallanger, 2020-01-14:
Not sure what you are asking for... Like I've said it's a long process with data analysis and depending on how the protocol is built it can be done or not.

frizzle101101, 2020-01-14:
Thats what im wondering do you follow a set of data analysis and testing
tequniques that you found was preety common when you were doing all the
other protocol's to init or ack the protocol. I'm just wondering how
1person is providing all the reverse engineering of the 2.4ghz protocols
which have been around for like 10years how do you get all that know how?
Or has this been in the works for years and just now took traction cuz the
t16 needed a tx module lol

frizzle101101, 2020-01-14:
I am saying this is amazing work gj dude

pascallanger, 2020-01-14:
Unless you have some hardware to ship in my direction, I'll close this issue.
Pascal

closed 2020-01-14 by pascallanger