pascallanger/DIY-Multiprotocol-TX-Module #317: R-XSR Frysky Firmware update no longer binds

spyder0069, 2020-01-18:
Frysky released a new accst firmware update 1/16/2020 for the R-XSR labeled v2.0.1.  This firmware will no longer bind with the previous taranis or multiprotocol module.  There is a v2.0.1 firmware also for the xjt internal and external module on the taranis which is also required to work with the new receiver firmware.  Notes from their firmware page are: 

1.Fixed the channel output error (uncontrolled servo movements) under certain conditions.

2.Strengthened correction and verification capability.
Note: Please update the firmware of all your radios, RF modules and receivers accordingly.

pascallanger, 2020-01-19:
I'm aware of it...
Pascal

closed 2020-01-19 by pascallanger

gretel, 2020-01-20:
@pascallanger thanks.. so change is coming?

pascallanger, 2020-01-20:
It has been released only 4 days ago... Give us the time to work on it.
They've added protections in the protocol so it's not a simple change to improve their protocol but instead an anti reverse engineering directed to this project and RX clones.
So the answer is may be yes, may be no. Wait and see...
Pascal

gretel, 2020-01-20:
didnt mean to push anybody.. take your time.. keep the good work up. thanks for the info!