pascallanger/DIY-Multiprotocol-TX-Module #319: Will FrSky accst version 2.0.1 be implemented.

Gaylian, 2020-01-21:
When will FrSky accst version 2.0.1 be implemented.

pascallanger, 2020-01-28:
All I can say is that I'm working on it. Time will tell.

MartinHugh, 2020-01-28:
That is good news, and it will take as long as it does, but are you in a position yet to judge whether an implementation will be feasable.  Or have they obfuscated/encrypted it so well that it is unlikely to be completed ?

Thanks.

pascallanger, 2020-01-28:
I can't say for now. There are multiple copy/clone protections which have been added. So like any protocol reversing, may be yes, may be no...

frizzle101101, 2020-02-03:
I was wondering if the issues they presented in the bug fix list are really issues or not kinda seems like bs to me to just give you a hard time, has anyone ever experienced these issues with the revere engineered accst: 
1.Fixed the channel output error (uncontrolled servo movements) under certain conditions. (sounds like possibly the jitters you see at center stick caused by noise, which seems to vary on the quality of the gimbals)
2.Strengthened correction and verification capability. (sounds like maybe added encryption)

MartinHugh, 2020-02-03:
https://frsky-forum.de/thread/3404-historie-und-zusammenfassung-der-wichtigsten-informationen-zum-thema-plötzliche/

MartinHugh, 2020-02-04:
As this problem seems to result from the RX's inability to deal with a frame error, can we would assume that the FrSky TX software is not actually at fault ?

If we can't assume this, and for some reason the frame error is not induced by a RF corruption in transit between the TX and the RX, and the reason is down to the TX software (in addition to the RX not being able to deal with bad data), what are the chances of introducing a similar problem into the multiprotocol implementation of FrSky TX code ?

pascallanger, 2020-02-04:
@MartinHugh The issue is protocol based therefore the problem exists in both the FrSky and Multi code. The RX software could have ben improved to be more robust but they decided to modify the protocol instead.
@frizzle101101 There is no encryption. Whils solving the issue by simply enabling the hardware CRC, they've added some anti copy/clone protections. This is different to encryption unless you consider simple XORs to be encryption...

MartinHugh, 2020-02-20:
Pascal, any movement on this one ?   How is the de-obfuscation going?

Gaylian, 2020-03-25:
Pascal, not sure if I am interpreting it wrong but in the multi.txt line 64 it seems to say FrskyX2,CH_16,CH_8,EU_16,EU_8 is there. I don't get on my TX display so I take it hasn't been done yet

pascallanger, 2020-03-25:
This is just a preparation work. The protocol code is not present therefore it won't show up on the TX.

Gaylian, 2020-03-25:
Thanks for info

pascallanger, 2020-03-29:
Frsky x V2.1.0 implemented in multi 1.3.0.79.
Please test and report.

MartinHugh, 2020-03-30:
Thank you for this.  Will attempt to update and do some bench testing in the coming days.  Does this update take account of the changes to the rx firmware last week (versioned as 2.1.0), e.g. for rx6r :

https://www.frsky-rc.com/rx6r/

pascallanger, 2020-03-30:
> Thank you for this. Will attempt to update and do some bench testing in the coming days. Does this update take account of the changes to the rx firmware last week (versioned as 2.1.0), e.g. for rx6r :
"Frsky X v2.1.0 implemented" means that I've implemented the FrSky release from last week versioned as ACCST D16 v2.1.0. The version numbers are matching, isn't it?
> Also, could you please link to the firmware, I can't see it at
> https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases
A release will only be added if reports are good. For now you need to build from source: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md
Pascal

pascallanger, 2020-03-30:
It's now available as a pre-release: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/1.3.0.79
Test the new FrSky X v1.x and v2.1.0 code (D16 FCC/LBT) but **fly with caution** and report success/failures.

MartinHugh, 2020-03-31:
Thank you for your work on this Pascal.  Please see report...

I am unable to fly easily at the moment because of the lockdown here in the UK
However I did some bench and range checks as follows

Transmitter
    Jumper T16
    Multimodule 1.3.0.79
    OpenTX      238N

Recievers
    X4R #1      X4RSP_EU_170322      (Frsky V1)
    X4R #2      X4R_ACCST_2.1.0_LBT  (Frsky V2)    

X4R #1  (Frsky V1)    
==================
Part of an existing aircraft installation
Was already bound to MULTI FrSky LBT 8Ch (1-8 with Telemetry)

I exercised channels 1-4 and all seemed normal
RSSI in the vicinity of the aircraft was 80-90 dbi

I did a range test and got to 40m (1.2Km) before RSSI warning at 40dbi triggered

All appears normal

X4R #2  (Frsky V2)    
==================
This was loose reciever mounted over 1m off the ground on a plastic bin.
I bound this to  MULTI FrSkyX2 LBT 8Ch (1-8 with Telemetry)

I connected 4 servos and exercised channels 1-4 and all seemed normal
RSSI in the vicinity of the aircraft was 80-90 dbi

I did a range test and got to 40m (1.2Km) before RSSI warning at 40dbi triggered

All appears normal

pascallanger, 2020-03-31:
So it all looks good from perspective. Nice!

Gaylian, 2020-04-01:
All OK on XM+ connected to a Omnibus F4 V3 FC.  RSSI on chan 16 works OK.  Haven't flown yet but it looks good. Thank you for all your hard work.

Dropsleutel, 2020-04-01:
So after reversing, whats your technical conclusion any benefits to move over to their new standard? 
Or basically just added some difficulties in protecting their designs?

pascallanger, 2020-04-02:
Like I've said since the begining, the only thing they have done to improve the reliability of the link is to enable hardware CRC. In fact, they could have just done this and be done.
But instead they have added additionnaly some copy protections. In V2.0.1, there was a copy protection on both the TX and RX sides during bind and normal packets. In V2.1.0, they went backward with only a protection during bind.
The protection I refer to is in fact a bundle of 4 all located in the bind packets:
1. Simple xor for obfuscation
2. 2 protections which are not reversed but the protections are not checked carefully enough so there is a way to trick the system.
3. In FrSkyD and FrSkyX v1.x the TX was sending during bind the hopping frequencies to be used by the TX and RX during normal operation which is not the case anymore in V2.x.x. It took some time and efforts to figure out but in the end this protection has been fully reversed.
Basically with what I've published, TX and RX clones are possible again until they release a new version...
Pascal

closed 2020-04-16 by pascallanger