pascallanger/DIY-Multiprotocol-TX-Module #323: FrSky D8 protocol lagging

Windyfpv, 2020-01-26:
Hello,
When pairing a FRSKY D8 receiver as the QX65 from Eachine, it seems the model is lagging to respond in Betaflight receiver and tab and currently lagging in real life.
After research, it looks like that when selecting the D8 mode in the internal module menu of the T16, the channels available is from 1 to 16 and not from 1 to 8 which could induce latency.
Is this a known issue?
Thank you.

pascallanger, 2020-01-26:
You need to tune the frequency as explained in the doc: https://www.multi-module.org/using-the-module/protocol-details/frsky-d

Windyfpv, 2020-01-27:
Hello Pascal,
Sorry to bother you again. Tried this method but still not working. When I pair my drone with my FRSKY X9d, it works like a charm but does not with the T16. On the T16 and as mentioned before, the bars on the rc page on betaflight are very laggy to respond. On the x9d, it's working perfectly. Do you have any ideas?

mariusair, 2020-01-28:
> Hello Pascal,
> Sorry to bother you again. Tried this method but still not working. When I pair my drone with my FRSKY X9d, it works like a charm but does not with the T16. On the T16 and as mentioned [...]

Hello Windyfpv, Pascal,
I had same trouble as you mentioned. My Jumper T16 with old frsky with D8 protocol got signal delay issue. When I push key to other side, the movement is delayed about 0.5S after key input.
Of course I updated latest firmware Multi module and opentx at all.
Below are 2 kinds of receiver got that error.
Jumper answered about this issue to change other frsky receiver but I am sure some protocol issue for old D8 frsky receiver. My D16 receiver and D8 with telemetry receiver hasn't this delay issue at all.
1) 2.4G 4CH Mini Frsky D8 Compatible Receiver With PWM
https://m.banggood.com/2_4G-4CH-Mini-Frsky-D8-Compatible-Receiver-With-PWM-Output-p-1143300.html
2) FB90 Quad with F3 EVO, builtin frsky d8 receiver
https://www.banggood.com/FRF3_EVO_BRUSHED-Flight-Controller-Built-in-Frsky-8CH-Sbus-Receiver-For-Eachine-QX95-QX90-QX90C-p-1105907.html?rmmds=detail-left-hotproducts__4&cur_warehouse=CN

mariusair, 2020-01-28:
Hello Windyfpv, Pascal,
I had a same trouble as you mentioned. My Jumper T16 with old frsky with D8 protocol got signal delay issue. When I push key to other side, the movement is delayed about 0.5 after key input.
Also recovery movement of each key is delayed about 0.5s after release of key to other side.
Of course I updated latest firmware Multi module and opentx at all.
Below are 2 kinds of receiver got that error.
Jumper answered about this issue to change other frsky receiver but I am sure some protocol issue for old D8 frsky receiver. My D16 receiver and D8 with telemetry receiver hasn't this delay issue at all.
1) 2.4G 4CH Mini Frsky D8 Compatible Receiver With PWM
https://m.banggood.com/2_4G-4CH-Mini-Frsky-D8-Compatible-Receiver-With-PWM-Output-p-1143300.html
2) FB90 Quad with F3 EVO, builtin frsky d8 receiver
https://www.banggood.com/FRF3_EVO_BRUSHED-Flight-Controller-Built-in-Frsky-8CH-Sbus-Receiver-For-Eachine-QX95-QX90-QX90C-p-1105907.html?rmmds=detail-left-hotproducts__4&cur_warehouse=CN

pascallanger, 2020-01-28:
I've never heard of this issue before and D8 has been out since a long time...
Is this problem appearing only when the receiver does not have telemetry back to the radio?
Can you compile the code without telemetry enabled to see if it changes anything?
Pascal

Windyfpv, 2020-01-28:
I am afraid I don't have the knowledge to recompile. Is there any tutorials on that?

pascallanger, 2020-01-28:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md

Windyfpv, 2020-01-28:
Hello Pascal,
I have made it to this part but don't know where to start to remove telemetry. Can you help me out please?

#if defined(FRSKYD_CC2500_INO)

#include "iface_cc2500.h"

static void __attribute__((unused)) frsky2way_init(uint8_t bind)
{
	FRSKY_init_cc2500(FRSKYD_cc2500_conf);	

	CC2500_WriteReg(CC2500_09_ADDR, bind ? 0x03 : rx_tx_addr[3]);
	CC2500_WriteReg(CC2500_07_PKTCTRL1, 0x05);
	CC2500_Strobe(CC2500_SIDLE);	// Go to idle...
	//
	CC2500_WriteReg(CC2500_0A_CHANNR, 0x00);
	CC2500_WriteReg(CC2500_23_FSCAL3, 0x89);
	CC2500_Strobe(CC2500_SFRX);
	//#######END INIT########		
}
	
static void __attribute__((unused)) frsky2way_build_bind_packet()
{
	//11 03 01 d7 2d 00 00 1e 3c 5b 78 00 00 00 00 00 00 01
	//11 03 01 19 3e 00 02 8e 2f bb 5c 00 00 00 00 00 00 01
	packet[0] = 0x11;                
	packet[1] = 0x03;                
	packet[2] = 0x01;                
	packet[3] = rx_tx_addr[3];
	packet[4] = rx_tx_addr[2];
	uint16_t idx = ((state -FRSKY_BIND) % 10) * 5;
	packet[5] = idx;
	packet[6] = hopping_frequency[idx++];
	packet[7] = hopping_frequency[idx++];
	packet[8] = hopping_frequency[idx++];
	packet[9] = hopping_frequency[idx++];
	packet[10] = hopping_frequency[idx++];
	packet[11] = 0x00;
	packet[12] = 0x00;
	packet[13] = 0x00;
	packet[14] = 0x00;
	packet[15] = 0x00;
	packet[16] = 0x00;
	packet[17] = 0x01;
}

static void __attribute__((unused)) frsky2way_data_frame()
{//pachet[4] is telemetry user frame counter(hub)
	//11 d7 2d 22 00 01 c9 c9 ca ca 88 88 ca ca c9 ca 88 88
	//11 57 12 00 00 01 f2 f2 f2 f2 06 06 ca ca ca ca 18 18
	packet[0] = 0x11;             //Length
	packet[1] = rx_tx_addr[3];
	packet[2] = rx_tx_addr[2];
	packet[3] = counter;//	
	#if defined TELEMETRY
		packet[4] = telemetry_counter;
	#else
		packet[4] = 0x00;
	#endif

	packet[5] = 0x01;
	//
	packet[10] = 0;
	packet[11] = 0;
	packet[16] = 0;
	packet[17] = 0;
	for(uint8_t i = 0; i < 8; i++)
	{
		uint16_t value;
			value = convert_channel_frsky(i);
		if(i < 4)
		{
			packet[6+i] = value & 0xff;
			packet[10+(i>>1)] |= ((value >> 8) & 0x0f) << (4 *(i & 0x01));
		} 
		else
		{
			packet[8+i] = value & 0xff;
			packet[16+((i-4)>>1)] |= ((value >> 8) & 0x0f) << (4 * ((i-4) & 0x01));
		}
	}
} 

uint16_t initFrSky_2way()
{
	Frsky_init_hop();
	packet_count=0;
	if(IS_BIND_IN_PROGRESS)
	{
		frsky2way_init(1);
		state = FRSKY_BIND;
	}
	else
	{
		state = FRSKY_BIND_DONE;
	}
	return 10000;
}	
		
uint16_t ReadFrSky_2way()
{ 
	if (state < FRSKY_BIND_DONE)
	{
		frsky2way_build_bind_packet();
		CC2500_Strobe(CC2500_SIDLE);
		CC2500_WriteReg(CC2500_0A_CHANNR, 0x00);
		CC2500_WriteReg(CC2500_23_FSCAL3, 0x89);		
		CC2500_Strobe(CC2500_SFRX);//0x3A
		CC2500_WriteData(packet, packet[0]+1);
		if(IS_BIND_DONE)
			state = FRSKY_BIND_DONE;
		else
			state++;
		return 9000;
	}
	if (state == FRSKY_BIND_DONE)
	{
		state = FRSKY_DATA2;
		frsky2way_init(0);
		counter = 0;
		BIND_DONE;
	}
	else
		if (state == FRSKY_DATA5)
		{
			CC2500_Strobe(CC2500_SRX);//0x34 RX enable
			state = FRSKY_DATA1;	
			return 9200;
		}
	counter = (counter + 1) % 188;	
	if (state == FRSKY_DATA4)
	{	//telemetry receive
		CC2500_SetTxRxMode(RX_EN);
		CC2500_Strobe(CC2500_SIDLE);
		CC2500_WriteReg(CC2500_0A_CHANNR, hopping_frequency[counter % 47]);
		CC2500_WriteReg(CC2500_23_FSCAL3, 0x89);
		state++;
		return 1300;
	}
	else
	{
		if (state == FRSKY_DATA1)
		{
			#ifdef MULTI_SYNC
				telemetry_set_input_sync(9000);
			#endif
			len = CC2500_ReadReg(CC2500_3B_RXBYTES | CC2500_READ_BURST) & 0x7F;
			if (len && len<=(0x11+3))// 20bytes
			{		
				CC2500_ReadData(packet_in, len);				//received telemetry packets
				#if defined(TELEMETRY)
					if(packet_in[len-1] & 0x80)
					{//with valid crc
						packet_count=0;
						frsky_check_telemetry(packet_in,len);	//check if valid telemetry packets and buffer them.
					}
				#endif
			}
			else
			{
				packet_count++;
				// restart sequence on missed packet - might need count or timeout instead of one missed
				if(packet_count>100)
				{//~1sec
					packet_count=0;
					#if defined TELEMETRY
						telemetry_link=0;//no link frames
						packet_in[6]=0;//no user frames.
					#endif
				}
			}
			CC2500_SetTxRxMode(TX_EN);
			CC2500_SetPower();	// Set tx_power
		}
		CC2500_Strobe(CC2500_SIDLE);
		CC2500_WriteReg(CC2500_0A_CHANNR, hopping_frequency[counter % 47]);
		if ( prev_option != option )
		{
			CC2500_WriteReg(CC2500_0C_FSCTRL0,option);	// Frequency offset hack 
			prev_option = option ;
		}
		CC2500_WriteReg(CC2500_23_FSCAL3, 0x89);
		CC2500_Strobe(CC2500_SFRX);        
		frsky2way_data_frame();
		CC2500_WriteData(packet, packet[0]+1);
		state++;
	}				
	return state == FRSKY_DATA4 ? 7500 : 9000;		
}
#endif

mariusair, 2020-01-29:
Hi Pascal

Please see attached mov file.

Mov file is explainng what I mentioned on github comment.

Thanks for your interesting and kindness

- mariusair

pascallanger, 2020-01-29:
The only thing you have to do is to open the _config.h file and look at the telemetry section.
You can disable telemetry completly by commenting #define TELEMETRY
Pascal

mariusair, 2020-01-29:
Need more information to compile without Telemetry.
T16 have to use noinv option, but I found invert_telemetry as default on config.h.
How can I disable invert_telemetry?
I used only disabled Telemetry bin file to update, I've got below error message. 
"Needs file stm-opentx-noinv" It might need noinvert_telemetry bin file. 
Tell me where can I find good help file to compile with detail options as Telemetry, Noinv, and so on.

Windyfpv, 2020-02-03:
Hello Pascal,
Did you have time to take a look at this issue?
Seems the telemetry trick does not work.
Thank you.

pascallanger, 2020-02-03:
Have you really tried to fully disable globally the telemetry? "Seems" is a vague word. The test posted above from mariusair does not disable telemetry...
I've got no hardware to reproduce your issue or test anything... So I've got no idea on how to do anything on this subject unless you do the tests to narrow it down.

Windyfpv, 2020-02-03:
OK Pascal, I will totally remove telemetry and I'll let you know.

tony0057, 2020-02-12:
Hello everyone i have exactly the same problem did you find a solution thank you very much .......

mariusair, 2020-02-12:
> Hello everyone i have exactly the same problem did you find a solution thank you very much .......

Compile with disable telemetry got error message  "Need file stm-opentx-noinv" when firmup to multi-module. But I didn't resolve that.

tony0057, 2020-02-15:
thank you for your answer i will keep looking for a solution

Typho81, 2020-02-22:
Hi,
Unfortunately I experienced the same issue! 
My setup is irangex irx4 lite and eachine beecore v2.0.
I use the multiprotocol in serial with a taranis x9 lite the strange thing is that ch range is 16 instead of 8. 
I tried to recompile the fw in different ways end different versions starting from 1.2.xx and something is changed respect the fw I found at the beginning, right now seems that the value read by receiver is not stable it change between the right value set by the tx and the minimum value.
I hope this can helps.
Let me know if you need logs or some other stuff.

Typho81, 2020-02-27:
Hi guys,
below some debug... I hope it helps...
let me know how if you need some additional info

20:19:30.431 -> Multiprotocol version: 1.3.0.73
20:19:30.529 -> Protocol selection switch reads as 0
20:19:30.629 -> Read ID from EEPROM
20:19:30.629 -> Module Id: 66788c22
20:19:30.629 -> Init complete
20:19:41.782 -> Protocol selected: 3, sub proto 0, rxnum 0, option 0
20:19:41.782 -> Proto=FrSky D, nbr_sub=0, Sub=, Opt=2
20:19:41.782 -> Short CB:18000
20:19:41.782 -> Short CB:15000
20:19:41.782 -> Short CB:2600
20:20:13.362 -> Protocol selected: 3, sub proto 0, rxnum 1, option 0
20:20:13.362 -> Proto=FrSky D, nbr_sub=0, Sub=, Opt=2
20:20:13.760 -> Protocol selected: 3, sub proto 0, rxnum 2, option 0

pascallanger, 2020-02-28:
No the debug output does not help at all since there is no debug at the protocol level... And anyway I have no idea where the lagging could come from.
You are saying that the output goes from the right value and then minimum value. Is the minimum value the RX failsafe value or something else?
The FrSky D8 protocol has been stable for a long time. I don't see why only some receivers are affacted, they must do something different or expect a slightly different timing... Unless I have access to the hardware to troubleshoot there is little I will be able to do.

Typho81, 2020-02-28:
I discovered that the minimum values that I see was the setup of Fail-Safe the real problem is the rxsignal on receiver that is unstable.

Typho81, 2020-03-07:
Ok we are closer to finding the problem, I succefully tested the version 1.1.6 and it works fine! the bug has been introduced at version 1.2.0 and it's still present.
I tried to compare the two release but there is a bunch of differences.

tony0057, 2020-03-10:
I wanted to say thank you for sharing your work with us ..............

tony0057, 2020-03-11:
hello where did you find version 1.1.6 i can't find it thank you very much friendly Tony

Typho81, 2020-03-11:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.1.6

Typho81, 2020-03-11:
> hello where did you find version 1.1.6 i can't find it thank you very much friendly Tony

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.1.6

tony0057, 2020-03-11:
super nice thank you very much

pascallanger, 2020-03-11:
@Typho81 I've found 2 differences between the 2 versions. One would affect the telemetry so let's forget it.
The second is the hopping frequency table which I think could explain your issue. Strangely the RX should use whatever table the TX is sending during bind but it looks like your RX might not like some of the values. As a test, we can revert the hopping table to what it was and see if you get a difference. If that's the issue then all the D8 models will have to rebound...

pascallanger, 2020-03-11:
@Typho81 Are you able to compile the code by yourself?

mariusair, 2020-03-13:
> @Typho81 I've found 2 differences between the 2 versions. One would affect the telemetry so let's forget it.
> The second is the hopping frequency table which I think could explain your issue. [...]

I think all D8 receiver don't have this issue because latest d8 receiver works well at same evironment.
Maybe frsky updated something relevant to this issue. Latest d8 receiver don't have this kind of issue at all.
I hope it will be resolved to use the machine with aged d8 protocol.

Typho81, 2020-03-13:
> @Typho81 Are you able to compile the code by yourself?

Yes I am

Typho81, 2020-03-13:
I think this will close definitely the issue! :-) it works good but the RSSI seems a little worse than before... 

diff --git a/Multiprotocol/FrSkyD_cc2500.ino b/Multiprotocol/FrSkyD_cc2500.ino
index 49a5aad..24f0a8e 100644
--- a/Multiprotocol/FrSkyD_cc2500.ino
+++ b/Multiprotocol/FrSkyD_cc2500.ino
@@ -93,9 +93,27 @@ static void __attribute__((unused)) frsky2way_data_frame()
        }
 } 
 
+static uint8_t __attribute__((unused)) get_chan_num(uint16_t idx)
+{
+  uint8_t ret = (idx * 0x1e) % 0xeb;
+  if(idx == 3 || idx == 23 || idx == 47)
+    ret++;
+  if(idx > 47)
+    return 0;
+  return ret;
+}
+
+void FrskyD_init_hop(void)
+{
+  for(uint8_t i=0;i<50;i++)
+  {
+    hopping_frequency[i]=get_chan_num(i);
+  }
+}
+
 uint16_t initFrSky_2way()
 {
-       Frsky_init_hop();
+  FrskyD_init_hop();
        packet_count=0;
        if(IS_BIND_IN_PROGRESS)
        {

pascallanger, 2020-03-13:
Ok this is what I thought. The problem with this modification is that everyone will need to rebind all the RXs... I'm wondering if I should add a sub protocol to choose between one or the other table. The problem is that opentx might not be able to use it...

Typho81, 2020-03-13:
> Ok this is what I thought. The problem with this modification is that everyone will need to rebind all the RXs... I'm wondering if I should add a sub protocol to choose between one or the other table. The problem is that opentx might not be able to use it...

I get you, I'll try and I let you know

pascallanger, 2020-03-21:
Fixed in 1.3.0.75

closed 2020-03-21 by pascallanger