pascallanger/DIY-Multiprotocol-TX-Module #324: Telemetry Support for Devo protocol

richardclli, 2020-01-28:
Did the current Devo protocol supports telemetry?  I tried to bind a MiniCP and seems there is no way to get the battery voltage.

pascallanger, 2020-01-28:
Devo telemetry is not supported.

richardclli, 2020-01-29:
I see, is it possible to support telemetry?  I can try to work this out. ^.^

richardclli, 2020-01-29:
I will check Deviation source code for reference to see how can I make this work.  It will be good if you can point out any details that can speed me up.  Thanks.

pascallanger, 2020-01-29:
Deviation has all the code for telemetry. I don't have sensors to connect on a Devo RX to develop and test so I've never done anything in this area.
I could have a look but I'm running after time currently...

richardclli, 2020-01-30:
No problem, I can see if I can work this out.  I have contributed Deviation in the past as well, I think I can make this work.

richardclli, 2020-02-01:
I have checked the source code of deviation and multi protocol module, seems the Telemetry framework are quite different.  Can you help to point out how the multi protocol module send back received data back to OpenTX?  Thanks.

richardclli, 2020-02-17:
I have work out some code and finally figured out that a single piece of code in deviation splitted into 2 parts: telemetry packet sent out by multi-module and decoded by OpenTX, which means to make it works, I need to modify both multi-module and OpenTX sources.

For a fall back test, I will first try to convert the DEVO telemetry data into DSM format and see if it works first.

richardclli, 2020-02-17:
Please give me some comments on this direction if possible, thanks.

pascallanger, 2020-02-17:
If you want to try to send data to OpenTX and get it displayed quickly using frsky hub. It's the easiest. Look at the bayang protocol as an example.

richardclli, 2020-03-11:
I tried to worked out some coding from Deviation Devo source code.  I can manage to port and compile the callback function for Devo telemetry, and I can manage to push out some output to Frsky Hub by setting the v_lipo1, v_lipo2, RX_RSSI, etc.  See if you may have some ideas on the following problems:

1.  RX_RSSI = CYRF_ReadRegister(CYRF_13_RSSI)&0x1F;  This line can never get non-zero RSSI, I got this from DSM coding, is there are initialization required?

2.  After reading using CYRF_ReadDataPacketLen(packet, CYRF_ReadRegister(CYRF_09_RX_COUNT)); and unscrambling, I can never found valid telemetry packets packet[0] & 0xf0 == 0x30.  It is quite strange.

I have attached my source for reference, see if you may found any hints.

[Devo_cyrf6936.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4316737/Devo_cyrf6936.txt)

richardclli, 2020-04-07:
I finally make it works, but the problem is related to some shared code with DSM, so could you give me some advice why you add RX abort operations when calling doing CYRF_FindBestChannels?  I check the code in deviation, in which the deviation version do not have these 2 lines:

//	CYRF_WriteRegister(CYRF_29_RX_ABORT, 0x20);		// Abort RX operation
	CYRF_SetTxRxMode(TX_EN);
//	CYRF_WriteRegister(CYRF_29_RX_ABORT, 0x20);		// Clear abort RX

The telemetry works after I remarked these 2 lines.

richardclli, 2020-04-07:
I realized what is the problem, this is obviously a bug that the Clear abort RX line is wrong, should put 0x00 instead of 0x20.

richardclli, 2020-04-07:
I have created a pull request, please review.  I have tested it with my Walkera MiniCP.

richardclli, 2020-04-27:
@pascallanger any comments on my pull request?

pascallanger, 2020-05-09:
I've done many comments but they haven't been addressed.
I'll see if I have time to merge the code and improve it this WE.

pascallanger, 2020-05-09:
Since none of review have been addressed, I've imported the change and done the recommended modifications from my branch.
Devo basic telemetry is available in 1.3.0.96.

Please test and report since none of the 2 Devo RX I have sends telemetry back...

closed 2020-05-09 by pascallanger