pascallanger/DIY-Multiprotocol-TX-Module #326: MPM Lite Flash procedure failed

cipotter, 2020-02-01:
Hi all!
I just got the **Taranis X-Lite Pro** and bought the **Vantac MPM Lite** Module in order to be able to bind with my **D8** tiny whoops. However, I am not able to flash it correctly, although I thoroughly followed all the steps described here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Flash_from_Tx.md

I tried with both the firmware multi-stm-opentx-taer-inv-v1.3.0.53.bin and multi-stm-opentx-taer-noinv-v1.3.0.53.bin, changing the option "Inverterd Com Port" accordingly. I used the bootloader xlite_rom.bin which can be found in erskyTxR222D7.zip and used the latest Flash Multiprotocol Module App from http://www.er9x.com/Ersky9xapps.html.
The whole procedure seems to run well until I come to the last step and actually flash the firmware, which always ends with FAILED.

I also noticed that the module has following led pattern when I boot the transmitter normally:
Green: ON / Red: Slow flash (0,5s on, 0,5s off)
During the few seconds the flashing procedure lasts, both leds are on.

Then I flashed the Module with my PC using Flash Multi v.0.2.4 and it seemed to work so far. However, when I press bind on my X-Lite Pro, the LED pattern does not change. (Internal Module: OFF, External Module: XJT Lite D8). Do I need a nighlty build in order to be able to select another External Module?

Another hint: I tried to bind the whoops with a FrSky XJT Lite external module (which should be pretty much plug&play, cause there is no firmware to flash) and it still does not work. I tried with LBT and FCC mode, as well as with another X-Lite Pro (both flashed to the latest versions available in www.frsky-rc.com). Note that I am able to bind with any quadcopter with D16 (FCC or LBT), but also have trouble with the R9MLite Pro module.

I have to admit I am pretty lost here because I have no clue which step I did wrong and is preventing from binding my D8 whoops. Any kind of help will be appreciated, thanks in advance!

Kind regards

pascallanger, 2020-02-01:
You need to select MULTI for the module type and then the protocol you want.
Note: they are shipping the module with a buggy bootloader which you need to reflash. Look at the recent closed issus you will find a full documentation.

closed 2020-02-01 by pascallanger

gretel, 2020-02-01:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/3594850/Vantac.MPM.Lite.Red.LED.Issue.docx

cipotter, 2020-02-03:
Hello Pascal,

thanks for the fast answer! Sadly The X-Lite Pro does not permit the option MULTI in the external module and XJT Lite is the only one that allows to choose the D8 protocol. That's why I asked if it is necessary to flash a nightly build to the transmitter? And if yes, do you know which one works great?

Regarding the buggy bootloader, I think I already got it flashed via PC so it should be no problem anymore.

gretel, 2020-02-03:
https://www.open-tx.org/2020/01/15/opentx-2.3.5

pascallanger, 2020-02-03:
In companion you must check "multi" in the build options (settings->seetings-build option).
For the bootloader upgrade, you must do it through FTDI. USB is not enough.
Pascal

cipotter, 2020-02-03:
Hello Tom,

thanks for the information, but I think I am not facing this issue. The green LED on the module stays green and I was able to flash the module via PC with Flash Multi.
However, the RED LED has the slow blink pattern:
Slow blink(on=0.5s,off=0.5s): serial has been selected but no valid signal is being seen on the RX pin.
I'm not sure if I'm doing something wrong or if it is a Hardware problem. As I said, I don't have the option to choose MULTI for the External RF Mode, so I suppose it has something to do with that.

pascallanger, 2020-02-03:
Yes and again, in companion you must select multi as a build option.

cipotter, 2020-02-03:
Hello,

thank you both for the help, it worked! Just to summarize what I did if someone reads this in the future:
- Flash the MPM Lite Module via PC with Flash Multi
- Flash my transmitter with OpenTX Companion 2.3 (I selected following build options: lua, noheli, flexr9m) --> It seems that the 2.3 version does not permit to choose multimodule and is selected by default (it is more like you cannot deselect it)
- Choose External module --> MULTI FrSky --> D8
- Normal bind procedure.

Disclaimer: I don't know what you mean with "For the bootloader upgrade, you must do it through FTDI. USB is not enough." but I didn't do it and worked anyways, so lucky me!

Thanks again and regards!

gretel, 2020-02-04:
@cipotter as @pascallanger said, you need to flash using a serial (non-usb) connection to update the bootloader. looks to me like you did not update the bootloader yet. the procedure is described in the document i have referenced earlier.
same for updating opentx, you just have to do it.

cipotter, 2020-02-04:
Hello @gretel,
I think you are right, I just flashed the firmware (multi-stm-opentx-taer-inv-v1.3.0.53.bin) via USB with the Flash Multi app. 
However, it now works and can bind with my D8 whoops and fly them without problem. So I won't try anything else, you know, don't touch a running system.
Regarding the opentx update, the latest version of opentx companion does not let you choose the build option multimodule, it seems it is always selected by default and you cannot unselect it.

gretel, 2020-02-04:
> However, it now works and can bind with my D8

chances are it works sometimes, not any time. for me it was like had to power-toggle tx four..five times until the binding succedded.

> it seems it is always selected by default 

right. `opentx 2.3.5` runs stable and smooth for me, so you should not worry much getting there.

cipotter, 2020-02-04:
@gretel You are definitely right, I guess I just was lucky the first time. It seems that I also need 4-5 power-toggles until it works fine. However, I don't have currently the HW described in the word document, so I cannot solve the problem right now.

Do you know if once booted up correctly it also remains this way or is there a possibility that the connection with the receiver breaks suddenly? It would not be safety critical because I only use it for indoor tiny whoops, but would be still annoying.

Thanks!

pascallanger, 2020-02-04:
Even with different hardware it's the same process, just chek the pins in the "compiling for STM32" page.
It's only a problem with power up. Once powered it will be stable.

cipotter, 2020-02-04:
Thanks a lot for your support! I will buy the adapter and try the procedure.
Just to be sure:
- during flashing the boot and 3.3V pins near the usb connector have to be bridged, right? I ask because in the picture there is just one cable at each pin soldered but not connected to each other.
- I can power the MPM Lite module via USB with 5V and the voltage at the 3.3V will be the right one so I don't have to bother about that?

gretel, 2020-02-04:
> * during flashing the boot and 3.3V pins near the usb connector have to be bridged, right? I ask because in the picture there is just one cable at each pin soldered but not connected to each other.

on powerup, the bootloader will launch the actual payload. but if the `boot` pin is pulled "high" (at 3.3volt) it will stay in the bootloader - so it is accessible and can be updated.

i'm not sure if it is sufficient to just short it on bootup. did solder and desolder mine..

> * I can power the MPM Lite module via USB with 5V and the voltage at the 3.3V will be the right one so I don't have to bother about that?

yes. paranoid people would use closeby usb ports like on the same bus to avoid any current creeps.

cipotter, 2020-02-04:
Awesome, thanks!

As soon as I receive the adapter and can try it will tell you how it went.

pascallanger, 2020-02-04:
Just shorting boot0 at startup is enough. By this I mean, short boot0 to 3.3v, apply power and then it's fine to release.

cipotter, 2020-02-15:
Hi!
Just wanted to confirm that the procedure in the document @gretel posted worked perfectly. Now The module always boots up at first try, thanks a lot!
--> https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/3594850/Vantac.MPM.Lite.Red.LED.Issue.docx