pascallanger/DIY-Multiprotocol-TX-Module #328: SLT Protocol problems with Jumper T16 Internal or External

MRC3742, 2020-02-08:
As others have discussed a problem exists when trying to bind a Genuine TR624 RX (one antenna) to the Jumper Modules using Protocol SLT and Sub Protocol V1_ch6.

The RX will bind but then the solid Red LED will intermittently flash causing loss of control to servo outputs when LED is out. Switching to Low power mode will turn the RX LED on Solid and servos work correctly now with limited range. The iRangeX External MPM seems to be less prone to this problem? It has been the theory that the Jumper Modules' RF hardware is less accurate and is possibly working on the edge of the frequency with no option to fine tune the NRF24L01+ chip via software.

Please review the attached captures of a Genuine Tactic TTX610 Transmitter (6 channel)
and compare them to the current code for any problems, additions or errors?
[Captures Tactic TTX610 Transmitter.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4173797/Captures.Tactic.TTX610.Transmitter.zip)

Thanks - Rich

pascallanger, 2020-02-08:
I've started to look at the dumps and noticed that the SLT protocol is running at 250Kbps (which I've totally forgotten). This is why it works on iRangeX and not on Jumper. 250Kbps need a more accurate crystal/caps and this is the problem with all the Jumper modules and explaining why some do work a little better than others.
I need to modify the protocol to use the CC2500 instead of the NRF24L01 for this type of bitrate, it should give better results.
I need to find out how the CRC is calculated on the NRF24L01 as this would be the 1st protocol.

MRC3742, 2020-02-08:
I knew that was a problem emulating other RF chips at 250Kbps but didn't realize it to be a problem with the same chips.......... but makes sense now.

I can also help with any testing needed - Rich

pascallanger, 2020-02-17:
@MRC3742 Please test the SLT protocol in v1.3.0.69
I would like to see if it's still working even if the LED is not solid just to see if I haven't broken anything.
If it does work then change the RF Freq tune value from 0=NRF24L01 to 1=CC2500. I'm not expecting it to work fist try since I haven't tested anything but with some luck ;-).
Pascal

pascallanger, 2020-02-17:
@MRC3742 Try v1.3.0.70. CC2500 emulation should work in this one...

gretel, 2020-02-17:
@pascallanger not sure if i miss something.. latest release is at `v1.3.0.53`?

pascallanger, 2020-02-17:
> @pascallanger not sure if i miss something.. latest release is at `v1.3.0.53`?

You need to build from the source if you want the latest version 1.3.0.71. I'm releasing only when people confirm it's working...

MRC3742, 2020-02-17:
@pascallanger I just downloaded and compiled 1.3.0.71 and testing on SLT Protocol does not work at all. RF Tune at "0" are any other value it does not attempt to bind.

I have tried using all subprotocols and tested with RF Tune set to -120, -80, -40, 0, 40, 80, and 120. Unfortunately will not bind to TR624 RX or any of the previously working Ominus models.
Rich

pascallanger, 2020-02-18:
@MRC3742 Can you test the version .72? It should work again for the NRF. I'm not sure why the CC2500 version was not working, I've tried a change so let's see as well.

MRC3742, 2020-02-18:
Using 1.3.0.72 now "0" works with NRF same as 1.3.0.69 with intermittent LED flashing.
Changing to any other value of RF Tune caused LED to go out completely.

Should I need to reset Bind on RX when switch to CC2500?
I have tried both, with and without bind but still no response.

pascallanger, 2020-02-18:
Ok it's already a good step. I'll check what the cc2500 is sending later.

pascallanger, 2020-02-20:
@MRC3742 Can you test .73?
Stupid mistake which made me lose a day, it should have worked since the begining with the CC2500...

MRC3742, 2020-02-20:
@pascallanger .73 is a success and works with RF tune offset values other than "0"
It works OK but seems to need more frequency offset for each change in step value.
I tested it on two MPM listed below.

Jumper T16 with Internal module (Jumper)
Value of  "0" causes bind LED to flash intermittently.
Values of +40 and -120 cause bind LED to flicker. (Set to -40 seems to be solid)

Jumper T16 with External IRX4 Plus
Value of  "0" causes bind LED to stay on steady (As noted this MPM seems more accurate)
Value of -128 causes slight flicker and +127 still solid LED

Both tested MPM Modules, using Genuine TR624 RX, bind LED never flashes or goes out at values of -128 or +127. Is there a way to increase the frequency offset per step? ( I think double the current amount would work ) Is there a simple way to change a value in the code i could test?

pascallanger, 2020-02-20:
This is a single byte register. You have already full access to it...
I'm confused since it seems to work fine from what you've written, isn't it?

MRC3742, 2020-02-20:
Yes it works but is difficult to find center as you need to change RF Tune by value of 10 and then watch for a slight flicker to try and get the proper offsets. I am familiar with RF Tune on the other CC2500 Protocols and you can turn the bind LED completely out at a much narrower band of + or - 

If it can't be done then it works fine as is. I just thought there might be an easy way to increase the frequency offset per step.

Thanks for all your time on this - Rich

pascallanger, 2020-02-20:
Nothing I can do. This will stay as it is.

closed 2020-02-20 by pascallanger