pascallanger/DIY-Multiprotocol-TX-Module #329: New Model Support Question

SeanTheITGuy, 2020-02-09:
Curious what the process is to request and/or work on support for new vehicles that are not yet supported.  I have the (extremely cheap) Eachine E013 (Small Pepper) and the controller it comes with is no bueno. Would love to try it on my T16.  I assume it is an NRF24 based device, though I haven't put it under the microscope yet.

I'm an accomplished hardware hacker and passable coder, so if nobody is working on this currently, would love to be pointed to some documentation regarding sniffing the protocol with SDR or however it is done to generate code to add support.

pascallanger, 2020-02-10:
The E013 is based on a LT5910 component which contains a LT8910 RF component.
Only 1 specific mode of the LT8910 out of many many that can be emulated by the NRF.
If you have a SDR try to record one frame and we will see if it's feasible...

SeanTheITGuy, 2020-02-10:
Apparently the equipment I have peaks at 1.7ghz.  Thanks for the response. Keep up the good work. I'll revisit if I get better SDR equipment.

closed 2020-02-10 by SeanTheITGuy