pascallanger/DIY-Multiprotocol-TX-Module #331: feature request: build using platformio

gretel, 2020-02-18:
i'd love to have this project build via https://platformio.org/ - having some experience i'd work on this, if @pascallanger feels it is beneficial :)

pascallanger, 2020-03-21:
The arduino build environment is stable so I'm not considering spending time to use something else for now.

gretel, 2020-03-21:
i was offering my help. thank you

pascallanger, 2020-03-21:
If you want to give it a go then feel free but I won't have the time to help... This is basically what I was saying.

AlessandroAU, 2020-06-16:
@gretel  Did you manage to get it compiling? I much prefer PFio over Arduino IDE nowadays.

gretel, 2020-06-16:
@AlessandroAU nope pascal said it is fine as is.

pascallanger, 2020-06-16:
Guys, I'm fine with whatever you propose if you think the other environment is better then try to get the same integration than the what we have with Arduino.

schwabe, 2020-06-16:
I used platform.io before. it is nicer because you get integration in my more editor and the build process is better than Arduino. E.g. speciyfing variants  with different defines is wayeasier That being said, you still need someone to make a working platform.io build process. I haven't build multi in a long time but iirc we used a patched sdk to be able to control the uart. That would be needed to be replicated in platform.io.

benlye, 2020-06-16:
I can take a look.  I've had a go at getting the firmware to compile in PlatformIO before and I think I know what needs to be done.

AlessandroAU, 2020-06-16:
I agree that PlatformIO is much easier for large projects with a lot of build variants. The editor is also much better suited to working with lots of files (can't even see all the files in the tabs in native Arduino!). 

At the moment I'm using PlatformIO as an editor and then building with arduino-cli, seems to work well enoguh.

benlye, 2020-06-17:
I have a bash shell script that I use to build FW for my modules.  I could have done it in PowerShell, but it's based on the same method Travis CI uses to test the code and create the release files so bash was logical at the time.  For each of my modules / configs it enables and disables the options I want and creates the binary files.  It uses arduino-cli as the compiler.

I can update my source repo and build the firmware for all the modules in a couple of minutes, and I never have to worry about merging changes in _config.h because I haven't actually made any changes there (or anywhere else).

For development work I can use VS Code as the IDE then compile with the script.

It runs very nicely in the Windows Subsystem for Linux (WSL2) on Windows 10 - I build 11 different FW variations in 2m 30s.

If you want to try it you'll need to change these lines:
```
# Path to the Multiprotocol folder , e.g. /tmp/DIY-Multiprotocol-TX-Module/Multiprotocol
SRCPATH=/mnt/c/Users/blye/GitHub/DIY-Multiprotocol-TX-Module/Multiprotocol

# Global ID to set on each module, e.g. 0x12345678
GLOBALID=0x12345678
```

And obviously modify/add/delete the sections which build for each module.

[build_myconfigs.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4791805/build_myconfigs.zip)

Just FYI, getting PlatformIO to work is going to take some time.  We know from experience that we have to be very careful about the compiler and core we use, so it's not as simple as picking an existing board/environment and compiling.  We'll need a completely custom package to make sure everything works exactly the way we need it to.

benlye, 2020-06-20:
I've been doing some work on this.  It wasn't easy, but I managed to get PlatformIO to build the firmware using the same version of the compiler we use in Arduino, and the same Maple core we use.  The firmware binary PlatformIO produced was ~6KB bigger than the one that the Arduino IDE produces.  It's not unusual to see different processes create different sized binaries (there is a couple of KB variance in the Arduino IDE depending on the platform it's run on), but this is extreme.

I want to dig a little deeper to try and understand why this is the case (I suspect it is to do with the pre-processor rather than the compiler) but, based on the level of complexity in building and publishing the packages and frameworks needed to compile the Multi firmware this way, I'm not sure this is worth pursuing at this time.  Maintaining the two completely separate build frameworks would also be a lot more work.

Problems with PlatformIO:
* The version of the compiler we use (2014-q1) isn't available as a ready-made PlatformIO package (the oldest one is 2014-q3) - to be consistent we would have to package and publish our version for all platforms
* We would have to package and publish the STM32F103 core we use
* Even with our compiler and core the STM32 bin file is ~6KB bigger than the one Arduino produces

What I **have** got working is a method to easily compile the firmware using the Arduino CLI in Visual Studio Code.  This only requires adding one or two VS Code config files to the repository.

Another working option is to use the Arduino extension in VSCode.  This method uses the Arduino IDE to compile the project.  It requires a little bit more configuration than the VS Code  + arduino-cli method, but it's pretty straightforward.

Both these methods have the advantage of using the tried-and-tested Arduino build and upload process, and would require very little future maintenance.  The only prerequisite is downloading arduino-cli or Arduino IDE.  The catch is that it only works in VS Code, and not other IDEs.

So, while this isn't PlatformIO, is it enough?

gretel, 2020-06-20:
@benlye did you set the optimization flags on `pio`? personally, i suggest not sticking with ancient toolchain because of portability issues time brings.

benlye, 2020-06-20:
Yes, all the optimization flags are the same, optimizing for size.

The thing that makes us cautious is that in the past there was an upstream toolchain change which caused a bug that made models crash.  The consequences can be bad, so we have to be very careful. 
 What we have today has been in use and tested for a long time, so changing it is not a trivial thing.

gretel, 2020-06-20:
@benlye alright. well sure things break on changes - progression requires effort. regarding `vscode` i'd prefer not to get into politics.

closed 2020-06-20 by benlye

gretel, 2020-06-20:
why do people around here close issues like that?

benlye, 2020-06-20:
Because there's not going to be any more work done on this.  I looked into it and it's not worth the effort/risk at this time.  Arduino works well, and there are options available to use a better IDE and still use the tried-and tested Arduino toolchain to build.

gretel, 2020-06-20:
guess you are right i dont feel like working on this for the second time.