pascallanger/DIY-Multiprotocol-TX-Module #332: SFHSS protocol binding/connection issue with Jumper 4in1 internal MPM

n2o2rox, 2020-02-18:
TX used: Jumper T16 with OpenTX version 2.3.5
Module used: Jumper 4in1 internal multiprotocol module with firmware version 1.3.0.53

I've recently updated to an internal module for my Jumper T16 transmitter. For some reason I cannot get  a solid bind or connection to my Futaba R2006GS receiver. When I try to bind it, it just blinks green. (It should be solid green light; with a solid green light I can make my servos move) It seems bound but could not get a solid green light. No servo response.  I tried the frequency tuning and that didn't work. I went down to 14 and below that the RX LED light blinks red indicating a lost connection. I also went up to 54 and beyond that it will also blink red. Nowhere in between it had a solid green light. Got the median of those values and had it fixed in the frequency tuning but still didn't work. Tried changing to low power mode but that dint help either. With the previous external module I had, it works flawlessly with v1.3.0.53.
I tested further the internal module by binding several receivers of different makes and protocols to see if the RF chip concerned is at fault. I tried several Hitec Optima' receivers and Frsky D8/X8R's. All of them worked flawlessly. Binding was instantaneous. So, the RF chip concerned works. 
I also built/compiled from source (1.3.0.69) for the internal module but had the same problem.
I'm at a loss here.

pascallanger, 2020-02-19:
Strange, the fact that you can tune means that the receiver sees the packets. The sfhss protocol has not been touched for a while. Do you still have the external module and it works with 1.3.0.53?
The firmware is the same for internal and external, no difference....

n2o2rox, 2020-02-19:
It is really strange. Yes, I still have the external module that works for
shfss with 1.3.0.53. Planning on something?

I'll try to do some more testing with it. Probably try to use other models
of Futaba sfhss receivers and compatibles. Worst case is to get me another
internal module. But that route would take a couple of weeks to a month
since i'll be getting it from China. And some things are not working well
with China. Shipments of products are somewhat slow.

pascallanger, 2020-02-19:
Since you can compile your own version, could you try to modify the file SFHSS_cc2500.ino line 236:
#define SFHSS_DATA2_TIMING	1625
You can change the value 1625 between 1600 or 1650, upload and try to see if you have some improvements.
Pascal

T16-Pilot, 2020-02-20:
I got the exact same problem on my Jumper T16 Pro Hall bought from TBS. It came with the internal module. 
The internal module is flashed with V 1.3.0.53, it works just fine with DSM2, DSMX, ASSAN and Hitec. But with S-FHSS the green light is flashing and the servos are stuttering. I also got a brand new Jumper 4 in 1 external module in my Horus X10 that is running 1.3.0.53 and it works just fine with my S-FHSS receivers. I tried the external modul in the T16 and the S-FHSS receivers is working great on the external module in my T16. I tried to roll back to 1.3.0.47 on the internal module. Still same problem...

T16-Pilot, 2020-02-20:
PS I also got the same flashing green LED on a XM+ D16 receiver, and it is having problems arming my quad....

MRC3742, 2020-02-20:
@T16-Pilot have you tried to tune the RF frequency on either of those Protocols on the Internal MPM?

[https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md](url)

Rich

T16-Pilot, 2020-02-20:
@MRC3742 Yes, done that. Did not fix the problem... The funny thing is that the external module with the same version of the firmware is working just fine in my T16 and Horus X10.

n2o2rox, 2020-02-21:
So, I'm not alone here. Does not work for my Futaba R2006GS receiver. Now, an XM+ D16.

However, I got hold of a FrSky Delta 8 to test their sfhss protocol. With this receiver, it binds well and servos respond well. If I get some flying time this weekend, I will do some field range/flight test. I'll see if I can borrow some different makes of Futaba SFHSS receivers and other compatibles and see what works and what doesn't. 

Also, I will do what Pascal had suggested about doing the compile mods. (That is if I still have time for this weekend :) )

pascallanger, 2020-02-21:
The test I've asked for is the most important one since this is the only thing that can be adjusted in this protocol. So try this first and see if you can have a solid bind.
I don't see what kind of information you will get by finding which RX works or not...

T16-Pilot, 2020-02-21:
@pascallanger I am not a programmer, so the code thing i do not know how to do. I am sorry.
What i did last night was re-flashing the internal module twice with 1.3.0.53 and then re-flashing OpenTX 2.3.5. After that i tried tuning (+20) the internal module and now my S-FHSS is working. My XM+ is also working now, but still flashing green light, not solid... i have not flight testet yet. Tuning the XM+ did nothing for me. The green LED was still flashing no matter what i did with the tuning.
First time i tried the S-FHSS was on OpenTX 2.3.3 and it work flawless out off the gate. Since then i upgraded OpenTX to 2.3.5 and the module from 1.3.0.47 to 1.3.0.53. The problems started after that upgrade. I am starting to think that it might be the way OpenTX is talking to the internal modul, because the external modul i running just fine with the same firmware version and OpenTX 2.3.5...?
@n2o2rox The FrSky Delta 8 is a multi protocol receiver. I got a lot of them. They bind to S-FHSS, FrSky D8 and Hitec Minima. Have you tried all off the protocols?

n2o2rox, 2020-02-25:
https://github.com/pascallanger I tried to modify the file SFHSS_cc2500.ino line 236:
#define SFHSS_DATA2_TIMING 1625 and changing the values between 1600  to 1650, (increments of 5) It didn't make any improvements.

https://github.com/T16-Pilot  Frsky  and Hitec protocols on  the Deltas worked.

pascallanger, 2020-02-28:
Frankly unless I have access to the RX which causes issue there is little luck that we will get to the bottom of the issue...
Pascal

richardclli, 2020-03-11:
I do not have any problems on S-FHSS receiver, however I just used some compatible receivers.  For S-FHSS protocol, you need to fine tune the frequency in the receiver.  You need to change the value until it can be binded and try to find the max and min of the tuning value that it will not lost connection and try to set something between min and max for best stability.

I am not sure if you have tried these steps.

n2o2rox, 2020-03-11:
@richardclli for now, my Futaba R2006GS receiver doesnt work with the sfhss protocol with my internal  module.  my other receiver,  the frsky delta 8, works fine. My problem is isolated on the Futaba R2006GS. Tuning and some program tweaks didnt help.

pascallanger, 2020-06-03:
@n2o2rox Unless I can get access to the this Futaba R2006GS receiver I'm not sure what I can do.

ceptimus, 2020-09-08:
I've seen exactly the same issue with a Futaba R2008SB receiver.  Tried 3 different transmitter modules, and the Futaba receiver won't successfully bind to any of them.  I did the fine tuning adjustment using flashing green / red LED, tried low power, etc...
The same transmitter modules will all bind to an X-BOSS AC900 V2 receiver in S-FHSS mode.

My guess is that the Futaba receivers will only bind to some limited range of transmitter IDs... and if we knew what that range was, we could modify the S-FHSS protocol to only use IDs in that range.

Or perhaps the protocol doesn't 100% emulate the output of a genuine Futaba transmitter...  The Futaba receivers are unhappy and won't bind, but the X-BOSS receivers are more tolerant?

I have to give back the Futaba receiver I borrowed today.  If I can borrow it again, I can do some experiments recompiling the source code with different fixed S-FHSS IDs, and see if my theory about only certain IDs being accepted is correct.

ceptimus, 2020-09-09:
My apologies. I now find I was using the wrong binding method for the Futaba R2008SB, and once I swapped to the correct method, the receiver linked properly and worked without any problems.  Please ignore my previous post, and sorry for bumping this old thread.
For anyone else who is as confused as I was, the way you bind a Futaba R2008SB to a multiprotocol module in S-FHSS mode is.
1. Have the transmitter switched on and transmitting normally - you never need to use the 'bind' function on the transmitter.
2. Switch on the receiver.
3. Press the little button in the slot **on the receiver** for two seconds within the first twenty seconds after powering up the receiver.
4.  Wait a few seconds during which time the receiver LEDs will flash different patterns.
5.  Bind process is now complete and the receiver should display a solid green LED, and from then on work normally with the MPM.
6.  There's no need to rebind at each switch on.

n2o2rox, 2020-09-10:
Finally, I got it to work for version 1.3.1.24. Been busy lately so testing was put to the side. Will test later for newer versions.

diego0815, 2020-12-17:
> Finally, I got it to work for version 1.3.1.24. Been busy lately so testing was put to the side. Will test later for newer versions.

Sorry for bumping this thread...

How was it solved? Finger troubles as well?

pascallanger, 2020-12-17:
2 comments above.

diego0815, 2020-12-17:
> 2 comments above.

Also for @n2o2rox in that case, so time to close this issue?

closed 2020-12-17 by pascallanger