pascallanger/DIY-Multiprotocol-TX-Module #333: 5ghz research

kenthinson, 2020-02-21:
This is not a bug report. I hope to collect knowledge here for development. 

I’m curious if anyone is looking into 5ghz aka DJI transmitter and protocols. In addition it’s very possible other companies in the future may adopt 5ghz. So starting research now is prudent I think. 

I’m curious what chips might be good options for creating a 5ghz module as well. Perhaps even a 2.4/5ghz WiFi chip could be reprogrammed to do both. 

Any other interesting thoughts from other devs?

Dropsleutel, 2020-02-29:
Look at this project: Open HD
https://www.rcgroups.com/forums/showthread.php?3154184-Open-HD-DIY-135ms-Latency-DIY-DIGITAL-HD-FPV-Discussion

pascallanger, 2020-03-21:
Not looking into that space for now.

closed 2020-03-21 by pascallanger