pascallanger/DIY-Multiprotocol-TX-Module #334: Can't bind to bugs 6

JodiSabar, 2020-03-11:
I just bought the original t16, i have flash radio firmware into opentx and update the external module to the latest. 
I bind it to my mjx bugs 3 mini and everything is ok except the flight range is reduced significantly from 300-ish m to just around 98m.
I try to bind it to my mjx bugs 6, when i click bind the bugs 6 lights stopped indicating that it is bind just fine but i can't do anything, no arming no flying, nothing. And when i click range i see no rssi, can anybody please guide me so i can enjoy my weekends flying my toy quads again?. 
Any response is much appreciated

MRC3742, 2020-03-12:
I fly a Bugs B6 with that same Radio T16 setup and it works fine.

Are you trying to use the "Bugs Mini" Protocol for the B6 ?
The Bugs B6 uses the "Bugs" Protocol.
It allows for 16 different Receiver No. (I'm using Receiver No. 1)

Rich

JodiSabar, 2020-03-12:
Hi Rich, i did use bugs protocol, can you tell me what version of opentx that you use on the radio?, I've notice that there is a critical update on 2.3.5 ( the one that i use), probably there is something with it.

MRC3742, 2020-03-12:
I'm using OpenTX 2.3.5 also but have updated my Multi Module to 1.3.0.73 (compiling my own)
But I have never had a problem with it using the latest firmware release (1.3.0.53)
There were a few problems with Bugs in earlier Multi Module Firmware. What version is your MPM running?

Rich

JodiSabar, 2020-03-13:
> I'm using OpenTX 2.3.5 also but have updated my Multi Module to 1.3.0.73 (compiling my own)
> But I have never had a problem with it using the latest firmware release (1.3.0.53)
> There were a few problems with Bugs in earlier Multi Module Firmware. What version is your MPM running?
> 
> Rich
> 
> Edit: Channel 5 needs to be setup to arm the Bugs B6 (+100% to arm)
> https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#bugs---41

I am using the latest firmware (1.3.0.53) for the MPM, also have edited the arm switch (channel 5) and other channel accordingly. Should i just delete and setup new model or re-flash both radio and MPM, im so lost here, what do you think?

MRC3742, 2020-03-13:
The binding procedure is a little different on that model I think?
You need to have the Transmitter in bind mode first and then plug the battery into the Bugs.

If that doesn't work your better of posting in the RcGroups Forum for more help.
It seems to be a support problem you're having and not an issue of the Multi Module.
https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module

Rich

pascallanger, 2020-03-13:
I agree, please post your issue on RCGroups where more people will be able to help you with your setup.

closed 2020-03-13 by pascallanger

JodiSabar, 2020-03-13:
> The binding procedure is a little different on that model I think?
> You need to have the Transmitter in bind mode first and then plug the battery into the Bugs.
>
> If that doesn't work your better of [...]

That i never try before, would try it soon after work and let you know.