pascallanger/DIY-Multiprotocol-TX-Module #340: PPM from SpektrumDX6i/DX5e not recognised

jonathanorsinger, 2020-03-21:
Hi,
I bought a jp4in1 module on AliExpress with a STM32. I want to use it with a Spektrum DX6i or DX5e, getting ppm signal from the training port. With both transmitters, I checked the output PPM signal and they look okay with the 6 channels and a frame length of 22ms.
However the signal seems not to be recognised by the module, as the blinking led at 0.5Hz says. I checked the connection on the PCB and everything is ok.
I tried with the precompiled binary for PPM and channel order TAER.
Is there a special jumper have to solder on the PCB? Or a config that must be changed in the code? My board is marked v1.2 (c) M. Blood.
Thanks for your help

pascallanger, 2020-03-21:
Hi,
Like mentionned in troubleshooting section: the https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Troubleshooting.md#input-mode---ppm , you most likely need to insert a 4.7k resistor between the PPM pin and +5V (or +batt).
Pascal

jonathanorsinger, 2020-03-22:
Thanks for your answer,
That didn't solve the problem but I found the solution: on my board there was a diode in series on the ppm signal path pointing outwards (cathode directly connected to the ppm pin in the 5-pin socket) and short-circuiting it was the way to go. I may be wrong but it sounds like the board was designed for a transmitter with pulled-up ppm outpout, whereas Spektrum transmitters may have it pulled-down... or maybe the module was expected to be used in serial mode....
But well doesn't matter since it works now
Still thanks for your quick answer Pascal.
Cheers, Jonathan

closed 2020-03-22 by jonathanorsinger