pascallanger/DIY-Multiprotocol-TX-Module #344: The 1.3.0.76~1.3.0.81 Can't binding with FRSKY V8 protocol receiver(V8FR-II HV)

cqmiao, 2020-04-14:
The 1.3.0.53 is OK，but 1.3.0.76~1.3.0.81 doesn't work~

pascallanger, 2020-04-14:
Are you sure?
I can't find any change in the code that would explain this...
I don't have a V8 receiver so I can't test myself but I'll ask someone to check.
Pascal

MikeBland, 2020-04-14:
Just tested a V8FR HV (not-II), and three V4FR receivers. All bound and worked. I had the "Freq" (option) value set to -25, the value I've found works well most of the time. However, only one of the V4FR receivers bound, I changed the Freq value to -45, then they both bound.

Mike

pascallanger, 2020-04-14:
@MikeBland Could you confirm which version you have been using? I imagine 1.3.0.87.

MikeBland, 2020-04-14:
1.3.0.86, as that was the version I had on the module.

Mike

cqmiao, 2020-04-15:
Thanks,I will try again.

closed 2020-04-16 by pascallanger