pascallanger/DIY-Multiprotocol-TX-Module #345: irangex lite

robthomson, 2020-04-16:
This is more a question than an issue.

On an irangex lite - does anybody know if there is a way to connect a 'bind button' to be used as a hardware option - rather than the serial interface?

Failing that.. is there any way to trigger the bind process to occur - that does not rely the serial connection?

This is for a 'mod' I am planning - essentially turning the irangex as a ppm transmitter module - similar to the old hack modules.

Rob

pascallanger, 2020-04-16:
You can do it but that requires really really small soldering. Why don't you go to an iRangeX+ module which is exactly done for the purpose you are pointing out?
Pascal

robthomson, 2020-04-16:
I need the small pcb form factor.

Hence wanting the irangex lite 

You any idea what pins would be needed to be soldered. 

Assuming a button off pins on the stm?

pascallanger, 2020-04-16:
Just look at the schematic: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Module_Build_yourself_PCB.md#-pcb-stm32-v10t-schematic

closed 2020-04-16 by pascallanger