pascallanger/DIY-Multiprotocol-TX-Module #347: XK X450 motor servos jerky

twistedwings, 2020-04-24:
Protocol XK, subtype X450

Binding works.
Frequency tuning ended up with 40.
RC control works but the motor servos jerk sporadically.
(about every 3 seconds).
The issue is not there with the original Tx.

Taranis X7
OpenTX 2.3.7
iRangeX IRX4 FW v1.3.0.86

nzjott, 2020-07-18:
same promble here with jumper t16 built in 4in1 module.

pascallanger, 2020-07-18:
Without the TX and model I don't see what I can do...

Githubbachelor, 2020-11-12:
I connect with XK450 new protocol best -60 Fine Tuning in Multimodul Jumper with Horus 12s. Everything seems to be OK but the motor servos are jumping in both directions on channel 4 for Ruder? Seems a problem with throttle and Rudder. There must be a mistake in the protocol? This happen alway, when the stick from rudder is at 0% and you want to try left or right direction, when throttle is over 50%. Servos from motors are jumping from one side to the other side uncontrolled. Not stable. In the monitor from TX you cannot see this reaction. Best situation was minimal reducing that phenomen with FInetuning from -40 to -60 - Best results but still there the problem.

pascallanger, 2020-11-12:
> I connect with XK450 new protocol best -60 Fine Tuning in Multimodul Jumper with Horus 12s. Everything seems to be OK but the motor servos are jumping in both directions on channel 4 for Ruder? Seems a problem with throttle and Rudder. There must be a mistake in the protocol? This happen alway, when the stick from rudder is at 0% and you want to try left or right direction, when throttle is over 50%. Servos from motors are jumping from one side to the other side uncontrolled. Not stable. In the monitor from TX you cannot see this reaction. Best situation was minimal reducing that phenomen with FInetuning from -40 to -60 - Best results but still there the problem. All thingswith latest firmware and software.Open Tx

As said previously, without access to the original TX and model there is nothing I can do.
Pascal

pascallanger, 2020-12-04:
Issue solved in release v1.3.1.78.
Pascal

closed 2020-12-04 by pascallanger