pascallanger/DIY-Multiprotocol-TX-Module #350: Support for the feiyue eagle-3 fy-rx02 receiver (red board)

gbraad, 2020-05-07:
I got hold of a feiyue 4channel receiver (fy-rx02),  which is supposed to be used with the feiyue brushless eagle 3 kit. I do not have a transmitter ATM.

I am unable to use it with MPM, but it seems to use bk2451 rfu+mcu

[image]
[image]

gbraad, 2020-05-07:
It should should work with the FY-TX01 transmitter from the FY03 Eagle3

pascallanger, 2020-05-07:
Without a TX and RX in my hands there is nothing I can do...
Pascal

gbraad, 2020-05-08:
A TX is on it's way. What can I do to sniff the signals? I have access to a logic analyzer, but since this is a RF+MCU I think the 'magic' happens inside the 8051 core. Would it be possible to listen in on the signals using a nRF24L01 for instance?

pascallanger, 2020-05-08:
You can use try to follow this article: http://travisgoodspeed.blogspot.com/2011/02/promiscuity-is-nrf24l01s-duty.html

gbraad, 2020-05-13:
Transmitter arrived yesterday and works like a treat, and got some additional receivers just in case I fry one (at is seems some can't handle a BEC with 6V as Tamiya ESC do). Anyway, will try to see what i can find over the weekend. Any code that I can run from an Arduino or STM32 would be appreciate as I haven't used the nRF24's low-level that much yet (only did a makecode extension to support them and send some packets back-and-forth).

pascallanger, 2020-05-13:
You can always start from the XN297Dump protocol.

pascallanger, 2020-06-03:
Any progress? Should I keep this issue open?

gbraad, 2020-06-03:
haven't found the time to look into this yet. all the equipment and
tools I have, and assistance from Goebish, ... just some distractions
aren't easy to control.

pascallanger, 2021-01-19:
No progress closing this issue...

knoopx, 2021-01-19:
Hey @pascallanger I'm actually revisiting this, I've got the v1 board and I'm trying to sniff packets using an nrf24. So far I'm struggling with the setup. Anyway if i make some progress I'll let you know.

gbraad, 2021-01-20:
@knoopx due to the situation I still got the radio here, so if you can share what you have tried it might be possible help can be offered.

knoopx, 2021-01-20:
@gbraad I've got a logitech unifying dongle and flashed https://github.com/marcnewlin/presentation-clickers firmware. So far I have not been able to identify the TX packets as I did not figure out the rate/channels yet.

benlye, 2021-01-20:
> @gbraad I've got a logitech unifying dongle and flashed https://github.com/marcnewlin/presentation-clickers firmware. So far I have not been able to identify the TX packets as I did not figure out the rate/channels yet.

Following.  I have a few Logitech unifying receivers laying around and never knew this was something that could be done with them.

pascallanger, 2021-01-20:
> @gbraad I've got a logitech unifying dongle and flashed https://github.com/marcnewlin/presentation-clickers firmware. So far I have not been able to identify the TX packets as I did not figure out the rate/channels yet.

To find at least one channel, use the multi scanner. You should see a couple of spikes. Take the value of one of them.

pascallanger, 2021-02-08:
Any progress? may be condiser sending me the hardware...

gbraad, 2021-02-09:
> Any progress? may be condiser sending me the hardware...

That was the idea. Wanted to give Goebish, but the pandemic prevented
me from traveling to Europe.

knoopx, 2021-03-19:
[image]

[image]

pascallanger, 2021-03-20:
Have you decoded one packet yet?

knoopx, 2021-03-20:
Not yet, im still learning how this whole thing works. So far I'm using universal radio hacker with a known protocol. I've been able to capture and demodulate but i have no idea how to make sense of the bitstream. I need to learn about how to decode the bit stream in order to match actual packet bytes. I guess each transceiver has its own way of encoding and modulating data and needs to be figured out? What are your tools of trade?

pascallanger, 2021-04-09:
> What are your tools of trade?

I'm using MATLAB to receive and decode the signal at a really low level by looking at the phase. Changes in the phase indicates the change from 1<->0, time between two changes gives the bitrate, then using these phase changes I decode the bit stream and rebuild in a table what could be the message. The table contains all variations MSB first, LSB first, inverted and not inverted as well as 7 bit shifts. From all these variations you need to find out which one contains most probably the message. Some stuff guide you through that and usually with 2/3 captures you have a pretty good idea.

If you do a capture on one of the RF freq using GNU radio I can easily look at it and give you the end table.

closed 2021-09-25 by pascallanger