pascallanger/DIY-Multiprotocol-TX-Module #351: Flysky RSSI

pan-sattan, 2020-05-09:
hi,
on my IRX4 with fw 1.3.0.91 is not possible to see any sensors from flysky X6b reciever... 

am i doing wrong something?? 

[image]
[image]
[image]

pascallanger, 2020-05-09:
Too close from the receiver may be...

pan-sattan, 2020-05-09:
now i tried through some walls from oposite rooms and still the same...

pascallanger, 2020-05-09:
Is it the only AFHDS2A RX that you have?
The only solution would be to go in debug mode and see if the module receives anything from the RX.
Since you have an external iRangeX debug is quite simple, I suggest that you learn how to compile a debug firmware  and launch a debug session. From there, I can guide on some firmware modifications to see what might happen.
Links:
- https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#option-3---compiling-and-updating-firmware
- https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Debug.md

MRC3742, 2020-05-09:
Observation on above pics.
No Multi Sub Protocol is shown for FSky 2A?
Possible version of OpenTX mismatch?

pan-sattan, 2020-05-09:
i have the latest fw for my taranis from companion software

later i will try debug mode, i have other recievers from fkysky, no one works... it shows me the rssi with flysky i6 radio, so i am sure, that its working

pan-sattan, 2020-05-09:
[image]

pascallanger, 2020-05-09:
Really strange this missing line...
Could you try one of the latest 2.3.8 nightly to see if you have the same issue?

pan-sattan, 2020-05-09:
maybe... which firmware should i use? radio: opentx or ersky? and inverted or not? 
[image]

pascallanger, 2020-05-09:
You are running OpenTX so OpenTX. For the inversion it should be inverted since it's an external module but the radio tells the module.
The problem of the line not showing up is not the module firmware but most likely the radio.
Can you test OpenTX 2.3.8 latet nightly?

MRC3742, 2020-05-09:
Going with changelogs latest stable version to run on OTX v2.37 would be MPM v1.3.0.87

Starting with version 1.3.0.91 "Protocols appearing in alphabetic order, only available in the latest OpenTX 2.3.8 nightlies (not published yet)" Maybe this is causing the problem?

Using the simulator in 2.3.8 nightlies adds a line "Type" below the "Mode" 
[image]

Link to OTX Nightlies is https://downloads.open-tx.org/2.3/nightlies/companion/ 
Then pick your OS, Install Companion nightly, Redownload firmware and flash to TX.

pascallanger, 2020-05-09:
@MRC3742 Thanks for helping out ;-)
1.3.0.91 does not need to have 2.3.8, it's only required if you want the protocols in alpha order. But I'm fine to be proven wrong.

pan-sattan, 2020-05-09:
now it works, looks like that was inverted/noninverted version problem...
thank you for advices, i very appreciate it

closed 2020-05-09 by pan-sattan

pascallanger, 2020-05-09:
It's not the inversion but the fact that you have selected erskyTX instead of OpenTX...
Pascal

pan-sattan, 2020-05-09:
yes, i choose wrong fw version, ersky

https://downloads.multi-module.org