pascallanger/DIY-Multiprotocol-TX-Module #353: Module unresponsive 50% of the time

glitch666, 2020-05-19:
I have a mpm vantac lite on my xlite s pro. It is firmware 1.3.0.95 and my opentx is 2.3.7. when I turn my tx on about 50% of the time the module has a solid red and green light and will not bind, also doesn't show the firmware version in the model menu. I have to power off and on my tx multiple times before it will have a solid green light and bind. Or I can put the module back off and on multiple times and then it will work. I've tried different fw versions on the module and it does the same thing.

pascallanger, 2020-05-19:
There is a known bug on this Vantac(=FrSky) module. The problem is the bootloader they are using. You need to upgrade it: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/3594850/Vantac.MPM.Lite.Red.LED.Issue.docx

closed 2020-05-19 by pascallanger

glitch666, 2020-05-19:
Perfect 👍 thanks kindly.