pascallanger/DIY-Multiprotocol-TX-Module #354: Spektrum Protocol temporary signal missing

FPVxfun, 2020-05-19:
Dear friend, I use your multiprotocol firmware both on an internal Jumper T16 and on an external 4IN1 iRangeX. I've tried with many different receivers (Spektrum, LemonRX, OrangeRX) with DSMX 22ms option. Actual fw is 1.3.0.91 but the issue happened with other previuos firmwares. Usually, even in near field (less than 50m between tx and rx), the signal is missing and the receiver manage by a "hold" in the servo movement. You could easily verify because you can't control the airplane for a couple of seconds. I had no issues up to last year.
Could you please verify Spektrum protocol?

pascallanger, 2020-05-19:
No one is complaining about DSM despite many users using it. Nothing at all on rcgroups apart from people saying that it works and sharing their experiences.
Are you sure you are really getting sevo freezes? Are you not just listening to the radio warnings about critical signals or even telemetry lost?
Have you verified your antennas? Some modules have sma and others rpsma, it's easy to mount the wrong antenna and having no range as a result.
So basically I don't know what it could be but it seems to affect only you...
Pascal

FPVxfun, 2020-05-20:
Dear Pascal,

My telemetry is switched off.

Servo freeze is real.

The module iRangeX 4 in 1 worked well with the original fw (I don't know
what version).

The hardware is the same, same radio, same model.

The antenna is the same and it is correct.

If you think could be some other, please let me know, I'm an electronic
engineer.

Thanks in advance

pascallanger, 2020-05-20:
No idea then... You are the only one to experience this issue...You can go on GitHub and go in the past to find a working version of the firmware, from there we can look at the difference.

benlye, 2020-05-20:
Could it be the radio if it is still running JumperTX?  I can't recall if the issue with JumperTX and DSM only affected the internal module or if external modules were affected as well.

@FPVxfun what firmware is on your radio?

FPVxfun, 2020-05-24:
OpenTx 2.3.7

FPVxfun, 2020-05-26:
OpenTx 2.3.7

FPVxfun, 2020-05-26:
Look at this...I'm not the only one:

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/293

pascallanger, 2020-05-26:
The issue has been solved for everyone excpet Rickster250 who is just pilling stuff that have nothing to see together. He is also stating things that are none sense like v13.0.76 works and not v1.3.0.91 where the DSM code has not been changed at all. He came to RCG forum and has been told a couple of things to solve his issues and never came back.

Your issue is completly different since it does affect both the T16 internal and external modules on a large number of RXs. I still think there is something on your setup. 50m range is not normal and shows that you either have checked LOW POWER or you have faulty modules. Again you are the only one to report issues on DSM.

pascallanger, 2020-05-26:
Please post on RCG on the T16 thread or Multi thread. Yoiu'll see that many people are using DSM. There has been plenty of range testing from a number of people without any issues.

pascallanger, 2020-05-26:
You are still complaining instead of looking for a fix.
You have been telling that on old FW is working fine with your setup. I've asked you 6 days ago to find which firmware is working for you: "You can go on GitHub and go in the past to find a working version of the firmware, from there we can look at the difference"
But all we receive each time is a single line of text...

pascallanger, 2020-05-26:
Going back to your issue something came back to my mind.
You said you tried Spektrum, LemonRX, OrangeRX with DSMX 22ms.
Can you give a full list of all the RX types you tried along with a link to the manuals?
DSMX 22ms sounds really strange to me, it would mean that none of the RXs you've tried have more than 7 channels and none of them have telemetry. Is that really possible?
Where does this DSMX22ms is coming from?
The radio/module should auto detect the RX type, is this the case or you've manually forced it?
How many channels have you delcared for the different RXs?
If they are real all DSMX 22, can you put the module in debug mode and give me the bind answer for each of them?

pascallanger, 2020-06-03:
@FPVxfun any answers?

FPVxfun, 2020-06-03:
The 1st plane is an Horizon Hobby Carbon CUB Safe+ BNF (bind and fly).
It is equipped with a controller code SPMA3232 by factory:

https://spektrumrc.com/Products/Default.aspx?ProdID=SPMA3232
and a satellite (serial receiver) code SPMA4647.

However I couldn't find any Instruction Manual on the web. It seems they're specific items for this plane.

The 2nd plane is equipped by a LemonRX 6ch diversity:

https://www.lemon-rx.com/index.php?route=product/product&path=72&product_id=132

The 3rd plane is equipped by a LemonRX 7ch stabilizer + telemetry module + satellite:

https://www.lemon-rx.com/index.php?route=product/product&path=76&product_id=131
https://www.lemon-rx.com/index.php?route=product/product&path=82&product_id=124
https://www.lemon-rx.com/index.php?route=product/product&path=72&product_id=135

Documentation could be found there:
https://www.rcgroups.com/forums/showthread.php?2916819-Official-Lemon-receiver-comparison-chart

https://www.rcgroups.com/forums/showthread.php?2164243-The-official-Lemon-stabilizer-receiver-thread-NEW-USERS-read-first-8-posts-)

https://www.rcgroups.com/forums/showthread.php?2266873-The-official-Lemon-DSM2-DSMX-Compatible-Telemetry-system-thread

pascallanger, 2020-06-03:
What about all the other questions?
It's obvious that some of your RXs are not dsmx/22. Selecting the wrong options for DSM is known to possibility get the RXs to freak out as pointed in the documentation. Have you let the system to auto detect everything? Was the auto detect working?
And all the other questions I've asked...

FPVxfun, 2020-06-03:
Unfortunately I had not enough time to test your last version with DSMX
autodetect. Maybe DSMX 22ms is not correct however I was able to bind. I
had loaded into a Multiprotocol module (which I'm sure it was working well,
coupled with same receiver and same plane) latest firmware of 1.2.1.85 (the
latest of 1.2 series). During the weekend I hope to have enough time to
test it. I will also try with your latest release with DSMX protocol
autodetect

pascallanger, 2020-06-03:
Bind will always work, it's independent of what's being selected. Auto detect is not a new feature... You should let the RX tell what it wants through the auto detect feature.

FPVxfun, 2020-06-07:
Yesterday tests:

SPMA3232 only works with DSMX 22ms. Binding with DSMX 11ms there are no output signals.

With version 1.2.1.85 I was not able to bind SPMA3232 receiver

closed 2020-07-15 by pascallanger