pascallanger/DIY-Multiprotocol-TX-Module #357: Any plans for FASST?

azaitsev, 2020-05-23:
Hi guys, awesome project. I'm surprised you haven't yet implemented FASST support. Have you any feature plans for that?

pascallanger, 2020-05-23:
If you are talking about Futaba FASST, I don't think it can be supported since they are using a special RF chip.
But I can have a look to see if it's doable to emulate it from another component. For these I would need someone to ship me a TX and RX. If you are willing to do it then say so otherwise I can only close your request.

azaitsev, 2020-05-23:
I have only one radio so I can't ship it. I can make photos of components inside or record an rf sample with hackrf.

pascallanger, 2020-05-24:
That won't help. Tons of data is needed to reverse a protocol. So I guess this is a dead end at least for now.

closed 2020-05-24 by pascallanger