pascallanger/DIY-Multiprotocol-TX-Module #358: FrSkyX2 protocol binding/connection issue with Jumper 4in1 internal MPM

redspecial, 2020-05-25:
Radio: Jumper T16 Pro Hall V2
I´ve got the same issue as it is still open with "SFHSS protocol binding/connection issue with Jumper 4in1 internal MPM" but I got it with FrSkyX2 EU LBT when trying to bind FrSky RXs (EU LBT firmware). When I bind a FrSky RX8R Pro with the internal Jumper 4in1 the receiver tells that it is bind successfully. When I reconnect only red led is blinking and no servo movement can be done. I have v1.3.1.1 on the internal 4in1 mpm and the latest nightly build of open tx 2.3.8 n220. Also tried it with the latest open tx stable release, same issue. 
I still got a taranis X9E and binding there with the same receivers works fine.
All the best
Jens

pascallanger, 2020-05-26:
"but I got it with FrSkyX2 EU LBT when trying to bind FrSky RXs"
You have added a "s" at RX, have you tried another RX?
Are you sure you are using V2.1.0? FrSky ACCST V2.0 and V2.1 are using the same bind but the normal packets are different.
Have you done the frequency tuning?
Pascal

redspecial, 2020-05-26:
Hi Pascal, 
I´ve tried it with two receivers, both RX8R Pro. I can tonight try it with another one, a RX4R. 
I´ll double check if it´s 2.1.0, but i recently updated it with the latest available ACCST firmware from FrSky. 
I´ve tried binding with different freq. tunes in +-20 steps. No difference. 
Jens

pascallanger, 2020-05-26:
They are still tweaking the RXs code: https://github.com/FrSkyRC/Firmware-Test/tree/master/ACCST%20firmwares%20v2.1.0
The RX8R Pro is there since some time but they have released a new code for the RX4R yesterday...

HazzaHFPV, 2020-05-26:
What I noticed with FrSkyX2 is, that I need to be further away from the receiver to get a solid link (green LED) compared to the previous version FrSkyD16 protocol.

I am using an XM+ with the internal T16 module. Binding works fine with both protocols, need to be about 0.5-1 meter away for solid link using FrSkyD16, 1-2 meters away for FrSkyX2. Any closer and I lose the link.
I regard this as being normal, but when testing on the bench it might lead to confusion.

redspecial, 2020-05-26:
Seems strange. I´ve now relfashed the RX8R-Pro with the new downloaded firmware V2.1.0 and it works. It still works on the taranis, too without reflashing the taranis tx module. The 2nd RX8R-Pro works without reflashing on the taranis and after reflash also with the T16 MPM. So I think the issue has been caused by the receivers, misflashing or sth.. So the issue can be closed, thank you!
I´ve just read your comment HazzaHFPV and yes it´s true, today I experienced the same with the 2nd RX8R-Pro. I was too close to the receiver after binding. But that initially was not the issue.

closed 2020-05-26 by pascallanger