pascallanger/DIY-Multiprotocol-TX-Module #360: XM Plus wil not bind to my Jumper T16 Pro or QX7S

rcflyerf15, 2020-05-31:
Hey guy's, Karl here. I'm having troubles trying to get an xm+ receiver to bind to my jumper t16. I have never had to much problems with these receivers, and they are in all of my quads. I tried messing with the frequency tune, and reflashing FCC firmware, but no dice. I even tried to bind it up using my QX7, but to no avail. Never had this much problems with a frsky xm-xm+ receiver before. looking for some advice or thoughts as to what might be going on. I had bought the Geelang Lightening 120X, which is what i want to bind it to. It came with the AC900 Frsky version, but i could not get it to bind either on the jumper t16, thats why i had bought the xm+, i just new it would bind, not, lol. Any help would be greatly appreciated. Thank you

ggmaniack, 2020-06-01:
Make sure you're not flashing the v2.1.0 ACCST firmware to your receivers. If you want to use that, you'll have to use the recently added FRSKYX2 mode.

pascallanger, 2020-06-03:
Could you use the RCG forum instead since I don't think this is a code issue but usage?
Thanks, Pascal

closed 2020-06-03 by pascallanger