pascallanger/DIY-Multiprotocol-TX-Module #361: Support older Pelikan Protocol

z3r0c00l2005, 2020-06-02:
Would it be possible to try supporting the version of Pelik

an protocol used on the sling RTF model (Possibly V2 but unable to confirm)  I've attached a Sigrok trace I think is the TX in bind mode
[TX_in_Bind.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4716408/TX_in_Bind.zip)

(May need to change the extension to .SR for PulseView to load it

z3r0c00l2005, 2020-06-02:
Should add that the Chip used is an A7015.  Supplied Transmitter is a MT-180H (FCC ID XJ6MT-180H) branded as a Pelikan Cadet 4 Lite

pascallanger, 2020-06-02:
The dump looks perfect. I can see that the start of the bind packet is the same but that's it.
The bind length is the same as a normal packet 15 bytes where the current code sends only 10 bytes.
15 -> fixed => common
04 -> version => common
FF FF DA 06 -> ID => common
52 63 AC 00 01 00 60 18 -> does not match
D1 -> checksum => common

Please do a dump in normal mode now.

Pascal

z3r0c00l2005, 2020-06-02:
[TX_On_RX_Off.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4717669/TX_On_RX_Off.zip)
[TX_Bound_Stick_Neutral.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4717670/TX_Bound_Stick_Neutral.zip)

As file filenames suggest that should be one dump with the Tx in normal mode and Rx off, and another with the Rx connected

pascallanger, 2020-06-02:
There is no information from RX to TX so we don't care if the RX is on or off.
As said in my previous post, the bind is missing.

z3r0c00l2005, 2020-06-02:
Lets try these....should be on with the TX switched on and attempting to bind and another powering up without requesting bind mode
[TX_Switch_On_In_Bind_Mode.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4717774/TX_Switch_On_In_Bind_Mode.zip)
[TX_Turned_On_No_Bind_Mode.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4717776/TX_Turned_On_No_Bind_Mode.zip)

pascallanger, 2020-06-02:
Thanks let me work on it from there.

z3r0c00l2005, 2020-06-02:
Great, thanks Pascal

pascallanger, 2020-06-02:
Can you try the version 1.3.1.8 available in the master branch?
This version is using your ID and frequencies so you should not have to rebind the RX. You can also try to bind the RX which should work as well.
The next step will be to try different IDs and frequencies but first thing try what has been posted.

pascallanger, 2020-06-02:
Ok can you try in the Pelikan ino file to comment the line #define PELIKAN_LITE_FORCE_ID (add //)?

z3r0c00l2005, 2020-06-02:
With the FORCE ID line commented out, the receiver feels harder to bind; although i can get it to bind after a little longer.  Servo movement is then really sluggish, almost a seconds delay between input and servo movement.  Un-commenting the FORCE ID and the servos are back to smooth movement

pascallanger, 2020-06-02:
Ok it means that the hopping freq algo is different... Let's try tomorrow to see if we can change the id to have more than one.

z3r0c00l2005, 2020-06-03:
using v 1.1.3.9 i can bind on both Rx ID 0 and 41, servo movement seems smooth now as well.

pascallanger, 2020-06-03:
RX ID is RX number, correct?
You haven't touched any of the defines, correct?

z3r0c00l2005, 2020-06-03:
Yes, RX Number defined on the TX during binding.

Nothing touched; just pulled down the latest copy of master and compiled it :)

pascallanger, 2020-06-03:
And you had to rebind at each change of RX number correct?
If yes then I call it done.

z3r0c00l2005, 2020-06-03:
Correct. Many thanks for all your help with this; if I notice anything strange i can raise a new issue :)

closed 2020-06-03 by pascallanger