pascallanger/DIY-Multiprotocol-TX-Module #367: Support for Microzone reciver

ss8688, 2020-06-11:
Microzone products sell well in here China,and they are cheap.It would be great if DIY-Multiprotocol-TX-Module can support  Microzone reciver. They may have 3  sub Protocol.

pascallanger, 2020-06-11:
Which RF component is in use?
Can you dump the SPI bus on the TX?

ss8688, 2020-06-13:
It use A7105

ss8688, 2020-06-13:
but what is ‘dump the SPI bus on the TX’？Do you mean Hardware Hacking？

ss8688, 2020-06-13:
@pascallanger I don't have one.but Iplan to buy one

pascallanger, 2020-06-13:
@ss8688 You can buy a cheap 8CH analyzer and connect on the A7105 SPI bus.
FYI, this is a long process to reverse a protocol with many trials. Before investing the time, I need to make sure that you are ready to do so.

ss8688, 2020-06-14:
@pascallanger [https://item.taobao.com/item.htm?spm=a230r.1.14.1.55c97f6aHAlACA&id=620031032239&ns=1&abbucket=12#detail](url)  Is this one work？

ss8688, 2020-06-14:
But they have 6 and 10 ch TX ，I only have a 8ch one，6 and 10 ch reciver not work with 8ch tx，They may use different protocol.

pascallanger, 2020-07-15:
@ss8688 Yes that's the logic analyzer I was refering too.

ss8688, 2020-07-18:
I find it really hard to weld the line on the pin，I should improve my welding skill。so，I will try it later

pascallanger, 2021-01-19:
No progress closing this issue...

closed 2021-01-19 by pascallanger