pascallanger/DIY-Multiprotocol-TX-Module #368: Wiki entry about fine tuning CC2500 protcols

fiechr, 2020-06-12:
I've done the tuning and everything worked fine. But there is one sentence that I don't understand:

> Once the Freq value is known it should be applied to all other models which use this protocol and, if they were previously bound, the receivers must be re-bound.

Does "this protocol" refer to all protocols mentioned at the top of the page? Or can different protocols using the same IC actually have different values for Freq?
If not, I recommend changing this to "the protocols mentioned above" instead of "this protocol".

The second thing I don't understand: Do I need to rebind after changing the Freq value using the transmitter settings? Or does this only refer to changing the value in the source?

HazzaHFPV, 2020-07-05:
Each protocol should be separately tuned and can have different fine-tuning values. For the process of fine-tuning the receiver has to be already bound and therefore does not have to be rebound once the correct value is entered (at least in my experience).

pascallanger, 2020-07-15:
I've updated the text to hopefully clear the questions.

closed 2020-07-15 by pascallanger