pascallanger/DIY-Multiprotocol-TX-Module #369: TM5 module on Taranis X9-lite Pro - working ok, but cannot flash from the radio.

Tygrys-1, 2020-06-13:
I know, that TM5 has some bootloader problem, but I find it working ok on Taranis X9 Lite Pro. I can also update MPM firmware with USB using the flash-multi with no problem. However I found two strange things:
 - The flash-multi cannot report the version of the software.
 - I cannot update the firmware from the radio, I got A No Sync error. It is told, than I have to flash it using the flash-multi once, which I have done many times before. And then it should be able to flash from the radio.

What am I missing?

kilrah, 2020-06-13:
You need to flash with an ftdi adapter to update the bootloader.

benlye, 2020-06-13:
It's also possible that the module also has the incorrect logic gate, stopping flashing from the TX working. We've seen that on one other TMX5 module.

Tygrys-1, 2020-06-13:
Thank you very much, @kilrah.
The second method worked smoothly and now I can flash from the radio.

Still, the "read module" from the flash-multi output like this:

Reading from MULTI-Module via native USB
Switching MULTI-Module into DFU mode ... done
Reading flash memory ... done

Firmware signature not found; extended information is not available. This is expected for modules with firmware prior to v1.2.1.79.

EEPROM Global ID:         Not found

MULTI-Module read successfully

-------------
As my firmware is now 1.3.1.9 - shouldn't it report something?

Also - on the Taranis version information screen I got 

Ext module
Module    No info

Is it as it should be?

benlye, 2020-06-13:
Your module may have the wrong STM32 part, which could stop Flash Multi from reading all of it. That is also a known issue with the TMX5. If it works in the radio then just don't worry about it.

Tygrys-1, 2020-06-13:
It is possible, because when I tried to backup the firmware from the flash-multi, the file size was 'bout 56k. Does this limit affects the functions of the MPM ?

Also - Should MPM report itself on taranis version info screen?

Tygrys-1, 2020-06-13:
@benlye Btw - great tools you have made, thank you for making live easier.

benlye, 2020-06-13:
It's an interesting problem.  The STM32F103C8, which I'm assuming your module has, is specified to have 64KB of flash, but it's well known that they (typically) really have 128KB.  

If your module really only had 64KB of flash it would not work at all so it seems like the tool Flash Multi is using to read the module is reading to 'spec' rather than what is actually there.  I have an F103C8 here somewhere that I'm going to do some testing with.

I can't say what the Taranis should tell you on the info screen - I don't have one.  You should see the Multi module FW version when you are selecting the protocol in model setup though.

Tygrys-1, 2020-06-13:
Don't know what chip is on my module. But I think you are right here.
It looks like Chineese manufacturers uses less costing versions, which were known to have more flash in real then in specs. The same applies ie. to BTT SKR mini E3 v1.2, where they used STM32F103RCT6, and flash it up to 512K ;-). 

Thank you for the support.

closed 2020-06-13 by Tygrys-1

kilrah, 2020-06-13:
> Also - Should MPM report itself on taranis version info screen?

No, only on the model setup page where you select protocols.