pascallanger/DIY-Multiprotocol-TX-Module #370: Support for E32-868t30d as separate 900mhz board

annoyinganime, 2020-06-13:
There are E32-868t30d modules, which contain sx1276, 1w amplifier and STM8L15x onboard - so everything what's needed to be a standalone 900mhz board. Many of us, including me, have transmitters with built-in 4in1 modules, and adding one external board for more protocols will be an amazing option to have. How hard it will be to port multiprotocol firmware with 915mhz-specific protocols onto STM8L15x?

pascallanger, 2020-06-13:
I've seen your post on the other issue, no need to create a dedicated entry. I don't have any time to look at it this is why I did not responded and let someone else who might want to do so..

closed 2020-06-13 by pascallanger

annoyinganime, 2020-06-13:
Sorry for bothering, I've just deleted it there. Il just copy it back