pascallanger/DIY-Multiprotocol-TX-Module #374: Spektrum SRXL2 too many channel issue

MattCall1995, 2020-06-16:
Hi,
I'm currently using a SPM4650 receiver. This one uses the SRXL2 protocol between the RX and FC
and uses DSMX between the RX and my TX16s (Multiprotocol v1.3.1.9 & opentx 2.3.9). 
When I bind in auto mode it detects 72 channels. This seems weird to me.
Coupled to that I get constant telemetry lost/recovered messages. All the discovered sensors show up in red. 
Does anybody have the same issue? Or found a work around?
Other DSMX receivers don't have any issues.

Thanks

kilrah, 2020-06-16:
Build from source, I believe it was fixed a few days ago.

pascallanger, 2020-06-16:
The number of channels showing 72 is corrected in the upcoming version of multi and already available on master of you compile by yourself. But it shouldn't affect anything. This is due to the RX sending an unknown flag. May be you know what this TX has special compare to others?

For the telemetry, I've never seen this SRXL2... I'm wondering if this is a multi issue (receiving the telemetry) or OpenTX issue (decoding telemetry).
What happens if the TV is not connected to the receiver? Is the telemetry stable?
Do you have an external module so we can do easily some debug on your hardware?

MattCall1995, 2020-06-16:
As far as I know this kind of RX is pretty new territory for spektrum. But that's it.
When binded the telemetry seems unstable and mostly incorrect. It kind of looks like it's refreshing. It twitches between red and black. 

> What happens if the TV is not connected to the receiver? Is the telemetry stable?
Do you mean when the FC isn't connected? That I tried but without succes.

I currently don't have an external module or a spare SPM4650.

pascallanger, 2020-06-16:
Yes I mean try the RX disconnected from the FC (sorry typing on my mobile).
I think you've said you tried but what's happening exactly in this scenario? What are you getting on the telem page? May be a video would help.

MattCall1995, 2020-06-16:
I just desoldered the RX from the FC. Telemetry seems stable.
Apparently I only checked if I got the same issue with the 72 channels last time I tried it. 
https://we.tl/t-XSLyAo7k3a
Added a video of situation when the RX is soldered and a picture when desoldered.
Had to send it with a wetransfer

SilkierLyric146, 2020-07-05:
Hi i want to report a similar issues with the same receiver and protocol: i have a FC wired with a SPM4650 that use the SRXL2  between the RX and FC and uses DSMX between the RX and my TX Jumper T18 Pro ,  i have encountered a lot of low signal issues, the RSSI signal is pretty low when i fly compared with my standard DX6 and i can't fly with my drone without have the fear to enter in a failsafe(which is activated when the fc detects that the RSSI signal is 20% lower for more than 4 seconds) which has happened more than once so far.
I have checked that the low power mode was disable and i have tried a lot of range test, my tx firmware is Multiprotocol v1.3.1.9 & opentx 2.3.8.

Also i have tried with different dsmx receiver that don't use SRXL2 but only DSMX and DSM2 and i have found that the low signal were a issues in all of them.
to specify i have tried:
1 x SPM4650 with SRXL2 -> low RSSI signal and failsafe problem
1 x Orignal Spektrum receiver mount on a Horizon hobby Inductrix -> the TX continued to report a low RSSI signal but i didn't have any real failsafe or total loss of drone controls.
2 x not original DSMX receiver but DSMX/DSM2 Compatible receiver (orangeRX) -> very low signal/RSSI i couldn't fly more than 10 meters without losing all the control of the plane 

i want to know if somebody can help or if i have a faulty module (i'am looking to try a fry sky receiver), thanks
ps: (my english is not very good so sorry is there are some error)

kilrah, 2020-07-05:
That receiver has "Fly-By Telemetry" so the RSSI (which when using multi is the strength of the telemetry signal and NOT the strength of the uplink) is useless, disable alarms.

HazzaHFPV, 2020-07-05:
If it happens with other receivers (other protocols) as well one thing I would check is if the u.fl connector of the internal 2.4Ghz antenna might have popped off or is not seated correctly. If you want to check be careful when removing the back cover of the radio as I think the 2.4ghz antenna is internally attached to the back.

SilkierLyric146, 2020-07-05:
Thank you all for the answers, so what about RSSI if you say it is a useless value i will ignore it but then what should I set in the fc of my drone as a parameter for the signal strength?  I would like to avoid flying without a failsafe parameter of course.  also I tried to disassemble the radio and I checked the antenna connector and it is well fixed both in the tx module and the welding in the antenna.
I will try with other receivers and other protocols and I will let you know, thanks again to all

pascallanger, 2020-07-05:
> Thank you all for the answers, so what about RSSI if you say it is a useless value i will ignore it but then what should I set in the fc of my drone as a parameter for the signal strength? I would [...]

OpenTX displays the TX_RSSI which is how well the radio receives the signal from the RX. It is NOT how well the RX receives radio. Your FC does not see this TX_RSSI. I don't know your FC should be configured but it cannot use this value anyway.

kilrah, 2020-07-06:
>  what should I set in the fc of my drone as a parameter for the signal strength? 

Nothing, there is just no info it can make use of.

Failsafe should not be linked to RSSI in the first place, just to signal loss. Many mechanisms for that on most FCs.

SilkierLyric146, 2020-07-07:
ok i understand, thank everyone

Dosh22, 2021-04-03:
Hi, all, i have problem with SRXL2 telemetry also. I have TX16S with OpenTX 2.3.11 and the multimodule with newest firmware (multi-stm-serial-aetr-v1.3.2.61). I have the SPM4650 reciever instaled in my Quad with Betaflight (4.2.8). 

I can control the quad without issues (no failsafes, reciever works fine).... But the telemetry doesn't work... I get this telemetry output on my TX16S:

[image]

Do you have idea, what cause this?

GregZuber, 2021-04-25:
I have also problems with SPM4650 (Carbon Cub S2). When binding for the first time (using the bind plug with the plane), it works well. I have telemetry data. But after I disconnect the plane battery, the blue diode flashes on the plane and the plane doesn't initiate (it doesn't react to any inputs from the transmitter). The only solution is to duplicate the model and rebind it again.

Dosh22, 2022-02-10:
Hi @OpenUAS, sorry for late answer.... Now I had MM v.1.3.3.7 on my radio and Edge TX 2.6 and the telemetry for Spectrum SRXL2 (SPM4650 reciever) looks the same as in pictures...