pascallanger/DIY-Multiprotocol-TX-Module #377: Frsky R9 868MHz loosing connection

mszulen, 2020-06-30:
Latest FLEX firmware in R9MM receiver and latest multimodule firmware in T18
915MHz works ok, but when switching over to 868MHz, connection drops every few secconds (red led start blinking) .

pascallanger, 2020-06-30:
Yes this is a known problem. Working on it

mszulen, 2020-06-30:
Thanks! Let me know if some testing it needed.

pascallanger, 2020-06-30:
868 is nearly fixed. I need to work on it a bit more but I have a stable link now.

mszulen, 2020-06-30:
Wow! That's an express fix! :) Thanks for working on it! :)

pascallanger, 2020-07-01:
I've pushed the 868 fix on github. If you compile by yourself you can test it.
On my side it works great.
Note that any previously bound RX needs to be rebinded for either 915 or 868.

pascallanger, 2020-07-01:
I've added but not available yet in OpenTX the bind options to output CH1-8 or CH9-16 on the RX. It also has the telemetry on/off option at bind but the telemetry is not supported (yet).

mszulen, 2020-07-01:
Perfect! Thanks a lot, Pascal!

pascallanger, 2020-07-03:
I've also added telemetry now.
Closing the issue as for me it's working.

closed 2020-07-03 by pascallanger