pascallanger/DIY-Multiprotocol-TX-Module #379: Guide on how to add new protocol?

zendes, 2020-07-04:
Hi,

I've seen several people asking for new protocol support, and I'm no different. I'm looking to add support for a Mini-Z (1/27th scale race/drift car from Kyosho). I've also seen you reply with asking for an SPI dump of the transmitter, but I haven't seen a guide on exactly what and how to capture, so I can provide you all the necessary data at once (or if we get very lucky a code sample).

I do have the transmitter, it's A7106 based and the SPI pins are clearly visible (the RF chip is on a small daughter board) so capturing data shouldn't be too hard. I have a scope and a Salaea Logic 8 clone to do packet captures. What I don't have is the procedure on what & how to capture the data. I also have access to the car itself and a transmitter with a 4in1 module, so I can try any code to test it.

[image]

[image]

Do you have any reading material or a set of steps I can follow to get this started?

PS: this transmitter appears to be a GT2EB clone from FlySky, but [specifically made for Kyosho](https://fccid.io/WIZKT531P/Label/Label-3613497). For another FlySky transmitter (NB4) there is a firmware update that specifically adds Mini-Z as a 3rd protocol. This probably means it's based on a FlySky protocol, but slightly changed so you have to buy their brand transmitter: Kyosho does sell a programmable upgrade from the RTF one (Syncro touch KT-432PT), and [added support for this protocol to the firmware](http://kyosho.com/mini-z-info/support/settings-and-firmwaresoftware/).

pascallanger, 2020-07-04:
Hey,
Adding support for this protocol should be quite simple.
Here are the steps:
 - solder your logic analyzer to SCS, SCK, SDIO and GND
 - do a dump of the SPI bus while powering up (start recording when it is off, turn on, wait and stop recording)
 - do a separate dump where you move channels (1 dump where you move only steering, 1 dump where you move only throttle and so on for all extra channels or features)
 - do a dump of the TX in bind, you want to capture the full bind so depending how bind works you want to start recording before launching a bind
It is important to save the captures with specific names.
For adding the protocol, since you are saying it might be a derivate from flysky we will add a sub protocol instead. If it's different we will have to add a new protocol. These are 2 different things so let's first get the dumps.
Pascal

zendes, 2020-07-05:
Awesome. I'll get to work (probably in the next few days, cause 4th of July/USA) and I'll report back with the requested files.

stanislavdavid, 2020-07-07:
Is it possible to add new Frsky ACCESS protocol in future? in #239 was some documentation but nobody mentioned it from that issue.

pascallanger, 2020-07-07:
ACCESS is not on my to do list.

zendes, 2020-07-09:
Here are the requested captures. The logic analyzer channel map is:
* 0: SDIO
* 1: SCK
* 2: SCS

The transmitter doesn't have a ch3 knob, so I was only able to dump throttle, steering and ch4/lights: [spi captures kyosho kt-531p flysky mini-z.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4894545/spi.captures.kyosho.kt-531p.flysky.mini-z.zip)

The transmitter also doesn't have a bind button. The binding process for these cars is:
1. push & hold a button on the car, then turn it on
2. power on transmitter as normal
3. profit

I've attached a capture of the transmitter as I went through this process, but it might just be the same as the normal powerup.

pascallanger, 2020-07-09:
Something is wrong with your captures. The sample rate is too low and therefore not capturing all the bits. You need at least to quadruple it.

zendes, 2020-07-09:
Made some new captures at a much higher sampling rate. Let me know if you want me to capture at different settings or if there's anything else I can help with.

[spi captures kyosho kt-531p flysky mini-z 10MHz.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4899646/spi.captures.kyosho.kt-531p.flysky.mini-z.10MHz.zip)

pascallanger, 2020-07-09:
Looks much better!

pascallanger, 2020-07-09:
First quick look this is not a flysky protocol... They are sending 37 bytes where flysky sends 21 bytes.
There are 14 channels being sent! for a simple 4 channel radio...
During the first few seconds, the radio sends bind packets then normal packets.
Bind packets consist of ID followed by the RF channels table to be used in normal mode
Normal packets contains the ID and channels.
There are a number of unknown bytes but I don't think they are relevant. I've only checked the startup file so let's see with the others.

zendes, 2020-07-09:
Interesting. Maybe Kyosho commissioned it from FlySky, but wanted it to be different as to be the only source for upgraded transmitters from the RTR one? Or could there be a technical reason why they did this?

As far as I know there are no 14ch transmitters for this protocol. It's all surface/car based, and besides throttle, steering, leds & possibly gyro there isn't anything more to control on their cars. The KT-432PT (the upgrade transmitter from the RTR kit) is also listed as a 4ch transmitter.

What do you think? Should we add all 14 channels (I don't have a way to test those), or just add the first 4 and wait for someone to ask for the others? If that ever happens, the person asking should have a receiver to test the extra channels with.

pascallanger, 2020-07-10:
Please test the new protocol Kyosho in the latest version of Multi v1.3.1.39.

zendes, 2020-07-10:
Will do. I should have an answer for you in the next day or two. This went way faster than expected, so thanks!

I also found the donation button. Can't literally buy you a drink, so this will have to do... :-)

zendes, 2020-07-13:
First tests look promising. Car binds immediately and both steering and throttle work as expected. I've ordered a gyro and lights, and they should arrive in about a week. I'll report back once I confirm those work as expected too, but I don't expect any problems there. I'll keep testing (playing with the car), but I believe we can call this one done. Thanks!

pascallanger, 2020-07-13:
Can you do a 1 line description for this protocol so that I can update the protocol page?
TX models, RX models, surface protocol...

zendes, 2020-07-13:
Sure. 

This is a surface protocol called FHSS. Kyosho has [4 protocols](http://kyosho.com/mini-z-info/support/radio-compatibility-information/):
* FHSS (this one), their current protocol, introduced in about 2017
* FHS: old, but still on the market in older stock (2014-2017)
* ASF/MHS: old/discontinued in about 2014

The list of available transmitters (that I know of): [KT-531P](https://rc.kyosho.com/en/82005.html) (this is the RTR one included in each box), [KT-431PT](https://rc.kyosho.com/en/82136.html) (this is the upgrade one Kyosho sells) and the Flysky [Noble NB4](https://www.flysky-cn.com/nb4) added support for this protocol in firmware 2.0.67.

Since these are 1/27th cars, they use an [integrated RX/Servo/ESC module](https://rc.kyosho.com/en/md211.html).

pascallanger, 2020-07-15:
Protocol completed, closing issue.

closed 2020-07-15 by pascallanger

ekaszubski, 2021-01-13:
@pascallanger @zendes I found this page while looking to control a [1/27 "MINI-Z Buggy Sports" item code 32083KA](https://rc.kyosho.com/en/32083ka.html) which comes bundled with the [Perfex KT-19 transmitter](https://www.kyoshoamerica.com/DIS--KT-19-Transmitter-for-MINI-Z_p_23895.html).

Per [Kyosho's MINI-Z Info page](http://kyosho.com/mini-z-info/support/radio-compatibility-information), this car uses their FHS (not FHSS) protocol. I pulled apart the transmitter and found that it uses the A7106. I snooped the SPI to the RF board with a Digital Discovery and captured the binding process, startup/config process, and individual channels. [mini-z-fhs.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5806718/mini-z-fhs.zip)

I compared the initial config of the A7106 for FHS to other existing protocol configs in A7105_SPI and it looks identical to [FLYSKY_A7105_regs values](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/A7105_SPI.ino#L290-L297). The binding and normal-op packets look exactly like the standard FlySky [protocol](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/FlySky_a7105.ino#L132) with no extensions as far as I can tell. I tried binding my RadioMaster TX16S to the car in FlySky/Std mode and it worked right away, just like the KT-19.

The channel order in the packet appears to be:
1. steering
2. throttle
3. lights
4. steering travel
5. Others: not sure

The steering trim knob seems to just add an offset to the steering value while the steering travel is actually transmitted as a separate channel.

Would it be possible to add Kyosho/FHS as an alias for FlySky/Std or would it make more sense to just add a note in the documentation somewhere?

Thanks for the awesome work on this project!

Other info:

[image]

[a7106_datasheet_v0.1preliminary.pdf](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/5806721/a7106_datasheet_v0.1preliminary.pdf)

pascallanger, 2021-01-14:
@ekaszubski
I would just add it to the documentation... Plese check: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-flysky---0

AndreasAusD, 2021-12-28:
> MHS/ASF protocol is not discontinued. The new Mini-z Evo has a receiver of those protocols and many people, included me is using these protocols grin Would be nice to include those protocols too

I would love to see the ASF protocol. I do have some older Mini-Z and a [KT-18](https://rc.kyosho.com/en/82001ebk.html) and would like to switch to the multiprotocol module.
Took a look inside the KT-18 :
[image]
[image]
so it's a CYRF6936

jafc76, 2021-12-28:
Nice Pictures! Yes, the ASF and MHS protocol is using a CYRF6936.
Would be nice to integrate these two protocols 😁 I have Kyosho Dnanos that are ASF and all my Mini-z are MHS, included the new EVO cars

AndreasAusD, 2022-01-09:
i will try to figure out a way to get the small transmitter board to work when it is not mounted to the main board.
@pascallanger what signals would need to be captured on the CYRF6936 ?
MOSI / SCK / SS ?

pascallanger, 2022-01-10:
SS / SCK / MOSI / MISO and don't forget to connect the GNDs together.

AndreasAusD, 2022-01-10:
i think it is not just a transmitter board, the microcontroller is also on this board. CYRF6936 connects to the PIC16F690 on the right. Only chance on connectiong the logic analyzer will be with some probe pins on the solder pads of the PIC, everything else is way to small for soldering any wires to it.
and for that to work i need to connect the board with some adapter as the components are facing towards the main board.

steff-wink, 2022-01-28:
Hello there, I have a KT-17 controller with an ATMEGA88 chip running all the pins to an NRF24L01 transmitter chip, all pins nicely accessible. This controller is for a Kyosho Minium Edge 540 airplane. Is this running with ASF, MHS or FHS protocols? Any ideas? Does it make sense to provide some dumps?
I gotta build myself a logic analyzer, maybe I can get some dumps out in the next days. Can't remember what year I bought that plane. Should this (and the discussion about the KT-18) be a new issue?

pascallanger, 2022-01-29:
It should be a new issue since both radios are completly different. A 8 channels logic analyzer costs $5-10 on ebay...

steff-wink, 2022-01-29:
already ordered a logic anaylizer yesterday, new issue with dumps is coming on tuesday, thx for your efforts!! Maybe I can even put together a PR, but that will take some time.