pascallanger/DIY-Multiprotocol-TX-Module #380: iRX4 Lite: Can't update via USB or via OpenTX

dgrat, 2020-07-05:
I ordered a new mutli module. 
When I try to update the firmware via USB according to the docu, I get a no DFU capable device error.
When I try to update the firmware via OpenTX in the transmitter menu, I get a NoSync error. 
What could be the issue?

kilrah, 2020-07-05:
Update the bootloader. https://github.com/benlye/MultiBootReloader

dgrat, 2020-07-05:
This does absolutely not work. I had partial success by using a FTDI programmer, soldering, bridging boot pins and flashing a recent version of the firmware. This enabled me to flash the firmware also via my transmitter too. However, using MULTI-BootReloader-0.1.0 does not work. Via USB I get the message below, via FTDI programmer it does not recognize the device. 

	Checking for MULTI-Module ...

	MULTI-Module USB Serial device found.

	The MULTI-Module firmware will be replaced by an application which will upgrade the module's bootloader.
	Once begun, DO NOT interrupt the process, otherwise the bootloader will be corrupted and the module will not function.
	If this happens a USB-to-serial (FTDI) adapter can be used to re-program the module.

	The upgrade takes 4-5 seconds and will be complete once the LED goes off and remains off for at least 5 seconds.

	Drücken Sie eine beliebige Taste . . .

	Resetting MULTI-Module to DFU mode ...
	maple-reset 0.5
	This program is Free Sofware and has NO WARRANTY

	https://github.com/benlye/maple-reset

	Reset sequence sent to COM5
	Waiting for DFU device ............ got it.

	Device reset successful in 410ms.

	Writing MULTI-Module Bootloader upgrade app to flash ...

	dfu-util-multi 0.7

	Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
	Copyright 2010-2012 Tormod Volden and Stefan Schmidt
	Copyright 2020 Ben Lye
	This program is Free Software and has ABSOLUTELY NO WARRANTY

	This version of dfu-util has been modified for flashing an STM32 MULTI-Module. It is intended for use with Flash Multi.
	Please report bugs to https://github.com/benlye/dfu-util-multi

	Filter on vendor = 0x1eaf product = 0x0003
	Opening DFU capable USB device... ID 1eaf:0003
	Run-time device DFU version 0110
	Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="UNDEFINED"
	Claiming USB DFU Interface...
	Cannot claim interface

dgrat, 2020-07-05:
I checked. Changing the bootloader via flash-multi does also not work. So I checked every combination: USB/FTDI with and without boot pins bridged. There is no way to reflash the bootloader. This means, the USB port is essentially useless.

pascallanger, 2020-07-05:
The message above shows that you have the wrong driver. You must upgrade your driver to libusk: https://github.com/benlye/flash-multi/blob/master/doc/Troubleshooting.md#re-installing-the-maple-dfu-device-drivers
Pascal

closed 2020-07-05 by pascallanger