pascallanger/DIY-Multiprotocol-TX-Module #384: Graupner HOTT... Losing Telemetry

Andrea76it, 2020-07-07:
Hi, I have a problem with the LUA "Graupner Hott" script to configure telemetry ... the message "No Hott Telemetry" always appears
- I use an iRX4 Lite module on my FrSky XLite Pro radio
- I have updated the Module to version 1.3.1.36 (but the problem also occurs with 1.3.1.24)
- I have connected a GR-12 + 3xG Receiver
- I do corrrectly the "RF Tune"
- The Radio correctly display the telemetry sensors
- The connection is stable
- When I open the "Graupner Hott" LUA script to configure the telemetry, the message "No Hott Telemetry" always appears and the radio loses the connection with the receiver.
Why does it happen?

Andrew

pascallanger, 2020-07-07:
Which version of OpenTX?
Have you updated the firmware on your RXs?

Andrea76it, 2020-07-07:
the version of Open TX is 2.3.9
the version of GR-12 3xG is 2.02
I tried the LUA Script that came with openTx: it displays strange characters and then crashes.
I tried your LUA script and it says "No Hott Telemetry"

Thank you

pascallanger, 2020-07-07:
The LUA script with OpenTX is not different from the one posted on this github... So I'm not sure about the difference you are seeing...
I know the standard GR-12 is working since I've used to rverse the protocol and I know a few people using it. I'm not sure what's different with this 3xG... Can you try different versions on it?
After I would need a dump from the module to see the telemetry packets without this there is nothing I can do.

pascallanger, 2020-07-07:
Do you have other receivers or just this one?

Andrea76it, 2020-07-07:
#I have several receivers:
Gr-24 3xG + 3A + 3M + Vario (9DOF + Vario)
Gr-18 3xG + 3A + Vario (6DOF + Vario)
Gr-12 3xG (3DOF)
Gr-12L (Satellite Version, only SUMD and Telemetry connection)
But they are mounted on model aircraft. Tomorrow evening I try to take one off and test it with a different model.
All receivers are updated to the latest firmware version.
The abbreviation 3xG indicates the presence of a Gyro on board the receiver. They are the newest versions.

I'm not sure that the two LUA scripts are identical, because they have different sizes in kb.

How can I do a dump from the module?

Thanks Andrea

robustini, 2020-07-08:
@Andrea76it maybe you're doing something wrong.
I use a GR18 regularly without any problem, the firmware is not even the latest, I still have 2.00 and 2.04 is available.

[image]

Andrea76it, 2020-07-08:
Hi pascallanger,
Hello Marco,
I did some tests with different receivers and I got this result:

Gr-12 3xG: does not work, the message "No Hott Telemetry ..." appears even though the sensors are correctly recognized in the Telemetry menu of FrSky. (Photos 1 and 2)

Gr-18 3xG + 3A + Vario: the receiver is correctly recognized, I manage the RX menu correctly, but I don't see the "GAM-4S Lipo" sensor, I only see the Vario sensor on board the receiver. (Photos 3)

Gr-12L (Satellite Version): the receiver is recognized correctly, I manage the RX menu correctly, but I don't see the "GAM-4S Lipo" sensor

Some idea?
I am attaching the photos

Regards, Andrea
[image]
[image]
[image]

pascallanger, 2020-07-08:
Are you sure you can't see the gam? You need to cycle through the connected sensors. Have you done it?

Andrea76it, 2020-07-08:
Hi, I solved the problem with the GAM sensor on the GR-18 receiver. It depended on the fact that the receiver was powered at 7.4v. I tried to power the 5v receiver and now the sensor is recognized correctly. This particular sensor (Lipo Cells) is powered by the receiver. Very strange, because the receiver can be powered up to 8.4v.
[image]
[image]

Instead, I have not solved the "No Hott telemetry ..." problem on the GR-12 3xG. I also tried another receiver of the same model (Art. No. 33576) and it behaves in the same way. I also tried to overwrite the firmware in case it wasn't the latest version, but nothing has changed. I am using the latest firmware version 2.02. Damn!
[image]

I'm trying another solution, that of building a Hott to SmartPort telemetry converter (Using a Teensy 3.2 with 2 serial ports), in order to use Hott Sensors attached directly to the FrSky receiver ... but it is very complex.

pascallanger, 2020-07-08:
As said previously, you need to dump the telemetry stream so I can see what this RX is sending...

Andrea76it, 2020-07-08:
can you tell me how can i do it?

robustini, 2020-07-08:
> 
> 
> can you tell me how can i do it?

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Debug.md

Andrea76it, 2020-07-08:
Thanks ... now I try!

robustini, 2020-07-09:
> 
> 
> Thanks ... now I try!

Capture 100/200 lines and copy them here.

Andrea76it, 2020-07-10:
Multiprotocol version: 1.3.1.37
Protocol selection switch reads as 0
Using bank 0
Read ID from EEPROM
Module Id: 6552e140
Init complete
RXID:  00 00 02 82 A0
Protocol selected: 57, sub proto 0, rxnum 4, option 30
Proto=HoTT, nbr_sub=2, Sub=Sync  , Opt=2
T1= 00 00 62 31 3F 62 00 31 00 00 00 62
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0D
T1= 00 00 64 31 3E 64 00 31 00 00 00 64
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0E
T1= 00 00 69 31 3F 69 00 31 00 00 00 69
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:09
T1= 00 00 63 31 3E 63 00 31 00 00 00 63
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0A
T1= 00 00 5B 31 3F 5B 00 31 00 00 00 5B
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0B
T1= 00 00 61 31 3E 61 00 31 00 00 00 61
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0C
T1= 00 00 64 30 3F 64 00 30 00 00 00 64
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0D
T1= 00 00 69 30 3F 69 00 30 00 00 00 69
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0E
T1= 00 00 63 31 3F 63 00 30 00 00 00 63
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:09
T1= 00 00 5A 31 3F 5A 00 30 00 00 00 5A
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0A
T1= 00 00 62 31 3E 62 00 30 00 00 00 62
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0B
T1= 00 00 66 31 3F 66 00 30 00 00 00 66
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0C
T1= 00 00 6A 31 3F 6A 00 30 00 00 00 6A
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0D
T1= 00 00 64 31 3F 64 00 30 00 00 00 64
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0E
T1= 00 00 5C 31 3F 5C 00 30 00 00 00 5C
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:09
T1= 00 00 64 31 3F 64 00 30 00 00 00 64
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0A
T1= 00 00 5C 31 3F 5C 00 30 00 00 00 5C
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0B
T1= 00 00 66 31 3F 66 00 30 00 00 00 66
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0C
T1= 00 00 5D 30 3E 5D 00 30 00 00 00 5D
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0D
T1= 00 00 67 31 3F 67 00 30 00 00 00 67
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0E
T1= 00 00 5E 31 3E 5E 00 30 00 00 00 5E
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:09
T1= 00 00 67 30 3F 67 00 30 00 00 00 67
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0A
T1= 00 00 5E 31 3F 5E 00 30 00 00 00 5E
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0B
T1= 00 00 68 31 3F 68 47 30 00 00 00 68
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0C
T1= 00 00 67 31 3F 67 47 30 00 00 00 67
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0D
T1= 00 00 61 31 3E 61 47 30 00 00 00 61
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0E
T1= 00 00 68 31 3F 68 47 30 00 00 00 68
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:09
T1= 00 00 60 31 3E 60 47 30 00 00 00 60
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0A
T1= 00 00 68 31 3E 68 47 30 00 00 00 68
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0B
T1= 00 00 69 31 3F 69 47 30 00 00 00 69
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0C
T1= 00 00 64 31 3F 64 64 30 00 00 00 64
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0D
T1= 00 00 5E 31 3F 5E 64 30 00 00 00 5E
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0E
T1= 00 00 62 31 3F 62 64 30 00 00 00 62
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:09
T1= 00 00 64 31 3E 64 64 30 00 00 00 64
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0A
T1= 00 00 67 30 3E 67 64 30 00 00 00 67
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0B
T1= 00 00 64 31 3F 64 64 30 00 00 00 64
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0C
T1= 00 00 60 31 3F 60 64 30 00 00 00 60
T0= 00 01 00 00 00 00 00 00 00 00 00 00
Sensor:0D
T1= 00 00 64 31 3E 64 64 30 00 00 00 64
Busy 25
Busy 52
Failsafe received
Busy 33
Busy 52
Busy 21
Busy 26
Busy 28
Busy 26
Busy 68
Failsafe received
Busy 58
Busy 27
Busy 29
Busy 74
Failsafe received
Busy 18
Busy 25
Failsafe received
Busy 52
Busy 28
Busy 28
Busy 50
Busy 32
Busy 26
Failsafe received
Busy 50
Busy 22
Busy 27
Busy 22
Busy 29
Failsafe received
Busy 26
Busy 18
Busy 45
Busy 48
Busy 46
Busy 21
Failsafe received
Busy 25
Busy 52

Andrea76it, 2020-07-10:
At line 122 I pressed on "Graupner Hott" LUA script.

Andrea

pascallanger, 2020-07-11:
The RX is not replying to the config text mode request...
Have you tried to use this RX with the NO_SYNC subprotocol? what happens if in this mode you launch a text config?

Note that you have failsafe activated which is going to introduce jitter on your servos/channels: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#HoTT---57

Andrea76it, 2020-07-11:
It works!!!! 
I have activated the No_Sync option and now it works correctly.
What is the difference between Sync and No_Sync? the polling of the sensors?
Andrea

pascallanger, 2020-07-11:
Humm, strange behavior compared to all other RXs.
No_sync is for no telemetry sync transmission. I recommend flying in sync mode for best telemetry performance.
I need more details to get an idea of what's happening.
With the power turned on on the RX at all time, what happens if you use first use sync, then switch the sub protocol to no_sync? First does the telemetry still get to the TX after the switch? Second can you access text config mode?

pascallanger, 2020-07-14:
@Andrea76it Have you tested?

Andrea76it, 2020-07-16:
Hi Pascal, sorry if I didn't answer you before, but I was busy at work and I was unable to take the tests you asked for. During the weekend I will perform all the tests and send you a detailed feedback.
Greetings, Andrea

Andrea76it, 2020-07-21:
Hi Pascal, these days I have done several tests.
If I change from Sync Mode No_Sync Mode I lose the connection with the receiver and I have to restart both the RX and the TX.

This is my configuration.
[image]
[image]

I can only access Text Mode if I'm in No_Sync mode.

I did another test: if I send many commands at the same time, sometimes it loses the connection and the signal goes to zero.
I lose both telemetry and the main radio signal that controls the servos.
I am attaching two sample videos, both of the telemetry and of the main screen.

https://youtu.be/3WhiqBn_x18

https://youtu.be/a7KtWzqC5zA

I don't think it's appropriate to fly with this receiver.

HazzaHFPV, 2020-07-21:
Failsafe in the radio has to be set to "receiver", otherwise it will interfere with the link. If I translate correctly it is set to "hold" in your case.

To set Failsafe correctly with the Hott protocol you have to set your Failsafe configuration once. These settings are then send to the receiver and the receiver saves them. Once this has happened you have to set Failsafe to "receiver", otherwise the Failsafe settings are send to the receiver in a regular time interval and disrupts the reaction shortly as the receiver saves the settings each time. It then looks like regular small failsafes as in your video. I hope O could explain it well as I made the same mistake in the beginning. It is written in the documentation:

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#hott---57

"Failsafe MUST be configured once with the desired channel values (hold or position) while the RX is up (wait 10+sec for the RX to learn the config) and then failsafe MUST be set to RX/Receiver otherwise the servos will jitter!!!"

closed 2020-08-25 by pascallanger