pascallanger/DIY-Multiprotocol-TX-Module #385: GRAUPNER HOTT RX Mode - More an Request then an issue

Xillius178, 2020-07-12:
Hello, 
would it be possible to install a "GRAUPNER HOTT RX" mode like that of FRSKY RX, FLYSKY RX, or DSM RX?

My ulterior motive: 
Could friends who have a Graupner radio control fly with me as student, 
since I also have an external module for the transmission? 

Would be Super Pascal if you could incorporate that ... 
Thank you for the rapid development of the code. 
A lot has developed in a year.

Greetings Xillius

kilrah, 2020-07-12:
Bind time doesn't affect anything since you can put the receiver in bind mode before initiating bind on the radio.

Xillius178, 2020-07-12:
Tell me colleague what problem do you have?
Do you have to add your mustard to every message I write? 
And no helpful criticism, or even expected help.

You don't seem to get it?

With some receivers I have to hold the bind button when switching on.
So and then I should press the OK button on the transmitter with the third hand, and that with my illness.

Better you show me that first, and then we'll talk about it again.

Unbelievable how you can simply ignore another person's disability.

kilrah, 2020-07-12:
What protocol requries you to hold the bind button while starting bind on the radio? Don't know of one.

pascallanger, 2020-07-12:
1. A RX for the HoTT protocol is not possible since the hopping table also has not been reversed completely.
2. Which protocol do you need me time to bind? For almost all protocols you can put the RX in bind then launch a bind on the TX.

HazzaHFPV, 2020-07-12:
I think there might be a misunderstanding. As you mentioned FrSky earlier:
To put FrSky receivers into bind you have to hold the bind/Failsafe button on the receiver, then power up the receiver. Then you can release the button, you don't have to keep it pressed.

Then you have all the time in the world to press bind on your transmitter.

The trainer function with Graupner works fine with  simple trainer cable (3.5mm audio jack), assuming your friends have a DSC-port on their Graupner Radio. It is not wireless, but it works. I have tested it between Graupner MZ-24 and Jumper T16.

On a final word: I don't think @kilrah was being disrespectful, he just simply pointed out that most receivers don't need the button to be pressed for the whole binding process. Kilrah is a driving and helpful force in development in our hobby and was simply being objective in his reply, it was not personal. Your reaction, however, was personal and impolite.

So, thank you @pascallanger and @kilrah for doing what you do and being so patient and understanding along the way.

pascallanger, 2020-07-14:
@Xillius178 Is there something to do or should I close this issue?

pascallanger, 2020-07-15:
No answer, closing

closed 2020-07-15 by pascallanger