pascallanger/DIY-Multiprotocol-TX-Module #386: Radiolink protocol support

zendes, 2020-07-13:
Hi. After completing the Kyosho protocol just now, I'm hoping you're in the mood to add another one.

This time it's a [Radiolink](http://radiolink.com.cn/doce/product-categorie-39.html) [RC4GS](http://radiolink.com.cn/doce/product-detail-131.html)/[RC6GS](http://radiolink.com.cn/doce/product-detail-135.html). These are generic 4/6 channel surface radios with several receivers available. They feature basic telemetry (signal strength & battery voltage), and offer a built-in gyro for stabilization.

I have the RC6GS radio and a few receivers that I can probe: [R4FGM](http://radiolink.com.cn/doce/product-detail-156.html), [R6F](http://radiolink.com.cn/doce/product-detail-137.html) & http://radiolink.com.cn/doce/product-detail-149.html. It's a cc2500 based system, and I was able to find [the transmitter module](https://www.aliexpress.com/i/4000597705561.html) separately.

Is the procedure to start this protocol the same? Attach logic analyzer to all pins, and capture startup, bind, and each of the 6 channels separately?

I have a variable lab bench power supply, so I can also capture the receiver with varying input voltages for the return telemetry if that's any help.

Let me know if you have time and what other information I need to supply to start.

RC6GS transmitter
[image]
[image]
[image]

R7FG receiver
[image]
[image]

R6F receiver
[image]
[image]

pascallanger, 2020-07-13:
Go ahead and do the exact same dumps.
The Kyosho protocol was easy since the id and hopping table was part of the bind packet. Having access to the RX   will be needed if that's not the case. But first do the TX dumps.

zendes, 2020-07-15:
Captures attached (at 10MHz). There is no special bind mode on the transmitter, you have it powered up and push&hold a button on the receiver, and that'll bind it. Let me know if you need receiver captures and from which scenarios.

[Radiolink RC6GS SPI captures.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4922878/Radiolink.RC6GS.SPI.captures.zip)

pascallanger, 2020-07-15:
Wow, this one is going to be really hard!
I haven't been able to find anything yet...
The check at the end looks like a sum but not as simple as that.
It hops on 16 frequencies but "randomely", there are 2 bytes in the payload moving all the time which might be linked but no idea how.
The channels are on 12 bits but they seem to be scrambled as well...

zendes, 2020-07-15:
I'll go make some captures of the simplest receiver I have, both with and without transmitter connected. Maybe that'll help.

Can you share a bit of the process you're using and how you're getting the capture data to be more manageable? I tried looking at it but scrolling through pulseview isn't exactly user friendly.

pascallanger, 2020-07-16:
I'll share a readable version of your dump when I'll be in front of my computer.
I'm not sure having a RX dump will help but it can't harm to have it.
What I think would be useful to get is:
- a really long dump without touching the radio
- a dump where you move really slowly the steering wheel in both directions (center->left->center->right)

pascallanger, 2020-07-16:
Here is the startup file: [startup.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4930406/startup.zip)
You can load it in excel, do a text to columns and apply filters to extract what you need. At least this is what I'm doing.
I would need the above to continue looking at it.

pascallanger, 2020-07-16:
0x01 | 0x3A | 0x99 | 0x22 | 0x3A | 0xFF | 0xF3 | 0x5F | 0x2C | 0xAA | 0x52 | 0x95 | 0xAA | 0xA0 | 0x0F | 0x00 | 0xFF | 0xFB | 0xDF | 0xFF | 0xFE | 0xF7 | 0xBF | 0xFF | 0xA1 | 0x0F | 0x7D | 0x03 | 0x00 | 0x00 | 0x00 | 0x01 | 0x53
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
-- | ID? | ID? | ID? | ID? |ST_7..0|7..4=THR_3..0,3=unknown,2..0=STR_10..8|THR_11..4| -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | Sum of all bytes&0xFF

Check=ok
Steering=strange the value seems to be on 11 bits, center 1023, low 170, high 1876. The 12th high bit value changes like if it was the most lower bit...
Throttle=12bits ppm format but centered on 1535 (instead of 1500, trim?), low 1109, high 1962.

pascallanger, 2020-07-16:
I really need a longer dump.
HOP1 3..0 is the index of the current RF channel in use from this table  {0x12,0xC6,0x66,0x72,0x1E,0x42,0x06,0x4E,0x36,0xAE,0x8A,0xA2,0x2A,0x7E,0x96,0x5A}
HOP2 7..4 values are between 0 and 7, all others are between 0 and F

pascallanger, 2020-07-16:
After spending a full day on it, if you want to know how the pseudo random generator is done you would need to offer a couple of more beers!!!
That was a hard one. I can't beleive that I've been able to reverse it in the end.
So basically, I can now generate a valid frame with the pseudo random generated value and the associated frequency. I also know how to inject throttle and somewhat steering. I haven't looked at the other channels but that should be a piece of cake compare to everything else.

zendes, 2020-07-20:
Here is a [r6f receiver startup dump](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4945669/r6f.startup.zip) and a [10 sec dump of the TX & RX on the same logic analyzer, while I slowly move the steering](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4945670/tx.rx.powerup.steering.sr.zip). I've added both SPI decoders, so they should load automatically. If not, the pin mapping is:

TX: 4 mosi, 5 miso, 6 clk, 7 cs
RX: 0 mosi, 1 miso, 2 clk, 3 cs

My last donation ID was 3BH75810W9931454U. If we call that "1 beer", let me know how many more you want. Happy to help fund the effort you're putting in! :-)

I have a second transmitter on the way, so once that comes in I should be able to get you another transmitter ID, if that's helpful.

Hope this is enough info. If not, lmk exactly what you need, how long you want the dump to be and what channels you want me to move and I'll get them. I've been out of town for a few days bc girlfriend birthday, but I'm back now.

pascallanger, 2020-07-20:
I was kidding about the extra beers ;-)
Since you are back, I've looked at the others channel dumps and I'm confused by some of them.. In some the throttle and steering are moving as well as a channel, in others I have more than 1 channel moving and I'm not sure if the channles are centered either... Can you make sure that all your channels are centered (no trim)  and there are no mixers when you do dumps?

pascallanger, 2020-07-20:
0x01 | 0x3A | 0x99 | 0x22 | 0x3A | 0xFF | 0xF3 | 0x5F | 0x2C | 0xAA | 0x52 | 0x95 | 0xAA | 0xA0 | 0x0F | 0x00 | 0xFF | 0xFB | 0xDF | 0xFF | 0xFE | 0xF7 | 0xBF | 0xFF | 0xA1 | 0x0F | 0x7D | 0x03 | 0x00 | 0x00 | 0x00 | 0x01 | 0x53
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
TELEM| ID? | ID? | ID? | ID? |CH|CH|CH|CH|CH|CH|CH|CH|CH|CH|CH|CH|CH|CH|CH|CH| CH|CH|CH|CH|CH|CH| HOP1 | HOP2 |UNK1|UNK2|UNK3|CHECK

TELEM=request telemetry from RX when equals to 0x03 otherwise 0x01.
CH: 16 channels are transmitted (far from 6). Each channel is on 11 bits: 1023 centered, 170 low , 1876 high. They surely can be extended since CH3 goes from 177 to 1883 (trim on this channel?). CH4 is a toggle between 341 and 1705, CH5 and CH6 are equal to 341 so I think they are also on a toggle. CH7 shows a value of 1000. CH8 must be an unconnected pin nearby CH3 since it moves with CH3 with a ratio. CH9 to CH14 are centered at 1023. CH15 and 16 shows a value of 1000.
HOP1 and HOP2 are the key of the hopping sequence.
UNK1,2,3 are unknown and fixed values.
CHECK=this is the sum of all payload bytes

pascallanger, 2020-07-20:
Telemetry packets received from RX
0x3A | 0x99 | 0x22 | 0x3A | 0x03 | 0xFF | 0x00 | 0x00 | 0x00 | 0x00 | 0x00 | 0x00 | 0x00 | 0x31 | 0x58
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
ID1 | ID2 | ID3 | ID4 |TYPE|VAL1|UNK|UNK|UNK|UNK|UNK|UNK|UNK|CHECK|VAL2
- ID: for me it confirms that this is the ID
- TYPE: might be that 0x03 is the TELEM answer of the request from the TX
- VAL1: changes slightly over time between 0xFD..0xFF -> unknown
- CHECK: This is the sum of all previous bytes
- VAL2: varies from 0x51 to 0x5A -> unknown and strangely unprotected from the check

pascallanger, 2020-07-20:
@zendes Can you do the different dumps:
- TABLE (dump from TX): This value might change every time the TX is powered on => I need some small dumps to see how it works (normal mode, not startup)
- RX telemetry (dump from TX): change the voltage as you suggested and do different dumps writing down the applied voltage in the file name
- RX telemetry (dump from TX): move away the RX from the TX to see the effect
- RX bind (dump from RX): put the RX in bind mode and do a dump with the TX off. That will give us some info on which RF channels it is looking for the TX.

zendes, 2020-07-21:
As requested, here are the dumps. All dumps were created with all channels on 0, where possible:
Ch1: steering, analog, 0
Ch2: throttle, analog, 0
Ch3: knob, analog, 0
Ch4: 2pos switch, -80 
Ch5: 3pos switch, 0
Ch6: 3pos switch, 0

[battery telemetry.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4951735/battery.telemetry.zip): I've saved the files under what the TX showed as received battery voltage, in 0.5v steps from 7.0v to 10.0v.

[rssi telemetry.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4951736/rssi.telemetry.zip): I've saved the files under an approximation of what the TX showed as received RSSI values. They change rapidly, but never more than a few steps.

[rx bind.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4951737/rx.bind.zip): I did a few "switch receiver to bind mode" (all within the first 1-2 sec of the capture) and one "startup + enter bind mode after about 4 seconds".

[tx table.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4951740/tx.table.zip): not sure what you meant by table, but I've turned the TX off and back on again in between each of these captures. If that's not what you need, lmk.

zendes, 2020-07-21:
Also maybe of note: the telemetry captures were all done using a different RX (R7FG) than the bind ones (R6F). Not sure if that matters, just thought I'd let you know.

zendes, 2020-07-21:
Just read something in the manual about a subsidiary ID. It's midnight here, but I'll go dive into that tomorrow. It sounds similar to a receiver ID in FrSky. It can be turned on and off, and when on can have 10 different values. It's used to distinguish between several bound receivers. So far I've had it off, but I'll make some captures tomorrow with all possible values.

pascallanger, 2020-07-21:
@zendes The telemetry is now reversed.
Check my previous posts for updated details on both TX frames and RX telemetry frames.
I'm waiting for the RX number dumps to see how it's done but at this stage I should be able to write the full protocol.

pascallanger, 2020-07-21:
I've just looked at the documentation you linked on the 2 TXs. This is pure marketing, under the hood it's completly different...
- Resolution 4096, in reality 2048
- FHSS on 67 channels, in reality 16
- "Only" 12ms from transmitter to receiver, while this is true (time for the packet to get over the air from the TX to RX) it gives a false impression since the frame rate is 20ms...

I've also noticed that the surface TXs can be used with the air receivers at least the R8EF and R8FM are listed. So my guess is that the T8FB and T8S radios are using the exact same protocol. To be confirmed...

zendes, 2020-07-22:
Somehow that doesn't surprise me...

Anyhow, I've figured out what they mean with the whole subsidiary ID. It's indeed a receiver ID, but when turned off it works similar to a wildcard so all bound receivers (irrespective of their bound ID) work. I've tested this by turning it on with ID 1, binding a receiver, then switching it to ID 2 and binding another one. When the feature is on, only the receiver bound to that ID works. When it's disabled, both receivers operate at the same time (wildcard). This might be the 4th byte in the ID field you found. I'll attach a capture with the receiver ID enabled, for several of the IDs.

zendes, 2020-07-22:
About the last 3 UNK bytes in the telemetry packet: I think those are for future expansions. There are RATE: 000rpm and T:NULL fields on the TX telemetry page, but [the manual](http://radiolink.com.cn/doce/UploadFile/ProductFile/RC6GSManual.pdf) has the following note: `The 'RATE: RPM' and 'T: NULL' telemetry functions are still under development. If there are new
developments, we will announce them to the Radiolink official website as soon as possible, thanks for your
attention.`. There also isn't an RPM sensor available for these receivers.

[image]

pascallanger, 2020-07-22:
For the RX_ID, they use the 1st byte TYPE to send it.
if(RX_ID) TYPE |= ((RX_ID+2)<<4)+4;
But I'm not sure if I should use it... On multi I'm modifying directly the TX global ID with the RX number so it appears like a different TXs to the RX. You have up to 64 RX numbers available. Here that would limit the number of RX numbers to 10 which might be enough but...
Let me know what you think.

pascallanger, 2020-07-22:
The failsafe channels are sent all the time. They are in fact the extra 8 channels being transmitted.
2 options:
- CH1-8 are normal channels and CH9-16 are failsafe values. 
- CH1-8 are normal channels and failsafe values are defined through the failsafe page on OpenTX. But there is no HOLD or NO_PULSE that you can set on that failsafe page.
Thoughts?

pascallanger, 2020-07-22:
I've pushed the RadioLink protocol on Master but I haven't verified anything.
By default I've left #define RLINK_FORCE_ID enabled meaning that it will use the ID from your radio. Test like this and tell me how it goes. If it works then comment that line (add //) and test again.
Telemetry should also work. I'm not sure of the timing but let's see.

zendes, 2020-07-22:
Will do. I recently got a Radiomaster TX16S, which has an internal module. I previously installed firmware by plugging my module in via USB and just flashing it via Arduino IDE, but OpenTX doesn't like the compiled file to flash via the TX. Is there a different procedure to compile/prep so OpenTX can flash the internal module?

zendes, 2020-07-22:
This protocol only supports 10 IDs and one wildcard. Are you saying that your way of setting the RX ID would only support the 10 IDs but not the wildcard? I was hoping to be able to use ID 0 as the wildcard, and then 1-10 for the specific RX IDs, but I'm not sure that's a big loss tbh. You can always just bind everything with the same RX ID and still get the "wildcard" behaviour.

pascallanger, 2020-07-22:
To flash through OpenTX you must use in Arduino the "Export compiled Binary" under Sktech.

pascallanger, 2020-07-22:
> This protocol only supports 10 IDs and one wildcard. Are you saying that your way of setting the RX ID would only support the 10 IDs but not the wildcard? I was hoping to be able to use ID 0 as the [...]

What I'm saying is that there are 2 ways to implement it:
- The RadioLink original way which is 10 RXs match+wildcard. Not that I still need a dump of the TX with ID=1 and a RX on.
- The multi way which is 64 RXs match but no wildcard.

pascallanger, 2020-07-22:
Still no test?

zendes, 2020-07-22:
It's 2pm here, so I'm at work. I'll be able to test once I get off, so in about 3-4 hours.

pascallanger, 2020-07-22:
Nearly midnight here so I will know only tomorrow. Please give as much as details as you can if it doen't work properly.

zendes, 2020-07-23:
Step 1: [TX & RX with ID set.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4964083/TX.RX.with.ID.set.zip)

zendes, 2020-07-23:
First test, with `RLINK_FORCE_ID defined`: bind succeeds, and channels move as expected. I get RSSI telemetry and both voltages in OpenTX.

Issues/observations:
1. Both voltages are 50% of what they should be. Bench PSU at 10.0v, radio shows 5.1v. Bench PSU at 4.0v, radio shows 2.0v.
2. Rx Battery voltage shows up as sensor A1 instead of RxBt.
3. RSSI shows a value of 225 with no unit, while the RadioLink TX shows -30dBm. An X8R connected to OpenTX shows 80dB.
4. I first through the servo moved more "jittery", but I'm not sure this is really a thing. Thought I would mention it, but maybe ignore this one for now until I have more data/find out I was wrong.

zendes, 2020-07-23:
Second test, with `RLINK_FORCE_ID commented out`: bind also succeeds, and channels again move as expected. Same issues as above.

I now believe issue 4 is because of dropped frames between the TX & RX, as I sometimes get "telemetry lost" from the TX. Tried with and without low power mode, and I've also done the cc2500 fine tune which did not improve the situation.

I've done a 10sec RX capture of me slowly sweeping ch1, and where the servo clearly moved in "chunks". I've also done a capture of the RX startup while having OpenTX powered up, in case that's useful. [v1.3.1.44 captures.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4964266/v1.3.1.44.captures.zip)

zendes, 2020-07-23:
To answer your other question about RX ID: no strong opinion, and I don't think it functionally matters to a user. This isn't really a "killer feature" on this system anyway. They sell it as "if your first boat gets stuck, you enable this and send out a second boat to rescue it while the first one is still powered up". You can accomplish the same thing by first binding everything to ID 1 and when necessary re-binding to ID 2. I'd say pick whatever feels more natural in the code.

zendes, 2020-07-23:
I didn't mention this, but both times (with and without `#define RLINK_FORCE_ID`) I was able to bind to both my R6F and R7FG receivers.

pascallanger, 2020-07-23:
Ok that's a good start.
Telemetry values are as expected. I'll fix the half voltage values. The naming and units are what they are since I'm using frskyd telemetry which I think is ok. Check the doc that I've written on this proto.

For the jerkiness, you can do 2 tests:
- disable this protocol telemetry in _config.h. (you'll find rlink hub something) and test. I think it should fix the jerkiness.
- increase this value https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/a9b7ab9a0679fcbf79cd3f119648f4d2299e6d17/Multiprotocol/RadioLink_cc2500.ino#L139 by step of 50 or 100. I'm thinking that's where it goes wrong.

pascallanger, 2020-07-23:
I've just pushed v1.3.1.45 which is fixing the voltage values. I've also increased the value as I suugested above. Please test.

pascallanger, 2020-07-23:
On the telemetry tab you have TQLY which is how many telemetry packets the TX is receiving 100=all, 0=none. What are you getting?

pascallanger, 2020-07-23:
Hummm this is what I was thinking:
With the ID 0x3A,0x99,0x22,0x3A the table is 0x12,0xBA,0x66,0x72,0x1E,0x42,0x06,0x4E,0x36,0xAE,0x8A,0xA2,0x2A,0x7E,0x96,0x5A
With the ID 0x62,0x4D,0xF9,0x27 the table is 0x06,0x36,0x5A,0x1E,0x72,0x2A,0x66,0x8A,0x42,0x4E,0xBA,0xA2,0x96,0x7E,0x12,0xAE

When the 1st table is used with the second ID (what is done right now in the code), then only 3 frequencies are in common explaining the large quantity of missing frames.
Now we need to understand how the ID influence the table.

pascallanger, 2020-07-23:
Table in another format:

0x3A | 0x99 | 0x22 | 0x3A | 1 | 15 | 8 | 9 | 2 | 5 | 0 | 6 | 4 | 14 | 11 | 13 | 3 | 10 | 12 | 7
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
0x62 | 0x4D | 0xF9 | 0x27 | 0 | 4 | 7 | 2 | 9 | 3 | 8 | 11 | 5 | 6 | 15 | 13 | 12 | 10 | 1 | 14

pascallanger, 2020-07-23:
Staying with FORCE_ID enabled, Do different dumps by changing the ID on this line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/75dc616130bfe854c2d684c3f9ae825092c0d85f/Multiprotocol/RadioLink_cc2500.ino#L44
Since we do not have overlaps due to the timezone, I'm giving you a long list of dumps to do on the RX with:
- \x00\x00\x00\x00
- \x00\x00\x00\x01
- \x00\x00\x01\x00
- \x00\x01\x00\x00
- \x01\x00\x00\x00
- \x00\x00\x00\x02
- \x00\x00\x02\x00
- \x00\x02\x00\x00
- \x02\x00\x00\x00
- \x00\x00\x00\x0F
- \x00\x00\x0F\x00
- \x00\x0F\x00\x00
- \x0F\x00\x00\x00
- \x00\x00\x00\x10
- \x00\x00\x10\x00
- \x00\x10\x00\x00
- \x10\x00\x00\x00
The goal is to see if all bytes of the ID are used to build this table and also get some values.

zendes, 2020-07-24:
First test using v1.3.1.45, no modifications (so RLINK_FORCE_ID defined)

* Receiver still binds as expected
* Voltages are now showing up correctly
* No irregularities on servo movement, no telemetry_lost messages
* Telemetry values R7FG: RSSI 219, TRSS 105dB, RQly 0, TQly 20
* Telemetry values R6F: RSSI 212, TRSS 85dB, RQly 0, TQly 20

I'll see how many ID dumps I can get you today...

pascallanger, 2020-07-24:
TQly of 20 means that all the packets from the RX are received. I forgot that the telemetry is every 5 packets... I'll correct that to be 100..0%.

zendes, 2020-07-24:
It's about 1am here, so I'm calling it a night. I went through your list and only got a bind on `\x0F\x00\x00\x00`, so then I tried some other IDs and from my first tests it looks like the first byte needs to be either 0x0F or 0x0E. Right now I have captures of 5 different IDs that bind. Let me know which other IDs you want me to try.
```
\x00\x00\x00\x00: doesn't bind
\x00\x00\x00\x01: doesn't bind
\x00\x00\x01\x00: doesn't bind
\x00\x01\x00\x00: doesn't bind
\x01\x00\x00\x00: doesn't bind
\x00\x00\x00\x02: doesn't bind
\x00\x00\x02\x00: doesn't bind
\x00\x00\x00\x0F: doesn't bind
\x00\x00\x0F\x00: doesn't bind
\x00\x0F\x00\x00: doesn't bind
\x0F\x00\x00\x00: binds, but about every sec loses telemetry
\x00\x00\x00\x10: doesn't bind
\x00\x00\x10\x00: doesn't bind
\x00\x10\x00\x00: doesn't bind
\x10\x00\x00\x00: doesn't bind
\x1F\x00\x00\x00: doesn't bind
\x2F\x00\x00\x00: doesn't bind
\x0E\x00\x00\x00: binds, but about every other sec loses telemetry
\x0D\x00\x00\x00: doesn't bind
\x0C\x00\x00\x00: doesn't bind
\x0E\x01\x00\x00: binds, but about every other sec loses telemetry
\x0E\x02\x00\x00: binds, but about every 2-4 sec loses telemetry
\x0E\x03\x00\x00: binds, and doesn't lose telemetry (no opentx warning)
```
[TX Addresses.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4970747/TX.Addresses.zip)

pascallanger, 2020-07-24:
If you still have a little time, can you try 0e 00 03 00 and 0e 00 00 03?

pascallanger, 2020-07-24:
From the dumps I've been able to extract only this:
ID | 3A | 99 | 22 | 3A | Table | 1 | 15 | 8 | 9 | 2 | 5 | 0 | 6 | 4 | 14 | 11 | 13 | 3 | 10 | 12 | 7
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
ID | 62 | 4D | F9 | 27 | Table | 0 | 4 | 7 | 2 | 9 | 3 | 8 | 11 | 5 | 6 | 15 | 13 | 12 | 10 | 1 | 14
ID | E | 0 | 0 | 0 | Table | 5 | 9 | 14 | 4 | 13 | 1 | 8 | 10 | 3 | 2 | 6 | 12 | 7 | 11 | 15 | 0
ID | E | 1 | 0 | 0 | Table | 14 | 4 | 1 | 3 | 0 | 9 | 7 | 10 | 15 | 8 | 5 | 2 | 11 | 12 | 6 | 13
ID | E | 2 | 0 | 0 | Table | 5 | 9 | 15 | 1 | 13 | 10 | 6 | 0 | 8 | 14 | 3 | 12 | 7 | 4 | 2 | 11
ID | E | 3 | 0 | 0 | Table | 1 | 9 | 13 | 4 | 8 | 3 | 7 | 10 | 11 | 14 | 6 | 5 | 0 | 15 | 12 | 2

All the others do not have any usefull information...

May be we should have done it differently:
ID | 3A | 99 | 22 | 3A 
-- | -- | -- | -- | --
ID | 3B | 99 | 22 | 3A 
ID | 3A | 9A | 22 | 3A 
ID | 3A | 99 | 23 | 3A 
ID | 3A | 99 | 22 | 3B 

The first line is your original radio id so don't try it but see what happens with the 4 other lines.

pascallanger, 2020-07-24:
Not sure if you are used to powershell but I'm attaching the script I'm using:
[CC2500_pv_chan.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4971552/CC2500_pv_chan.zip)
To use it:
- in pulseview do a right click on the RX SPI window part where the data appear and select "Export all annotations"
- save the file with a .txt extension
- open the powershell script, edit the second line and put the filename from the step above
- run the script in the directory where the file is
- the last ouput line of the script will give you something like: ID=	E	0	0	0	Table=	5	9	14	4	13	1	8	10	3	2	6	12	7	11	15	0 which is what I'm pasting here

zendes, 2020-07-24:
[TX IDs similar to original.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4974187/TX.IDs.similar.to.original.zip)

I'm using an OSX system, so no powershell unfortunately. I was able to find a faster way of getting the captures, so I took some time during lunch today & did the ones you asked for.

If you think there's value in getting a bunch more, I'll take a crack at automating some of this. Can you let me know what ID range you'd like the table of?

pascallanger, 2020-07-24:
I can't check the files until tomorrow unfortunately...
What I'm wondering is if there is a loop happening at some stage or if it's a pure pseudo random output.

If you can automate, then may be we could get 256 consecutive samples.
To go faster without reflashing the module, you can use the rx number. Just replace the last byte of your ID by RX_num and then in the GUI change the value. The protocol will immediately switch to the new value.

All I need from my point of view is the annotation text export from pulseview.

pascallanger, 2020-07-25:
I've checked your last file. The full 4 ID bytes are used to create the order of the hopping table...
Touching the 1st byte seems to be a bad idea since the RX is freaking out as we've seen on your previous dumps.

ID | 3A | 99 | 22 | 3A | Table | 1 | 15 | 8 | 9 | 2 | 5 | 0 | 6 | 4 | 14 | 11 | 13 | 3 | 10 | 12 | 7
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
ID | 3A | 9A | 22 | 3A | Table | 2 | 7 | 0 | 1 | 9 | 3 | 5 | 10 | 12 | 15 | 13 | 4 | 8 | 11 | 6 | 14
ID | 3A | 99 | 23 | 3A | Table | 4 | 6 | 1 | 11 | 8 | 0 | 9 | 12 | 3 | 14 | 15 | 7 | 13 | 10 | 2 | 5
ID | 3A | 99 | 22 | 3B | Table | 11 | 10 | 5 | 14 | 12 | 15 | 4 | 7 | 0 | 8 | 9 | 1 | 2 | 13 | 6 | 3
ID | 62 | 4D | F9 | 27 | Table | 0 | 4 | 7 | 2 | 9 | 3 | 8 | 11 | 5 | 6 | 15 | 13 | 12 | 10 | 1 | 14
ID | E | 0 | 0 | 0 | Table | 5 | 9 | 14 | 4 | 13 | 1 | 8 | 10 | 3 | 2 | 6 | 12 | 7 | 11 | 15 | 0
ID | E | 1 | 0 | 0 | Table | 14 | 4 | 1 | 3 | 0 | 9 | 7 | 10 | 15 | 8 | 5 | 2 | 11 | 12 | 6 | 13
ID | E | 2 | 0 | 0 | Table | 5 | 9 | 15 | 1 | 13 | 10 | 6 | 0 | 8 | 14 | 3 | 12 | 7 | 4 | 2 | 11
ID | E | 3 | 0 | 0 | Table | 1 | 9 | 13 | 4 | 8 | 3 | 7 | 10 | 11 | 14 | 6 | 5 | 0 | 15 | 12 | 2

zendes, 2020-07-26:
Making progress on the automation. I have a script that'll launch sigrok-cli to do the capture and then push the bind button via a GPIO. All I need to do is bump the RX ID by one every time it beeps at me. Here are the first captures: [TX ID 00-63.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4977210/TX.ID.00-63.zip)

I haven't figured out how to make sigrok-cli export the SPI decoder in the same format as pulseView. I can get the bytes, but not the MOSI vs MISO direction or timestamp. This is what I can get now:
```
sigrok-cli -d saleae-logic16:conn=20.27 --config samplerate=10M --channels 0,1,2,3 --protocol-decoders spi:miso=1:mosi=0:clk=2:cs=3 --protocol-decoder-annotations spi=miso-transfer:mosi-transfer --samples 100M

spi-1: 0F 24
spi-1: FB 00
spi-1: 0F 21
spi-1: BF 00
spi-1: 0F 01 3A 99 22 04 FF FB DF FF FE F7 BF FF FD EF 7F FF FB DF FF FE F7 BF FF FD EF 7F 61 25 00 00 01 6D 46 B1
spi-1: FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
spi-1: 0F
spi-1: 3A
spi-1: 0F
spi-1: 3A
spi-1: 0F
spi-1: 36
spi-1: 0F 0F
spi-1: 0A 36
spi-1: 0F
spi-1: 34
spi-1: 1F
spi-1: 3A
spi-1: 1F
spi-1: 36
spi-1: 0F 0F
spi-1: 0A 66
```

The captures are just the raw bytes with an SPI decoder attached, but not the annotations from pulseView yet...

pascallanger, 2020-07-26:
If you can get the output you are showing then I can write a decoder for it in python so you can run it.
I'm trying to get sigrok-cli to give me this output but I'm failing so far. Can you share a full file of the above?

pascallanger, 2020-07-26:
Ok running the latest nightly I'm getting the same output.
If you add --protocol-decoder-samplenum to the command line you can see which lines are related to each other. Since you have miso first on your output you know that for a same samplenum the first line is miso then second line is mosi. From there we can decode.

pascallanger, 2020-07-26:
Here it the python script:
[rlink_table.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4977624/rlink_table.zip)
To use it:
- redirect the output of sigrok-cli in a text file:
`sigrok-cli.exe -i 63.sr -P spi:miso=1:mosi=0:clk=2:cs=3 -A spi=miso-transfer:mosi-transfer --protocol-decoder-samplenum > output.txt`
- execute the python script giving the file name as input:
`python rlink_table.py output.txt`
- if all works well you get this ouput:
`ID 3A 99 22 3F Table 2 12 8 9 1 11 10 0 14 15 3 5 13 7 4 6`

pascallanger, 2020-07-26:
Here is my script output:

ID | 3A | 99 | 22 | 0 | Table | 3 | 12 | 1 | 7 | 11 | 10 | 14 | 6 | 13 | 9 | 0 | 5 | 4 | 8 | 15 | 2
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
ID | 3A | 99 | 22 | 1 | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 2 | Table | 10 | 4 | 15 | 9 | 8 | 5 | 7 | 3 | 0 | 6 | 11 | 2 | 14 | 1 | 13 | 12
ID | 3A | 99 | 22 | 3 | Table | 11 | 4 | 10 | 5 | 9 | 8 | 6 | 2 | 1 | 3 | 7 | 15 | 14 | 12 | 13 | 0
ID | 3A | 99 | 22 | 4 | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 5 | Table | 13 | 1 | 15 | 2 | 10 | 3 | 9 | 8 | 11 | 12 | 4 | 0 | 6 | 14 | 5 | 7
ID | 3A | 99 | 22 | 6 | Table | 6 | 7 | 8 | 0 | 12 | 10 | 9 | 1 | 11 | 2 | 3 | 14 | 15 | 5 | 4 | 13
ID | 3A | 99 | 22 | 7 | Table | 0 | 7 | 3 | 8 | 13 | 14 | 11 | 10 | 4 | 12 | 6 | 15 | 1 | 2 | 5 | 9
ID | 3A | 99 | 22 | 8 | Table | 0 | 4 | 7 | 6 | 8 | 11 | 9 | 12 | 10 | 14 | 1 | 5 | 13 | 15 | 2 | 3
ID | 3A | 99 | 22 | 9 | Table | 15 | 4 | 12 | 1 | 11 | 14 | 9 | 13 | 2 | 5 | 8 | 3 | 6 | 7 | 10 | 0
ID | 3A | 99 | 22 | 0A | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 0B | Table | 10 | 11 | 2 | 8 | 9 | 5 | 0 | 3 | 13 | 14 | 15 | 1 | 7 | 4 | 6 | 12
ID | 3A | 99 | 22 | 0C | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 0D | Table | 6 | 5 | 0 | 2 | 8 | 10 | 3 | 7 | 12 | 4 | 15 | 14 | 13 | 9 | 11 | 1
ID | 3A | 99 | 22 | 0E | Table | 13 | 4 | 8 | 7 | 12 | 9 | 15 | 14 | 5 | 2 | 11 | 10 | 3 | 1 | 0 | 6
ID | 3A | 99 | 22 | 0F | Table | 3 | 15 | 0 | 1 | 10 | 4 | 9 | 2 | 5 | 6 | 14 | 13 | 8 | 12 | 11 | 7
ID | 3A | 99 | 22 | 10 | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 11 | Table | 2 | 12 | 7 | 8 | 4 | 14 | 9 | 10 | 13 | 11 | 0 | 5 | 6 | 15 | 3 | 1
ID | 3A | 99 | 22 | 12 | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 13 | Table | 5 | 10 | 8 | 13 | 6 | 9 | 1 | 3 | 14 | 7 | 0 | 2 | 15 | 11 | 12 | 4
ID | 3A | 99 | 22 | 14 | Table | 12 | 7 | 15 | 4 | 6 | 5 | 3 | 14 | 11 | 2 | 0 | 10 | 8 | 13 | 9 | 1
ID | 3A | 99 | 22 | 15 | Table | 6 | 0 | 1 | 5 | 12 | 9 | 2 | 8 | 3 | 4 | 13 | 14 | 11 | 10 | 7 | 15
ID | 3A | 99 | 22 | 16 | Table | 4 | 14 | 15 | 3 | 2 | 11 | 10 | 8 | 5 | 12 | 9 | 7 | 0 | 1 | 13 | 6
ID | 3A | 99 | 22 | 17 | Table | 5 | 11 | 12 | 0 | 4 | 8 | 9 | 7 | 3 | 15 | 13 | 1 | 14 | 6 | 10 | 2
ID | 3A | 99 | 22 | 18 | Table | 10 | 4 | 15 | 6 | 13 | 1 | 3 | 5 | 14 | 7 | 11 | 12 | 0 | 2 | 9 | 8
ID | 3A | 99 | 22 | 19 | Table | 15 | 3 | 10 | 14 | 4 | 11 | 6 | 0 | 12 | 9 | 1 | 8 | 7 | 5 | 13 | 2
ID | 3A | 99 | 22 | 1A | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 1B | Table | 11 | 3 | 7 | 1 | 6 | 4 | 9 | 14 | 2 | 5 | 15 | 0 | 8 | 13 | 10 | 12
ID | 3A | 99 | 22 | 1C | Table | 4 | 15 | 0 | 10 | 12 | 9 | 11 | 14 | 6 | 3 | 5 | 1 | 8 | 7 | 13 | 2
ID | 3A | 99 | 22 | 1D | Table | 2 | -1 | 1 | 12 | 9 | 6 | 4 | 7 | 13 | 8 | 0 | 11 | 3 | 10 | 5 | 14
ID | 3A | 99 | 22 | 1E | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 1F | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 20 | Table | 4 | 6 | 10 | 0 | 5 | 1 | 14 | 2 | 13 | 15 | 12 | 7 | 8 | 3 | 9 | 11
ID | 3A | 99 | 22 | 21 | Table | 6 | 13 | 12 | 7 | 9 | 2 | 3 | 4 | 1 | 10 | 15 | 0 | 11 | 14 | 8 | 5
ID | 3A | 99 | 22 | 22 | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 23 | Table | 2 | 7 | 14 | 11 | 0 | 4 | 5 | 1 | 10 | 3 | 15 | 12 | 6 | 8 | 9 | 13
ID | 3A | 99 | 22 | 24 | Table | 3 | 7 | 13 | 12 | 15 | 8 | 1 | 14 | 4 | 0 | 5 | 11 | 9 | 2 | 0 | 0
ID | 3A | 99 | 22 | 25 | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 26 | Table | 4 | 3 | 1 | 10 | 6 | 8 | 15 | 5 | 14 | 7 | 0 | 12 | 9 | 13 | 11 | 2
ID | 3A | 99 | 22 | 27 | Table | 6 | 11 | 14 | 13 | 2 | 5 | 15 | 9 | 12 | 0 | 1 | 7 | 3 | 4 | 8 | 10
ID | 3A | 99 | 22 | 28 | Table | 6 | 8 | 11 | 12 | 14 | 15 | 10 | 1 | 7 | 3 | 13 | 9 | 2 | 0 | 5 | 4
ID | 3A | 99 | 22 | 29 | Table | 15 | 7 | 4 | 8 | 2 | 10 | 9 | 11 | 0 | 3 | 12 | 13 | 5 | 1 | 6 | 14
ID | 3A | 99 | 22 | 2A | Table | 1 | 6 | 8 | 12 | 3 | 5 | 7 | 0 | 2 | 14 | 9 | 15 | 4 | 10 | 11 | 13
ID | 3A | 99 | 22 | 2B | Table | 3 | 12 | 15 | 0 | 7 | 10 | 11 | 1 | 8 | 2 | 4 | 5 | 9 | 6 | 13 | 14
ID | 3A | 99 | 22 | 2C | Table | 13 | 1 | 0 | 5 | 9 | 7 | 2 | 6 | 14 | 4 | 15 | 11 | 8 | 12 | 3 | 10
ID | 3A | 99 | 22 | 2D | Table | 12 | 10 | 13 | 5 | 8 | 0 | 7 | 6 | 3 | 0 | 0 | 9 | 0 | 11 | 2 | 14
ID | 3A | 99 | 22 | 2E | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 2F | Table | 15 | 14 | 4 | 9 | 2 | 13 | 3 | 5 | 11 | 1 | 10 | 7 | 12 | 6 | 8 | 0
ID | 3A | 99 | 22 | 30 | Table | 4 | 15 | 5 | 9 | 12 | 7 | 8 | 6 | 3 | 1 | 10 | 2 | 11 | 14 | 13 | 0
ID | 3A | 99 | 22 | 31 | Table | 14 | 7 | 8 | 3 | 5 | 4 | 10 | 6 | 2 | 13 | 12 | 0 | 11 | 15 | 9 | 1
ID | 3A | 99 | 22 | 32 | Table | 11 | 0 | 1 | 6 | 7 | 3 | 10 | 2 | 9 | 5 | 13 | 15 | 8 | 14 | 4 | 12
ID | 3A | 99 | 22 | 33 | Table | 5 | 12 | 14 | 11 | 9 | 15 | 3 | 1 | 6 | 2 | 10 | 4 | 7 | 0 | 8 | 13
ID | 3A | 99 | 22 | 33 | Table | 5 | 12 | 14 | 11 | 9 | 15 | 3 | 1 | 6 | 2 | 10 | 4 | 7 | 0 | 8 | 13
[... cut ...]
ID | 3A | 99 | 22 | 3B | Table | 11 | 10 | 5 | 14 | 12 | 15 | 4 | 7 | 0 | 8 | 9 | 1 | 2 | 13 | 6 | 3
ID | 3A | 99 | 22 | 3C | Problem! |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
ID | 3A | 99 | 22 | 3D | Table | 3 | 6 | 9 | 12 | 0 | 15 | 1 | 11 | 4 | 14 | 10 | 7 | 13 | 2 | 5 | 8
ID | 3A | 99 | 22 | 3E | Table | 8 | 14 | 12 | 10 | 2 | 1 | 6 | 11 | 9 | 7 | 15 | 4 | 13 | 0 | 3 | 5
ID | 3A | 99 | 22 | 3F | Table | 2 | 12 | 8 | 9 | 1 | 11 | 10 | 0 | 14 | 15 | 3 | 5 | 13 | 7 | 4 | 6

pascallanger, 2020-07-26:
Problem! is indicating a problem. For example:
- 3C=60, the RX timing is fully wrong.
- 04, not enough frames to sync
- ...

pascallanger, 2020-07-26:
Here is an updated version of the script: [rlink_table.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4977700/rlink_table.zip)

pascallanger, 2020-07-26:
But from what I see above there is not a big chance that there wiil be a loop... If that's the case then the only way to go and have different IDs is to store in the fw ID and corresponding table but that's taking a lot of flash space...

alexdibi, 2020-07-26:
Maybe given my inexperience it is just stupid.... but since I read just out of curiosity and to understand something, I thought but with the same TX id on 2 different receivers the hopping table is the same or varies also based on the ID of the receiver?

[37 more replies]

closed 2020-08-01 by pascallanger