pascallanger/DIY-Multiprotocol-TX-Module #387: Devo telemetry not working with RX705

kristjanbjarni, 2020-07-13:
I have been testing the new Devo telemetry with my Walkera QR X350 Pro which is using the Devo RX705 receiver. The problem is that I get just fixed values. A1, A2 shows 0.0V and the RSSI shows 93, but this doesn't change at any time. When I turn off the quadcopter I get "Telemetry lost" on the remote so something is obviously being sent. On the original remote Devo F12E, I can get the battery voltage and the GPS coordinates which I am very interested in getting. Is there any way I can help in debugging this?

pascallanger, 2020-07-15:
I'll check what's happening with the voltage. Adding GPS will be a major change...

pascallanger, 2020-07-16:
@kristjanbjarni 
Which radio do you have?
On your Devo F12E where do you see the battery voltage appearing? volt1, volt2 or volt3? only volt1 and volt2 are displayed so I'm wondering if your value is in volt3...
93 is the max rssi value, it goes from 93 down to 0 in steps of 3. Are you sure when you get the quad away from the remote (or a couple of walls apart) that this value doesn't change?

kristjanbjarni, 2020-07-16:
I have the Jumper T16 Pro V2

But sorry it seems I was mistaken. I was checking with a battery on storage voltage 3.8v and it showed as 0.0, but when I tested with a battery that was at 3.9v, then A2 showed as 0.4v. After I changed the ratio to 1470 then it matched the actual voltage under load. I also tested the RSSI and you are correct after I moved away and a couple of walls it went down to about 60 before loosing telemetry. However there seems to be a problem with the voltage, I tried running the quadcopter under load for a few minutes and after some minutes the voltage dropped to 0.0 but when stopping the motors it went back up to correct voltage. Not sure if it's just a problem with the voltage monitor on my end or how the telemetry is configured.

richardclli, 2020-07-27:
Current Devo telemetry only works with Micro helis like MiniCP, V120D02S.  Full telemetry is not supported yet.  The current implementation of OpenTX and Multi-protocol splits telemetry capturing (in multi-module source code) and telemetry decoding (in OpenTX code), adding Devo full telemetry need to change both multi-protocol code and OpenTX code.

And the key is that I do not have the Walkera telemetry hardware to test the telemetry implementation.  If you can help, I can see if I can work this out.

kristjanbjarni, 2020-08-03:
I didn't know that the V120D02S had telemetry (I own one myself) or is that some added hardware?

Anyway I actually have a spare RX705 receiver that I could gift and mail to you or possibly sponsor a RX705 receiver bought locally? if you want. Although not sure if having only the receiver is enough for testing. But I think it's possible to hook up any standard NMEA (or Ublox?) GPS to the telemetry input on the RX705 to get live GPS coordinates.

richardclli, 2020-08-04:
Yes, MiniCP has telemetry, V120D02S has telemetry but need to upgrade firmware to activate if your firmware is old, not sure if it is still available now.

Anyway, I will try to port and build the original Devo telemetry support from Deviation and see if this works for you.  Please help me to test it out once it is ready.  The full telemetry support is already there in Deviation source code, it includes temperatures, voltages, and GPS information.  But the porting need to modify both the multiprotocol source code and opentx source code, so it means more works and need to flash both the radio and the RF module to activate.

richardclli, 2021-03-01:
@pascallanger Hey, saw that you have committed DEVO telemetry changes.  I will test it out this weekend. ^.^

pascallanger, 2021-03-01:
@richardclli That would be perfect. Do you have sensors (temperature, RPM, voltage) and GPS?
I've done quite some modifications which I have no idea if they work (I hope so) and if I've broken something (I hope not)...

pascallanger, 2021-03-02:
I've tested like I could almost everything and I think I have something working, please report.

pascallanger, 2021-04-18:
Completed

closed 2021-04-18 by pascallanger

khmewes, 2023-07-16:
I know this is closed, but can I get some insight from you on the telemetry port of the RX705 receiver. What sensor would work with it (Voltage, GPS etc.) I am using the RX705 as receiver with a Fly Wing H1 controller connected via the sbus data port (in a modded Walkera V450D01) and it works fine with the devo protocol on the RM TX16S.  I get RSSI but no voltages. What sensors/interface would be needed to get voltage and perhaps GPS to work?

pascallanger, 2023-07-17:
All the supported sensors are in the doc : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#devo---7

khmewes, 2023-07-18:
Thanks Pascal.

I was more looking for info about the interface type of the RX705 telemetry port, i.e. sbus, ibus or general serial port. Whatever Walkera designed there. I wanted to connect a voltage sensor like the FlySky FS-CVT01 ibus based one, but it wasn’t recognized at first then I inverted the signal and the led stayed on. But no communication with the transmitter (TX16S). The CVT01 connected to the FS-i6AB receiver sensor port shows up as A3 in the telemetry list with the FlySky protocol. So is this even possible with the RX705 and Devo or does this only work in combination with the Walkera QRX350 flight controller. If so I will just give up on the whole thing and use the FS-i6A receiver instead. But that’s no fun for us tinkerers.

 

Cheers,

 

Klaus

pascallanger, 2023-07-19:
I have no idea. This is the wrong place to search for this information. You should try on some Devo forums.