pascallanger/DIY-Multiprotocol-TX-Module #388: Support of Jeti Protocol?

guerradr, 2020-07-13:
Hi There,
Is it possible to add the JETI RC Protocol or do they use different hardware? I do have access to a JETI TX with some guidance I could assist to gather required data.

pascallanger, 2020-07-14:
Hi, sorry it's not doable. None of the current components can emulate it.
Pascal

valfa14, 2020-08-20:
And is it planned for next versions?

pascallanger, 2020-08-20:
I can't say...

gibbi, 2020-08-20:
it current products spektrum is using a CC2560, but they are still emulating the phy layer of a cyrf for backwards compatibility. So that chip could replace the EOL cyrf and enable JETI.

pascallanger, 2020-08-20:
I haven't looked at the encoding of Jeti so I can't say.
CC2560 is one of the component that could emulate the cyrf, like we are emulating the xn297 with the NRF and CC2500. I'm currently looking at another one which I think might be more interresting.

valfa14, 2020-08-20:
Working Jeti protocol would be totally magic, as they do great stuff but their radios are 💸 💸 . I'm not very familiar with all protocols requirements and their implementation but it would "open market" for many modellers in central Europe as it was one of the first 2.4ghz available system.

diego0815, 2021-01-06:
@valfa14  I was looking into jeti but ended up with a TX16S instead. Would have loved spending my money with a European brand, but it was not within my budget :-(

Jeti offers compatible RF modules though, maybe open TX could implement their telemetry serial protocol, to avoid the external telemetry display box.

onki69, 2021-02-15:
Hi, I have just opened some Jeti Receiver to find out what 2G4 hardware is used:
The R9, an older ATMEGA based RX from 2014 uses AT86RF231 with CC2591 as HF Front end and UPG2214 Antenna Switch.
The REX10 (from 2016) is STM32F302 based and uses AT86RF231SE2431L as HF Frontend. This also applies to the REX10A (from 2018).

Hope that helps.

diego0815, 2021-02-15:
The AT86RF231 is a:
High Performance RF-CMOS 2.4 GHz Radio Transceiver Targeted for IEEE 802.15.4™, ZigBee®, 6LoWPAN, RF4CE, SP100, WirelessHART™ and ISM Applications

As per the product page:https://www.microchip.com/wwwproducts/en/AT86RF231

It is a classical zigbee chipset and can't be emulated by any of the 4(5) chipsets presently on the module. It uses direct sequence spread spectrum based on phase shift keying, as far as I can tell the other chipsets do all fsk (maybe the cc2500 can also do ASK or OOK, but that also won't help). Yeti and Radiolink (not tje existing surface protocol) both need a zigbee chipset, like a cc2420 or whatever the market has for us. 

So as @pascallanger classified it it's a won't do, as the hardware is lacking those capabilities.

pascallanger, 2021-10-20:
I'm not sure if I should close it since people don't check closed issues and they open new ones. Adding the "Won't do" status was for me a way to show that I'm aware and won't work on it...

raphaelcoeffic, 2026-11-16:
FYI: I've just managed to communicate properly with a Jeti Duplex R6 receiver (bind, data, telemetry, remote JETIBOX). These old receivers however use only a limited set of channels. Newer receivers use more channels and I guess the bind reply contains some capability bit which is not yet identified. I'll have to retest with my friend's DS-14 and REX10.

onki69, 2026-11-16:
AFAIK Jeti uses 16 channels with 5MHz channel width.
old ATMega based R-receivers could not be updated to the LBT compatible firmware. This only works with receivers which display the firmware version in the Jetibox view. Same applies to the TX modules. AFAIK only the TU2 can be used "LBT compatible".
The updated R receivers should work similar to the REX series.

raphaelcoeffic, 2026-11-16:
> AFAIK Jeti uses 16 channels with 5MHz channel width. old ATMega based R-receivers could not be updated to the LBT compatible firmware. This only works with receivers which display the firmware [...]

The 5MHz channels (std IEEE) is what the Duplex R6 I have here does. The REX10 I sniffed yesterday used additional 1MHz channels.
The reason why the R receivers cannot use the 1MHz channels is because they use the AT86RF230, which does not support the 1MHz channels. Later on, they move to AT86RF231 which supports the additional channels.

As of now, I have only seen a "EU" receiver (my guess), so I cannot tell how the module or receiver differs in FCC mode. AFAIK, the only "EU" specificities is LBT, but we'd need somebody outside EU or with FCC firmware to tell us. The sniffer as described [here](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/507#issuecomment-6635361903) could be made to sniff Jeti frame as well, which could help identify a non EU-mode.