pascallanger/DIY-Multiprotocol-TX-Module #390: FlySky AFHDS2A and 16 channels

wimalopaan, 2020-07-15:
FlySky AFHDS2A receivers as the FS-iA6B output the channels 15 and 16 as zero (as well as 17 and 18). 

A conversation with technical support of FlySky gives confidence, that these receivers should output allways 18 channels. In case of OpenTX / MPM this is clearly up to channel 16.

So, channels 15 and 16 are somehow lost on the way OpenTx -> MPM -> RX. Looks like a bug.

Additional. is it possible to transfer all 18 channels (say channel 17 / 18 from OpenTX)?

pascallanger, 2020-07-15:
They is no bug, 14 channels is the expected behavior.
The protocol can only send 14 channels (at least at once). Unless they modified it recently,  there was no way for afhds2a to send more than 14 channels.
If you still think that more channels are doable then provide a  SPI dump with 16/18 channels enabled and moving so that they can be identified.
Multi supports 16 channels and won't be increased.

wimalopaan, 2020-07-15:
This is the spec from FlySky:

[image]

pascallanger, 2020-07-15:
Sure the iBus protocol can transport up to 18 channels since they've designed it with future in mind. But it does not mean that you can send all this channels with AFHDS2A.
Again give me a AFHDS2A SPI dump with these extra channels in use and I'll add it.

pascallanger, 2020-07-15:
I could try to send the channels as shown above. But I have no way to test if it will even work...
If I post a test source will you be able to compile by yourself and try it out?
Pascal

wimalopaan, 2020-07-15:
Yes, sure. I'll try that.

pascallanger, 2020-07-15:
Please quickly test the new branch: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/AFHDS2A_17CH
It is sending 16 channels from the TX plus RX_LQI on channel 17.

wimalopaan, 2020-07-15:
Just installed arduino in Arch Linux, but got exceptions. Tried to use arduino-builder, but that produces errors reading hardware-file ....

Is it possible to compile the source using make or cmake? I'm not an arduino user at all. All my µC projects use make or cmake ...

Would it be possible to provide me with the binary (STM32, internal, AETR)?

pascallanger, 2020-07-15:
Here you are:
[multi-stm-1.3.1.40.bin.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/4927020/multi-stm-1.3.1.40.bin.zip)

kilrah, 2020-07-15:
An iBus frame between RX and destination always carries 18CH, but that doesn't mean the RF link between radio and receiver does. 

Just like an SBUS frame which is always 16 channels (actually 18, but the last 2 are on/off only) yet some receivers that only support 8 channels over the air like FrSky D8 will output SBUS.

As Pascal said it was likely future-proofing, they already knew they would make a new protocol, and unsurprisingly AFHDS3 has 18 channels

wimalopaan, 2020-07-15:
CH15: ok
CH16: ok
CH17: outputs 2000
CH18: always 988

Looks ok?

wimalopaan, 2020-07-15:
Can we do something useful with CH17 and CH18?

pascallanger, 2020-07-15:
CH17 will contain the link quality indicator. It's currenlty not working since I've forgotten to enable it in one place...
CH18 will be unused.

I fear that old FCs which are not applying masks on the incoming ibus values will stop working... I might want to add a compilation flag for compatibility purpose...

wimalopaan, 2020-07-15:
Regarding CH17/CH18: would it be possible to transfer OpenTX CH17/Ch18 on that? Does that require a change in OpenTX?

I don't think that this would cause any problems, since FlySky themselves must have gone the same path ...

pascallanger, 2020-07-15:
It'll cause issues with all old Betaflight FCs not running the latest code: https://github.com/betaflight/betaflight/commit/588166fa56a33c01aeb2f26be4a8b0b3d0751154#diff-e54a9126c8070d840575f673ad79217a
I haven't checked the other FCs code but I'm pretty sure it's the same for everything reading the IBUS values.

Adding CH17/CH18 would require changes in OpenTX and Multi. I won't do it but feel free to create your own fork.

Pascal

wimalopaan, 2020-07-15:
Ok, I see the problem not masking the upper bits.

Well, a compilation-option looks very inflexible to me for the average user. MAybe it'll be better to have an option IBus16,PWM additionally to IBus,PWM and IBus,SBUS?

Further question: if I want to do my own fork, the data transfer from OpenTX to MPM uses 18 Channels? So it may be an option to hijack sbus ch 17/18?

pascallanger, 2020-07-15:
The earlier firmware should not be used. My mod was quick and dirty to see if that could be done.
I've updated the master branch with a better implementation. May be you can test it after fixing your environment.

For 17/18 channels, you need to modify in opentx pulses/multi.cpp to send the additiononal 2 channels. On the multi side, you need to adjust the buffer size, number of channels and position of the additional bytes (HoTT, Sport...). This is not hard to implement.

wimalopaan, 2020-09-07:
Looks like SBus-Output still have the same Problem.

pascallanger, 2020-09-07:
???

wimalopaan, 2020-09-07:
Even with your recent change we have only ch1-14 if one chooses AFHDS2A with PWM,SBUS setting. ch15,16 are not in the sbus-stream that the receiver e.g.FS-iA6B produces.

pascallanger, 2020-09-07:
Looks like this change is only half baked...

wimalopaan, 2020-09-07:
If you want me to check a patch, please let me know.

pascallanger, 2020-09-07:
There is nothing to patch. What's being sent is independent of ibus vs sbus. This is the RX.

wimalopaan, 2020-09-07:
Really? So, with your recent change, pwm.ib16 gives 16 channels on receiver ibus-output, thats verified. But pwm,sbus sends also 16 channels, but the receiver outputs only 14. So, this is an FlySky RX issue?

pascallanger, 2020-09-07:
Humm, I've checked the code doesn't care about sbus or ibus. But you can only select ibus with 16 channels. You should have started with that... From your messages above only ibus was concerned...

wimalopaan, 2020-09-07:
> Humm, I've checked the code doesn't care about sbus or ibus.

ok, so all 16 channels are send thru the air?

> But you can only select ibus with 16 channels. 

As I said above, pwm,ibus (14 channels) and pwm.ib16 (16 channels) are ok.

> You should have started with that... From your messages above only ibus was concerned...

Sorry for hijacking this issue.

Should I open a new issue, that with pwm,sbus only 14 channels are outputted from the rx?

Or is this clearly a flysky-issue? Then I will contact flysky support.

pascallanger, 2020-09-07:
Test multi v1.3.1.62

kilrah, 2020-09-07:
Nobody uses CH17-18 so it wouldn't be surprising FlySky didn't implement them on their SBUS output.

wimalopaan, 2020-09-07:
It is not about ch17-18. it is about ch 15-16!

pascallanger, 2020-09-07:
Please test as said above...

wimalopaan, 2020-09-08:
Yes, will do that and be back with the results.
Thank you for your effort.

wimalopaan, 2020-09-08:
Mmh, well, I see only these two changes:

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/commit/e2972a582375b0cb677d9fd987348be2231af8e2#diff-b4d3d77640c4649a696af2753732310f

So, nothing really changed and the subprotocol do not show up in the radio ...

pascallanger, 2020-09-08:
Try again.

wimalopaan, 2020-09-08:
Ok, now looks ok.
Thank you very much!

closed 2020-09-14 by pascallanger