pascallanger/DIY-Multiprotocol-TX-Module #391: What are the values in the frames FrskyD

JPdeRie, 2020-07-15:
Hi,
I am trying to read the battery status of the drone through FrySkyD with the iRX4 lite.
I just want to serial print the battery voltages.
Now I found the code where this information is stored but I'm a bit confused

In Telemtry.ino around line 358 there is this function that puts the bytes into the frame and send it over serial.
And from what I understand is A1 the whole battery voltage and A2 voltage per cell.
The problem is when I read this out its not what I expect, like 52. Where does this get converted to a float, or is this not related to the battery?

RSSI however gives me the values I expect, but that's kinda logical because it's not in floats. 

If anyone can help me with this, would be greatly appreciated!
Thanks in regards
`
void frsky_link_frame()
{
	frame[0] = 0xFE;			// Link frame
	if (protocol==PROTO_FRSKYD)
	{		
		frame[1] = pktt[3];		// A1
		frame[2] = pktt[4];		// A2
		frame[3] = pktt[5];		// RX_RSSI
		telemetry_link &= ~1 ;		// Sent
		telemetry_link |= 2 ;		// Send hub if available

    	debugln("we got the battery: %d",frame[1]);
    	debugln("we got the battery: %d",frame[2]);
    	debugln("we got the RSSI: %d",frame[3]);
	}
	else
		if (protocol==PROTO_HUBSAN||protocol==PROTO_AFHDS2A||protocol==PROTO_BAYANG||protocol==PROTO_NCC1701||protocol==PROTO_CABELL||protocol==PROTO_HITEC||protocol==PROTO_BUGS||protocol==PROTO_BUGSMINI)
		{	
			frame[1] = v_lipo1;
			frame[2] = v_lipo2;
			frame[3] = RX_RSSI;
			telemetry_link=0;
		}
	frame[4] = TX_RSSI;
	frame[5] = RX_LQI;
	frame[6] = TX_LQI;
	frame[7] = frame[8] = 0;
	#if defined MULTI_TELEMETRY
		multi_send_frskyhub();
	#else
		frskySendStuffed();
	#endif
}`

kilrah, 2020-07-15:
A1 and A2 are the analog inputs on some receivers, not external sensors. You must say what receiver you use and what sensor is giving you the voltage to know where it is.

pascallanger, 2020-07-15:
You've got the right value. 52 means 5.2V as simple as that.
Pascal

JPdeRie, 2020-07-15:
aah thank you all!

closed 2020-07-15 by JPdeRie