pascallanger/DIY-Multiprotocol-TX-Module #392: Unable to bind the X8R without telemetry

Salto42, 2020-07-16:
Unable to bind the second receiver without telemetry.
Configuration:
      Jumper T16   2.3.9 otx  
       Internal MPM   V1.3.1.36   FrSkyX2  LBT(UE)

       Two   X8R

Works with MPM V1.3.1.24

pascallanger, 2020-07-16:
Can you confirm that you are getting the popup menu with channels selection and telemetry?

pascallanger, 2020-07-16:
I think I've found the issue...

pascallanger, 2020-07-16:
I've just pushed a fix in master v1.3.1.41. If you compile by yourself the code it should work. Please report.

Salto42, 2020-07-16:
> Message du 16/07/20 10:30> De : "pascallanger" > A : "pascallanger/DIY-Multiprotocol-TX-Module" > Copie à : "Salto42" , "Author" > Objet : Re: [pascallanger/DIY-Multiprotocol-TX-Module] Unable to bind the X8R without telemetry (#392)> >
>
> I've just pushed a fix in master v1.3.1.41. If you compile by yourself the code it should work. Please report.

Bonjour Pascal
Tout fonctionne correctement avec la dernière édition.
Un grand merci à vous pour votre promptitude, votre travail et votre patience...

closed 2020-07-16 by pascallanger