pascallanger/DIY-Multiprotocol-TX-Module #396: Clone subprotocol for DSM

digitalonkel, 2020-07-20:
I really like the clone mode for FrSky protocol.
It would be great, if you could implement a clone mode for DSM as well (even if this would not support different RX no)

Thanks, Rainer

pascallanger, 2020-07-20:
It's easy to do from a protocol perspective but I don't know how to do it from a GUI perspective.
Saving the other TX ID is just adding under the DSM RX a Clone sub protocol (like for FrSky).
The problem I have is really on the DSM protocol and how to indicate that the Cloned mode should be active. I can't do like on FrSky and add just one Cloned entry since DSM requires 5 different subprotocols to work. I see 2 options:
- Activate the Cloned mode when the RX number is equal to 63. Simplest solution.
- Add yet another GUI line to add a check box for Cloned. The OpenTX guys are going to kill me... May be it can be instead a bind box like FrSky where you select normal or cloned, I don't know what the OpenTX team will think about this. The advantage of this solution is that if you clone a multi module and not an original TX then the RX numbers would work.
Let me know if you see another solution or which one you think is the best.

digitalonkel, 2020-07-21:
Hi Pascal,
Thank you for taking care.
For short term implementation the first option is ok from my point of view, just to have the option (and I don't want you get killed ;-)
Regards, Rainer

3djc, 2020-07-22:
No, we are not that bad, we will only leave him in boiling water for a couple of hours, or if we are in a really bad mood, something really cruel, like forcing him to read all of Trump's tweet.

Back to the point, if it is based on rx 63, how do you select RX num ? I guess you would still need it , no ? Or are you eluding that if you clone a Spektrum TX, then rx num can't be used, not sure I understand

digitalonkel, 2020-07-22:
😂
Yes for the clone mode every RX no would be the same. Not ideal, but ok for me.
Regards Rainer

pascallanger, 2020-07-22:
@3djc To do things well, I would need a selection string, something like "Cloned: No, Multi, Spektrum". There are still 2 usable bits in the option value to store this info.

3djc, 2020-07-22:
Still would like to understand, why would cloning a multi be different from clone a Spektrum TX ?

pascallanger, 2020-07-22:
Because when I implemented the protocol I did not had access to a Spektrum TX and I've never looked how they are dealing with the RX number and I still don't know right now... Spektrum implementation must be different than the one on Multi.
So there are 2 options:
 - I change the way Multi deals with RX num to comply with Spektrum. It means that all models need to be rebound...
 - I implement the Spektrum way just for the cloned mode so you have both worlds at the same time able to use RX numbers

3djc, 2020-07-22:
Well, safety first, as always, should be the rule. It seems to me to be a bit dangerous that the UI suggest RX num are in place (letting user think they actually work) (or maybe I misunderstood an you have rxnum working, but not the way Spektrum does it ?)

Anyhow, I think I would favor complying with Spektrum all the way

pascallanger, 2020-07-22:
RX num works independently of any protocols, this is why it appears on evey protocol. This is it: "you have rxnum working, but not the way Spektrum does"
I'm thinking changing DSM at this stage is going to be kind of a big thing for people having to rebind all their RXs but may be not... Personnaly I don't care...
Even if I do Spektrum all the way, I still need a menu entry to select "Mode: Normal or Cloned".

pascallanger, 2020-08-25:
@3djc not sure how to proceed with this...

cynfewl, 2020-09-29:
@pascallanger can you have a different TX ID in the clone slot for each OpenTX RX number or is there only one across all OpenTX RX numbers?

If you can do a slot for each OpenTX RX number then you can just set this up in a similar way to the existing FrSky D8 cloned mode, where the ID in the cloned TX ID is used and the OpenTX RX number has no effect on the cloned model. The user could then choose which OpenTX RX number they assign (as this would be purely cosmetic) and each cloned slot would still only control the correct receiver.

If there is only one slot per radio for each protocol DSM2_1F/DSM2_2F/DSMX_1F/DSMX_2F then we would need to have some additional interaction between the OpenTX RX number and the clone TX ID in order to control multiple receivers with an existing bind, most logically that the user must manually create the model in OpenTX with the same RX number as the model number on the original Spektrum radio (or multiprotocol module). This would be similar to how the FrSky D16 cloned modes currently work (or at least my reading of how they work).

For me the goal of cloning a DSM radio is to allow me to migrate to OpenTX with multiprotocol and this would be easiest if it does not require me to rebind the receiver as part of the process. I am happy to bind the Spektrum radio to the multiprotocol module multiple times though, as rebinding through the radio doesn't require digging a receiver out of the foam in which it is currently buried.

Unless I am missing the point of the comments above, which is entirely possible.

johnnym007, 2022-01-26:
Is there any way to move this one foward, possibly with EdgeTX? 

"I'm thinking changing DSM at this stage is going to be kind of a big thing for people having to rebind all their RXs... But may be not... Personnaly I don't care..."
Not a problem for me, but I guess it maybe more of a problem for other people.

Alternatively is this something I could hardcode and compile in my own version of code?

johnnym007, 2023-08-22:
@pascallanger - Are you aware of a way of exacting the `cyrfmfg_id` from a Spektrum radio without the model match ID (mmID)?  I can see that the Spektrum protocol XOR the mmID with cyrfmfg_id[3] (this looks the same as the multi implementation).
I am working through the implementation of a cloning DSM and whilst I can detemine this information easily myself and make things work I would like to make it a easy as possible for others wanting to clone DSM.
The only option I can think of currenlty is that the RX_num would need to be set to the the mmID prior to cloning the TX. And the only reliable way I can find of determing the mmID is to export the model from the Spektrum radio and get the value from the resulting SPM file.
Am I missing an easier way of extacting the `cyrfmfg_id` without the mmID?

pascallanger, 2023-08-22:
Have you verified how the model match is done on a genuine DSM TX ? I've implemented a xor in my code but I have no idea how Spektrum is doing it, they could use a simple addition on one byte of the ID or the full ID or something else. Based on that after we can think about solutions.

johnnym007, 2023-08-23:
Spektrum are using the same method on the DX6 and NX8.  I think I have found a way to reliably determine the `cyrfmfg_id` without the model match ID.

Create a new model in the Spektrum TX - it will be named in the following format:
**xx**: Acro
Where xx represents the model match number.  This method works even if models order has been changed or models have been deleted.

cyrfmfg_id[3]=(packet_in[3]^RX_num)^0xFF (where RX_num = **xx**-1)

I have already added subprotocols for cloning and erasing the cloned `cyrfmfg_id`, in my own fork of the code, using the FRSKY cloning as a reference.  Currently this is storing the `cyrfmfg_id`, 'type' & 'channels' as I'm guessing these might be useful for the next part of the implementation.

I only own EU radios so I can only test with the DSMX protocol.

pascallanger, 2023-08-23:
Sounds good

johnnym007, 2023-08-23:
I've been thinking about how best to implement this and reading the above...I think the easiest way would be to simply have a Cloned: No, Yes tick box added to the DSM protocol.   The user would then need to select the correct RX_num, subprotocol and channels based on the original (Spektrum) bind.  I cannot really see a way around this as the cloned TX has no way of knowing this information.  Ideally once the Cloned: Yes was selected the bind option would be disabled to avoid confusion.  Maybe the Cloned tick box could only be displayed if there is a valid (not 0xffs) cloned `cyrfmfg_id`.  Not sure if this is possible.

The second option would be to add new Cloned subprotocols to the current DSM protocol.  I guess the available cloned subprotocols could then be limited until DSM2 can be tested (implemented).

Another limitation would be a maximum of 63 models (probably ok for most folks).

Any thoughts on this?  I guess we would need to discuss with OpenTX & EdgeTX devs?

johnnym007, 2023-08-24:
@pascallanger I think the best way forward would be to use one of the unused bits in the flags byte (flags |= 0x03) to indicate whether a Cloned tickbox should be displayed.  Do you have any problems with me using this approach?

pascallanger, 2023-08-24:
I think this is fine. Need to work with EdgeTX and OpenTX to get this in.

johnnym007, 2023-08-24:
Is there a contact for EdgeTX and OpenTX or is it best to open github issues?

pascallanger, 2023-08-24:
Contacts:

- 3djc above is an OpenTX dev.
- raphael https://github.com/raphaelcoeffic is an EdgeTX dev.

But I think opening github issues and link them together would be great

3djc, 2023-08-24:
Ok for the tick box, but what I don't get is the full process for a user to clone, how is it done ?

pascallanger, 2023-08-24:
I would implement it the same way as frsky.
You bind the TX you want to dupe using the RX protocol selecting the cloned sub protocol and the correct Rx number. Then you go to DSM protocol and check the box. This will tell the module to use the cloned I'd.

johnnym007, 2023-08-24:
Agree with Pascal above ^^^^^

pascallanger, 2023-08-24:
And when it's not cloned what do you display ?

3djc, 2023-08-24:
Normal

johnnym007, 2023-08-24:
Would you send this value on byte 26 bit 2?

`// Multi V1.3.X.X -> Send byte 26, Protocol (bits 7 & 6), RX_Num (bits 5 & 4), invert, not used, disable telemetry, disable mapping`

pascallanger, 2023-08-24:
instead of this unused bit 2, don't we have some free bits in the option byte ?
I'm fine if you do the implementation.

johnnym007, 2023-08-24:
Yes - ok.  So If I am looking at this correctly for DSM then we can use bit 5 of the option byte:
optionValue (Byte 3)
Bits 0..4    num_ch
Bit 5         Not Used
Bit 6         Frame Rate
Bit 7         Max Throw

3djc, 2023-08-25:
@pascallanger , je crois que Ben (RM) essaye de te contacter sur Skype ;)

3djc, 2023-08-27:
Will you raise a PR here ? I would like to bump 'required' mpm firmware level in edgetx

johnnym007, 2023-08-27:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/pull/877

pascallanger, 2023-08-27:
It's in the test version 1.3.3.30 : https://downloads.multi-module.org/latest-test/