pascallanger/DIY-Multiprotocol-TX-Module #398: Flysky x6b reciever behaving wierd with radiomaster tx16s

LasseLandmand, 2020-07-22:
Okay, so this is kinda a weird problem. I succeeded in binding the RX to the transmitter, and I was also able to read the RSSI value. But when I connected my quad with mamba mk2 stack to my pc, the readings from the receiver tab was weird. For example, if I put the throttle at 10% beta flight would show that value (1100us) but then after a few seconds, the value would jump to 1500us. This would continue to happen on all my channels.

Also, nothing should be wrong with the x6b receiver because it was working perfectly with my previous flysky i6 transmitter and with the same quad setup.

I am using the FLYSKY AFHDS2A protocol with my tx16s internal multiprotocol, and this transmitter is also working perfectly with the same protocol and other flysky receivers.

The transmitter is updated with the lastest version of open tx (2.39) and i am running the latest of Multiprotocol (v1.3.1.36)  

Looking forward to any ideas or input :)

pascallanger, 2020-07-22:
Have you bound the receiver before upgrading the module? Can you rebind it and see if you have the same issue?
Never heard of an issue with this protocol...

LasseLandmand, 2020-07-23:
I have rebound multiple recievers with the transmitter and they all have show the same problem. At this point i am thinking of just going out and buying 3 new frsky recievers. But thanks for the help.

pascallanger, 2020-07-23:
Do you have issue only on this receiver? In other words have you tested another RX model?
Would you be able to ship me this RX?

pascallanger, 2020-08-25:
This issue has been opened for a long time without any new comments for more than a month... Can it be closed?

activa73, 2020-09-16:
I have a same  problem with all my X6B reciever and my Radiomaster TX16S with telemtry.

Telemertry is very unstable and lost then all 10 or 15 seconds with the model close or far to me.

I don't understand why? I use last FW for TX16S and use this SW revision for mutiprotocol : multi-stm-opentx-aetr-noinv-v1.3.1.49.bin.

I'm open to test somes solutions.

BR

UncleTone, 2020-09-18:
would it be possible to port the FlyPlus protocol to the MPM to get the vario and other telemetry to work properly. on the RM16s

pascallanger, 2020-09-18:
No idea what flyplus is, neither rm16s...

hpoorthuis, 2020-09-20:
I have a same problem with all my X6B receiver and my Radiomaster TX16S (latest Software)
Do get Messages that Telemetry is Lost. sometimes  every 3 Seconds sometime longer. 
on the Telemetry Tab i see that all Values go to Red for about 1 to 2 Seconds.

With an other Receiver TGY-IA10 (FlySky FS-IA10B) i do not have this Problems. it looks a little bit that the Receiver is loosing Telemetry packets 
i did Take some Photos the FlySky FS-IA10B have 6 Telemetry Parameter the X6B has 7 Parameter and the Stars are Flashing 2 x Black then one Round Missing ( with Telemetry Lost Message)
[https://drive.google.com/file/d/1GhYWUb7L3o2GzWMsgqn9m1BDPqaKWgqj/view?usp=sharing](FlySky FS-IA10B )
[https://drive.google.com/file/d/1355ku9QlPAFDwFht-ixSHjcecyN3xQtN/view?usp=sharing](FlySky X6B)
[https://drive.google.com/file/d/1ub6bvYsldRYyTJ2KlZKAlcbzUc3fqWwN/view?usp=sharing](FlySky X6B)

Firmware of the X6B:
[https://drive.google.com/file/d/1tPQ7izlu5-XkqrHpp2eWYaMZsGx4x5cu/view?usp=sharing](14 Channel Firmware)
[https://drive.google.com/file/d/1dtQkyaLYALKZzGwZNCq3S8EdJbXqW6k3/view?usp=sharing](8 Channel Firmware)

pascallanger, 2020-09-20:
I might have a fix for the x6b issue but I need to do some research in the coming days.
I hope you will be able to test quickly any new code I'll publish on master.

UncleTone, 2020-09-21:
Hi Pascall,
Sorry for the confusing message.
i and some friends are very grateful for the work you have done on the
multiProtocol Modules.
We fly gliders and have started using FlySky receivers and the altitude
Attachment FS-CAT01.
The latest open source firmware for the FS-i6 Tx is the FlyPlus

https://github.com/qba667/FlySkyI6/releases/tag/1.7.6

This upgrade allows the vario to work correctly and would be a great
addition to our RM TX16s transmitters
regards
Tony McPhee

pascallanger, 2020-09-21:
> Hi Pascall, Sorry for the confusing message. i and some friends are very grateful for the work you have done on the multiProtocol Modules. We fly gliders and have started using FlySky receivers and the altitude Attachment FS-CAT01. The latest open source firmware for the FS-i6 Tx is the FlyPlus https://github.com/qba667/FlySkyI6/releases/tag/1.7.6 This upgrade allows the vario to work correctly and would be a great addition to our RM TX16s transmitters regards Tony McPhee
> […](#)
> -- Regards Tony

Can you open your own issue since it is completely unrelated to the current one?
I'm puzzled since I've added support for all Flysky sensors not so long ago... But let's talk on this other issue.

pascallanger, 2020-09-21:
@hpoorthuis @activa73 @LasseLandmand 
Can you test v1.3.1.67 and see if it fixes your issue?

pascallanger, 2020-09-23:
@hpoorthuis Are you sure that you were testing 1.3.1.67/65 in the same conditions (same distance, orientation...). I would think that the change can either improve or do nothing but not degrade. Can you make sure to retest in the exact same conditions?

hpoorthuis, 2020-09-23:
No did test the 1.3.1.67 short distance at Home , and the 1.3.1.65 on the Field (Flying)
Will see what can Find out, but need a few Days

pascallanger, 2020-09-23:
I've pushed 69 but frankly I don't think it will change anything...

hpoorthuis, 2020-09-27:
Today i detected that after I remove the TBS Module the Problems are gone.
so it seems to go in an other direction.
any Idee?

pascallanger, 2020-09-27:
May be the TBS module was always transmitting and therefore preventing your module to receive correctly the telemetry signal.

activa73, 2020-09-27:
Hello, very strange because, I have exactly the same issue without external
module. I don't think the issue is due to external module. Weather is very
bad and I can't test new release. I keep you in touch.

sinoseque, 2020-10-14:
any update on this? i have same problem with 3 x6b receivers and v1.3.1.69

pascallanger, 2020-10-14:
> any update on this? i have same problem with 3 x6b receivers and v1.3.1.69

Is the issue still opened?

sinoseque, 2020-10-14:
> > any update on this? i have same problem with 3 x6b receivers and v1.3.1.69
> 
> Is the issue still opened?

it is, but part 2 not, anyway is there something i could help with? i mean, could help some debug traces or something like that?

activa73, 2020-12-01:
Hello Pascal, have you try my reciever and show the issue? Thank you.

pascallanger, 2020-12-02:
> Hello Pascal, have you try my reciever and show the issue? Thank you.

Sorry it's on my desk untouched. I've got some personnal issues getting in my way to get anything completed lately...

activa73, 2020-12-02:
Pas de problème.

Bonne continuation

emolina82, 2021-02-11:
I had both problems, first the receiver in betaflight worked weird. It was fix by himself ... Trying different protocols sbus ibus and didn't happen anymore. But I still having the problem with telemetry (lost and recover)

If you find a solution .. let us know, I'm running the last version of multi module and opentx under Radiomaster TX16s Multi hall gimbals

Best regards

pascallanger, 2021-02-14:
Try v1.3.2.45 from here: https://downloads.multi-module.org/latest-test/
The X6B generates loads of FEC errors making the telemetry really sparse. I've disabled the FEC check and only care if the CRC is correct. I hope it won't have bad impacts on telemetry values since only the CRC16 is protecting the payload...

emolina82, 2021-02-15:
hi @pascallanger, I've tried this change and looks like it works like a charm.

I haven't flight yet with this firmware (I suppose it is safe ¿?) but I could do if there is not risk

I've restored to the version v1.3.2.30 and the problem still happens

If you need more test on my side, please let me know and I'll try to help

Thank you so much!!!

pascallanger, 2021-02-15:
You should be able to fly safely with this test version as I'm near a release.

emolina82, 2021-02-15:
Ok, I will try as soon as possible (but maybe it'll take some time due to covid lockdowns ... )

Is it only affect to FlySky X6B receptor or to all FlySky receptor?

BR

activa73, 2021-02-15:
Hello All,
I'm, happy that you have find a solution for this issue.
BR

pascallanger, 2021-02-15:
@emolina82
As said earlier the change affects all RXs.

pascallanger, 2021-02-15:
@activa73
As soon as I have good reports i can send you the RX back.

activa73, 2021-02-16:
Thank you Pascal, take your time.

pascallanger, 2021-03-02:
@LasseLandmand @hpoorthuis @emolina82 @Tygrys-1 Has anyone tested the change?
I still have @activa73 RX that I've used to troubleshoot and I would like to send it back...

Tygrys-1, 2021-03-02:
Change tested on version 1.3.2.51. Looks like all 5 receivers I tested now are working ok. Telemetry is stable and values looks correct. Good work and thank you for this. :-) I needed to rebind some of my receivers, but AFAIR this was announced some version ago, so no problem.

Whatever this FEC is, if it adds more safety for the protocol than it might be valuable to have for other receiver type. Can you make an option or a sub protocol for the X6B-s, and enable FEC on other AFHDS2A receivers?

I want to share my observations from the tests. With 4 of my receivers everything was smooth, but the last one did act different. 
1) I powered the receiver and transmitter and it was bound, but no telemetry. I deleted the sensors and tried to discover - none was found.
2) I rebinded the receiver and tried to discover sensors again. At first it has not found anything. After some time (1 min?) sensors have been discovered, but stayed in red, occasionally blinking black, with low TRSS values ie. 12 and with low RSSI. After some time (20s) sensors stayed black, but still changing and with low values on TRSS. 
3) I tried to move the transmitter and reoriented it's antenna, and after a few seconds "normal" TRSS of 248 started to show, sometimes dropping to 100 or so, and after that (next 20s or so) it stabilized itself at constant "good" values, and now it is stable. 4) I tried to switch off/on receiver, transmitter and everything is rock solid and ok. 

As I remember, this receiver stopped to send the telemetry during my earlier tests and was not doing that even with the original i6X Flysky radio. Now it returned "back to live", but it was not an instant improvement, but rather a process. 
Have you any idea, what happened? Does the receiver uses some "autotune" or maybe other error correcting algorithm, which made it "out of sync" with telemetry on my previous tests with the earlier multiprotocol software, and now it returned back to the telemetry working state?

pascallanger, 2021-03-02:
Thanks for testing.
No idea what your 4th RX is doing...
Let's close this issue then.
@activa73 can you PM me to make sure I have the correct return address?

closed 2021-03-02 by pascallanger

activa73, 2021-03-02:
Hello Pascal,
what is your PM address please?

pascallanger, 2021-03-02:
hpnuts on RCGROUPS

emolina82, 2021-03-02:
hi @pascallanger 

I tested the last weekend and everything looks fine. I only had two telemetries lost alerts, I suppose this is normal in some scenarios.

Thanks!!!!