pascallanger/DIY-Multiprotocol-TX-Module #400: Trouble understanding the payload of FrSky_X

JPdeRie, 2020-07-23:
Hi Im learning more and more about using the multimodule and I want use it to have the drone send senor values to my pc.
Now I'am using FrSky_X and I am trying to puzzle out how this package is organized.
If I read the RX buffer of the telemetry of the frsky_x by uncommenting the debug lines in ReadFrSkyX(), I get something like this
Telem: 0E 50 9F 02 E3 42 05 7E 1B 10 10 02 00 A1 AF 1F B1

And after following the code I came upon the fact that buffer[7] to [12] is the payload : 7E 1B 10 10 02 00.
But then this data disappears in p->payload[i] = buffer[i+7] 

Now what I hoped was that the first 2 bytes are the sensor ID and the other 4 bytes are the sensor values. But I cant find any correlation with the ID's that are set in betaflight, and the ones I added (Ive tested this on the X-lite and there they do show up in the telem tab). There might be some bit shifting and masking that I dont know about, but I lost my trail.

again, if someone has some information or can give some direction, that would be greatly appreciated.
I also want to say that this repository has been a great help for me so far, Thanks!

Kind regards

pascallanger, 2020-07-23:
The telemetry that you are looking at is a stream. It means that it does not contain the full message. You need to  concatenate a few of them to get the message, this is what the code is doing after, grouping them together, unstuff, then send to the radio.
Unless you want to rewrite the parser, you should put a debug output here instead since you want both SerialWrite (the one pointed out and the one below) to be echoed on your PC: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/75dc616130bfe854c2d684c3f9ae825092c0d85f/Multiprotocol/Telemetry.ino#L665

JPdeRie, 2020-07-23:
Ive been trying this out but still have no idea what to correlate this with.:

Telem: 0E 50 9F 02 4E 11 05 7E 1B 10 30 07 00 52 EB 29 B3
Short CB:800
Telem: 0E 50 9D 02 EE 62 04 6D 00 00 00 07 00 DF 7A 2A AF
Short CB:800
Telem: 0E 50 9F 02 4E 62 04 6D 00 00 00 07 00 9D EB 2A B3
Short CB:800
Retry:2
pktx:  1B 10 30 07 6D 00 00 00 00
p[i] (dec) 27 16 5 241 2 35 15 0
p[i] (hex) 1b 10 05 f1 02 23 0f 00

P[i] is what the serial_Write line would print
pktx is the pkg that comes out of proces_sport_data()

The thing is, I know betaflight sends the sensor ID and the sensor data over FrSkyX, and the X-lite is able to show this data on the screen through the multimodule. So i know that this information must be somewhere in these packages but I cant find it.

pascallanger, 2020-08-25:
Have a check at OpenTX. All the telemetry decoding is done there.
Start here and make your way through: https://github.com/opentx/opentx/blob/9f4ab50e1bc139d15e247c3774df2be487f33cd0/radio/src/telemetry/multi.cpp#L389

closed 2020-08-25 by pascallanger