pascallanger/DIY-Multiprotocol-TX-Module #401: Radiolink AIR Protocol (AT- Series Specifically)

maybenikhil, 2020-07-23:
I recently transitioned from Radiolink to a Radiomaster TX, but I want to be able to bind to my older quads without switching out their RXes. I have pretty much no experience previously with breaking down RF protocols, but the chip in the AT9S is a cc2533. I can try to help give data for the protocol, but currently do not have access to a logic analyzer.

kilrah, 2020-07-23:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/386

pascallanger, 2020-07-23:
The multi module does not have the RF component to support/emulate the AT9S hardware.
I'm currently adding support for the air protocol Radios/T Series  and surface protocol Radios/RC.

maybenikhil, 2020-07-24:
Alright. Im gonna look at the AT9 (non-s) but sounds good.

pascallanger, 2020-07-24:
Keeping it open so people stop asking...

maybenikhil, 2020-07-24:
Ah, smart. And, checked the AT9, and (of course) it's also a CC2533. Hopefully eventually it will be supported but i love the work you are doing.

diego0815, 2021-01-06:
@maybenikhil cc2533 is a ZigBee radio (ieee802.15.4) there are a couple of protocols based on this most famous is jeti if I'm not wrong. To get this going we would need a suitable HW first, then someone who would be capable and willing to reverse the protocol ;-) 

I'm dreaming up some ideas about using an nrf52840 for part of the xn297 and NRF24L01 protocols, plus it would also come with a 802.15.4 TRX but unfortunately for the foreseeable future this will remain in dreaming mode, at least as this concerns me.

drtrigon, 2021-04-24:
Will the RadioLink R9D receiver be supported, or does it use the wrong chip?

pascallanger, 2021-04-24:
As you can see on the vendor website this is a RX for AT radios so no... The list of compatible receivers is listed here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#radiolink---74