pascallanger/DIY-Multiprotocol-TX-Module #402: No sync error during Multimodule update

rgtoth, 2020-07-25:
Leave shortened version of **multi-stm-opentx-aetr-inv -v1.3.1.36.bin** in firmware file for external flashing. Flashing starts but then get error **no sync**
1. Irange4 in 1 module
2. previous version 1.3.0.95
3. other flash method will not erase nor overwrite

pascallanger, 2020-07-25:
Looks like you need to update your bootloader. Check the latest version of Flash Multi to do this.

benlye, 2020-07-25:
You need to use the latest version of Flash Multi to replace the bootloader on your module.
https://github.com/benlye/flash-multi
https://github.com/benlye/flash-multi#upgrade-module-bootloader

Video showing the process:
https://www.youtube.com/watch?v=WNCxTXLtU1o&t

rgtoth, 2020-07-25:
flash-multi-0.4.2   STMMulti4.bin will not load
Loading: multi-stm-opentx-aetr-inv -v1.3.1.36.bin,  could not find multimodule in DFU mode- failed
DFU needs installation  where?

benlye, 2020-07-25:
Did you install the drivers?

rgtoth, 2020-07-25:
More specific
Went into application actions Installed USB drivers ok :  reset module to dfu mode ok: tried to upgrade module bootloader - failed - serial baud set 115200.
Install firmware - looking for DFU  - failed

benlye, 2020-07-25:
What are the LEDs on the module doing? It sounds like it might be leaving DFU mode and switching back to serial mode before the app tries to write to it.

Can you try the process here instead? It does the same thing, but might be quick enough to get around the problem.

https://github.com/benlye/multiBootReloader/

benlye, 2020-07-25:
It will be interesting to know if MultiBootReloader script works.  I have a theory which may explain why Flash Multi isn't working and will work on it tomorrow.

rgtoth, 2020-07-25:
I used bootreloader.bin  tried to flash as firmware. error: **was compiled w/o usb support** use another firmware file After several hours of trying your input and other input I could not erase or write to the module. I must have corrupted the software when I updated last time. It is working  ok but it just keeps blinking. Can I remove blinking?

benlye, 2020-07-26:
You cannot flash bootreloader.bin like firmware.  You needed to follow the process as documented - it's really simple, so I don't know why you would try it any other way.  I've no idea what state your module will be in now, and recovering it is probably going to be harder.

If it's 'just blinking' it's not working.  You need to follow the instructions you've been given, and if they don't work you need to report the actual errors you get.

Start by plugging the module in and seeing what device shows up in Device Manager.  That should tell us where you are currently at.  If you can make a video of the LED patter that would also help.

rgtoth, 2020-07-26:
Went into zadig and replaced driver per info. then I downloaded the **multi-stm-opentx-aetr-inv -v1.3.1.36.bin**, without incident. Working again
thanks for the guidance and the help

closed 2020-07-26 by rgtoth

stanislavdavid, 2020-10-20:
Hi I have same problem but I made fresh update of the module over flash multi with this result:
Reading MULTI-Module via USBasp
[1/2] Reading flash ... done
[2/2] Reading EEPROM ...  done

Multi Firmware Version:   1.3.1.69 (AVR)
Expected Channel Order:   AETR
Multi Telemetry Type:     OpenTX
Invert Telemetry Enabled: True
Flash from Radio Enabled: True
Bootloader Enabled:       True
USB Serial Support:       False
Serial Debug Enabled:     False

EEPROM Global ID:         0x96B28DBB

MULTI-Module read successfully

Then I try to upload different bin file from radio and it show me No Sync error
I have latest versions of all
Multimodule is from banggood atmel mega328p YRFM02_V1.2

stanislavdavid, 2020-10-20:
This is the log from writing firmware
Writing MULTI-Module via USBasp
[1/5] Erasing flash ... done
[2/5] Setting fuses ... done
[3/5] Writing bootloader ... done
[4/5] Writing firmware ... done
[5/5] Verifying flash ... done

MULTI-Module firmware written successfully.

benlye, 2020-10-20:
@stanislavdavid this issue is closed and you have a different problem (it can't be the same thing because you are using an Atmega module). Can you open a new issue?